margelo/react-native-vision-camera #3069: 🐛 Crash on device without any cameras

alkhatera, 2024-07-10:
### What's happening?

When I used the package with a device that had cameras, it worked perfectly. However, when run with any device without cameras, as soon as the app starts it crashes.

**Note**: I put a sample Camera Device since the bot will close any issues without Camera Device

### Reproduceable Code

```tsx
const cameraRef = React.useRef<Camera | null>(null);
const devices = useCameraDevices();

const [cameraType, setCameraType] = React.useState<CameraDevice | null>(
    devices.length ? devices[0] : null,
  );

const codeScanner = useCodeScanner({
    codeTypes: ['code-128', 'code-39', 'code-93', 'codabar', 'ean-13', 'ean-8'],
    onCodeScanned: codes => {
      onReadBarcode(codes[0].value || '');
    },
  });

return <View>
<RNCamera
          ref={cameraRef}
          device={cameraType}
          isActive
          audio={false}
          focusable
          codeScanner={codeScanner}
          enableZoomGesture
          resizeMode="cover"
        />
</View>
```

### Relevant log output

```shell
Camera LENS_FACING_FRONT verification failed java.lang.IllegalArgumentException: No available camera can be found at androidx.camera.core.CameraSelector.select(CameraSelector.java:107) at androidx.camera.core.impl.CameraValidator.validateCameras(CameraValidator.java:124) at androidx.camera.core.CameraX.lambda$initAndRetryRecursively$2$androidx-camera-core-CameraX(CameraX.java:336) at androidx.camera.core.CameraX$$ExternalSyntheticLambda0.run(Unknown Source:12) at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) at java.lang.Thread.run(Thread.java:923)
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 6,
  "minZoom": 1,
  "maxExposure": 4,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 0,
  "minExposure": -4,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

iMin Swan

### VisionCamera Version

^4.4.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-07-10:
This is intended behaviour. Are you not using TypeScript? In your example code, you could pass `null`/`undefined` to `device={...}`, which is not allowed by VisionCamera. **This should give you a TypeScript error already.**

So what you need to do (which is also explained in the docs/examples here) is to only render the Camera component if `device` is non-null.

See this code in the README:

[image]

alkhatera, 2024-07-10:
Hi @mrousavy 

Don't worry about my example. The issue is that I have a device that has NO CAMERAS. As soon as the app runs it crashes with the error:
`Camera LENS_FACING_FRONT verification failed java.lang.IllegalArgumentException: No available camera can be found at androidx.camera.core.CameraSelector.select(CameraSelector.java:107) at androidx.camera.core.impl.CameraValidator.validateCameras(CameraValidator.java:124) `

You can reproduce the issue by using an Android emulator and remove the the front and back cameras from the emulator.

farkerhaiku, 2024-07-16:
> is weird, I don't get it - how can this code be reached if device is null? Usually this should throw a no-device error. Are you sure this device does not have any Cameras?

A few weeks ago, my pixel 7 pro started to see this same error using the expo-camera library. I can assure you it has cameras :) I believe this started after a recent android security update. This week I started to see the same issue in crashlytics on Samsung, so I think they got the same update.

mrousavy, 2024-07-17:
Thanks for the info! I guess then there's nothing I can do, we can only wait for a fix in either CameraX or Android OS.

closed 2024-07-17 by mrousavy

MFazio23, 2024-08-08:
We're facing this same issue on our app that wasn't there previously. The crash is coming from `CameraDevicesManager.initialize()`, and specifically the `ProcessCameraProvider.getInstance(...).await(...)` call in there.

Is there any reason this exception cannot be handled so the app is still usable, and instead we can reference something like `Camera.getAvailableCameraDevices()` to make sure the camera is available?

mrousavy, 2024-08-09:
@MFazio23 well can you post the exception? I wonder what throws in that func.

MFazio23, 2024-08-09:
> @MFazio23 well can you post the exception? I wonder what throws in that func.

It's `androidx.camera.core.InitializationException`:
```
androidx.camera.core.InitializationException: androidx.camera.core.CameraUnavailableException: Device reporting less cameras than anticipated. On real devices: Retrying initialization might resolve temporary camera errors. On emulators: Ensure virtual camera configuration matches supported camera features as reported by PackageManager#hasSystemFeature. Available cameras: 0
	at androidx.camera.core.CameraX.lambda$initAndRetryRecursively$2$androidx-camera-core-CameraX(CameraX.java:433)
	at androidx.camera.core.CameraX$$ExternalSyntheticLambda0.run(Unknown Source:12)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
	at java.lang.Thread.run(Thread.java:1012)
Caused by: androidx.camera.core.CameraUnavailableException: Device reporting less cameras than anticipated. On real devices: Retrying initialization might resolve temporary camera errors. On emulators: Ensure virtual camera configuration matches supported camera features as reported by PackageManager#hasSystemFeature. Available cameras: 0
	at androidx.camera.core.CameraX.lambda$initAndRetryRecursively$2$androidx-camera-core-CameraX(CameraX.java:433) 
	at androidx.camera.core.CameraX$$ExternalSyntheticLambda0.run(Unknown Source:12) 
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145) 
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644) 
	at java.lang.Thread.run(Thread.java:1012) 
```

farkerhaiku, 2024-08-09:
We switched to the expo-camera library, but it suffered from the same problem. we used the patch-package library to wrap the offending call in a try catch block and propogate an error back up to react native, allowing us to show a different view when something goes bad at the native level

farkerhaiku, 2024-08-09:
Are you only seeing this issue on pixel devices @MFazio23 ?

MFazio23, 2024-08-09:
> Are you only seeing this issue on pixel devices @MFazio23 ?

Nope, we're getting prod crashes on Samsungs and Motos as well.

MAKARD, 2024-08-19:
> It's `androidx.camera.core.InitializationException`:
>
> ```
> androidx.camera.core.InitializationException: androidx.camera.core.CameraUnavailableException: Device reporting less cameras than [...]

The same for me, affected devices are

- Xperia 1 IV
- P30 Pro 
- Pixel 7a
- Galaxy J6+
- V2022
- moto g pro
- Galaxy A13
- Xperia 10 II
- Redmi Note 11
- Pixel 6
- Pixel 8 Pro
- Xperia 1

MAKARD, 2024-08-19:
jfyi I made this patch to at least prevent the app from crashing
```
diff --git a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraDevicesManager.kt b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraDevicesManager.kt
index 581d003..1604c32 100644
--- a/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraDevicesManager.kt
+++ b/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/CameraDevicesManager.kt
@@ -62,13 +62,30 @@ class CameraDevicesManager(private val reactContext: ReactApplicationContext) :
   override fun initialize() {
     super.initialize()
     cameraManager.registerAvailabilityCallback(callback, null)
+
+    val eventEmitter = reactContext.getJSModule(RCTDeviceEventEmitter::class.java)
+
     coroutineScope.launch {
-      Log.i(TAG, "Initializing ProcessCameraProvider...")
-      cameraProvider = ProcessCameraProvider.getInstance(reactContext).await(executor)
-      Log.i(TAG, "Initializing ExtensionsManager...")
-      extensionsManager = ExtensionsManager.getInstanceAsync(reactContext, cameraProvider!!).await(executor)
-      Log.i(TAG, "Successfully initialized!")
-      sendAvailableDevicesChangedEvent()
+      try {
+        Log.i(TAG, "Initializing ProcessCameraProvider...")
+        cameraProvider = ProcessCameraProvider.getInstance(reactContext).await(executor)
+        Log.i(TAG, "Initializing ExtensionsManager...")
+        extensionsManager =
+          ExtensionsManager.getInstanceAsync(reactContext, cameraProvider!!).await(executor)
+        Log.i(TAG, "Successfully initialized!")
+        sendAvailableDevicesChangedEvent()
+        eventEmitter.emit("CameraDevicesInitializationFinished", null)
+      } catch (exception: Exception) {
+        val exceptionMap = Arguments.createMap()
+
+        exceptionMap.let {
+          it.putString("name", exception.javaClass.canonicalName)
+          it.putString("message", exception.message)
+          it.putString("stack", exception.stackTraceToString())
+        }
+
+        eventEmitter.emit("CameraDevicesInitializationFinished", exceptionMap)
+      }
     }
   }
 
diff --git a/node_modules/react-native-vision-camera/lib/typescript/Camera.d.ts b/node_modules/react-native-vision-camera/lib/typescript/Camera.d.ts
index 84d8b74..2b19008 100644
--- a/node_modules/react-native-vision-camera/lib/typescript/Camera.d.ts
+++ b/node_modules/react-native-vision-camera/lib/typescript/Camera.d.ts
@@ -240,6 +240,8 @@ export declare class Camera extends React.PureComponent<CameraProps, CameraState
      * If you use Hooks, use the `useCameraDevices()` hook instead.
      */
     static addCameraDevicesChangedListener(listener: (newDevices: CameraDevice[]) => void): EmitterSubscription;
+
+    static addCameraInitializationListener(listener: (error?: Error) => void): EmitterSubscription;
     /**
      * Gets the current Camera Permission Status. Check this before mounting the Camera to ensure
      * the user has permitted the app to use the camera.
diff --git a/node_modules/react-native-vision-camera/src/Camera.tsx b/node_modules/react-native-vision-camera/src/Camera.tsx
index d0cd767..6ece88e 100644
--- a/node_modules/react-native-vision-camera/src/Camera.tsx
+++ b/node_modules/react-native-vision-camera/src/Camera.tsx
@@ -418,6 +418,11 @@ export class Camera extends React.PureComponent<CameraProps, CameraState> {
   public static addCameraDevicesChangedListener(listener: (newDevices: CameraDevice[]) => void): EmitterSubscription {
     return CameraDevices.addCameraDevicesChangedListener(listener)
   }
+
+  public static addCameraInitializationListener(listener: (error?: Error) => void): EmitterSubscription {
+    return CameraDevices.addCameraInitializationListener(listener)
+  }
+
   /**
    * Gets the current Camera Permission Status. Check this before mounting the Camera to ensure
    * the user has permitted the app to use the camera.
diff --git a/node_modules/react-native-vision-camera/src/CameraDevices.ts b/node_modules/react-native-vision-camera/src/CameraDevices.ts
index 5866143..6d94cd5 100644
--- a/node_modules/react-native-vision-camera/src/CameraDevices.ts
+++ b/node_modules/react-native-vision-camera/src/CameraDevices.ts
@@ -12,6 +12,7 @@ const constants = CameraDevicesManager.getConstants()
 let devices = constants.availableCameraDevices
 
 const DEVICES_CHANGED_NAME = 'CameraDevicesChanged'
+const DEVICES_INITIALIZATION_NAME = 'CameraDevicesInitializationFinished'
 // eslint-disable-next-line @typescript-eslint/no-explicit-any
 const eventEmitter = new NativeEventEmitter(CameraDevicesManager as any)
 eventEmitter.addListener(DEVICES_CHANGED_NAME, (newDevices: CameraDevice[]) => {
@@ -24,4 +25,7 @@ export const CameraDevices = {
   addCameraDevicesChangedListener: (callback: (newDevices: CameraDevice[]) => void) => {
     return eventEmitter.addListener(DEVICES_CHANGED_NAME, callback)
   },
+  addCameraInitializationListener: (callback: (error?: Error) => void) => {
+    return eventEmitter.addListener(DEVICES_INITIALIZATION_NAME, callback)
+  }
 }
```

There is a listener for the initialisation event because I'd prefer not to lose visibility on this issue in production

fabricio-v, 2024-08-27:
any solution for this problem?

andho, 2024-08-31:
I'm doing development using Waydroid (no cameras), and a blank app crashes when vision camera is installed. @mrousavy can this issue be reopened and/or the patch integrated?

mrousavy, 2024-09-02:
I don't have any free time for this right now. But this can be fixed in around 4 hours or so I think with enough testing.

The patch above is a workaround which I don't want to merge. A good fix is to simply tell JS that there are no cameras, which is exactly what is expecetd here.

I guess I'd do this thru GitHub sponsors.

corbella83, 2024-10-18:
> Camera

I don't see why the patch is not a good fix for you. Libraries should never crash, and if they do for some reason, they just have to propagate the exception so that the user can decide what to do. It's well known that is a bad practice to launch a code in background and don't handle the exceptions. And that's what the patch is just about: notify any error at the initialization process instead of crashing. 

Also your "good fix" won't prevent other crashes that may occur at the initialization process.

Brma1048, 2024-11-28:
any updates on this?

hoanght1995, 2025-01-02:
I encountered this error when running e2e testing too. any update?