margelo/react-native-vision-camera #3070: 🐛  intermittent Crash Due to Permission

rsainiWin, 2024-07-11:
### What's happening?

I want to take a picture.
Sometime very rare but when a user takes a picture this leads to a crash going via the crash reporting tool. It says it happened due to NullPointerException: Attempt to invoke virtual method 'void android.view.View.dispatchCancelPendingInputEvents()' on a null object reference
Backtracking further it takes me to the Camera Module

### Reproduceable Code

```tsx
// hooks
  const {hasPermission: hasCameraPermission, requestPermission} =
    useCameraPermission();
  const handlePermission = async () => {
    await requestPermission();

    const audioPermission = Camera.getMicrophonePermissionStatus();
    if (audioPermission !== 'granted') {
      await Camera.requestMicrophonePermission();
    }
  };
  useEffect(() => {
    handlePermission();
    navigation.setOptions({headerShown: false});
  }, [navigation]);

return (
{hasCameraPermission && device != null && isEnabled && (
                <PinchGestureHandler
                  onGestureEvent={onPinchGesture}
                  enabled={isActive}>
                  <Reanimated.View
                    onTouchEnd={onFocusTap}
                    style={StyleSheet.absoluteFill}>
                    <TapGestureHandler onEnded={onDoubleTap} numberOfTaps={2}>
                      <ReanimatedCamera
                        style={StyleSheet.absoluteFill}
                        device={device}
                        isActive={isEnabled}
                        ref={camera}
                        onInitialized={onInitialized}
                        onError={onError}
                        enableHighQualityPhotos={false}
                        onStarted={() => 'Camera started!'}
                        onStopped={() => 'Camera stopped!'}
                        format={format}
                        photoQualityBalance={'balanced'}
                        // fps={fps}
                        // photoHdr={format?.supportsPhotoHdr && enableHdr}
                        // videoHdr={format?.supportsVideoHdr && enableHdr}
                        lowLightBoost={
                          device.supportsLowLightBoost && enableNightMode
                        }
                        enableZoomGesture={false}
                        animatedProps={cameraAnimatedProps}
                        exposure={0}
                        // enableFpsGraph={true}
                        orientation="portrait"
                        photo={true}
                        video={false}
                        audio={hasMicrophonePermission}
                      />
                    </TapGestureHandler>
                  </Reanimated.View>
                </PinchGestureHandler>
              )}
              {!hasCameraPermission && (
                <View
                  style={[
                    StyleSheet.absoluteFill,
                    {
                      flexDirection: 'column',
                      alignItems: 'center',
                      justifyContent: 'center',
                      backgroundColor: themeGradient.lightColors?.grey0,
                      paddingHorizontal: 24,
                    },
                  ]}>
                  <View style={{gap: 24}}>
                    <Icon
                      name="camera-off"
                      type="feather"
                      size={isTablet() ? 128 : 72}
                      style={{opacity: 0.6}}
                      color={'white'}
                    />
                    <Text
                      style={[
                        TextFontStyles.textBody,
                        TextColorStyles.white,
                        TextAlignmentStyles.center,
                      ]}>
                      You've not given access to the camera, click on the button
                      below to give access.
                    </Text>
                    <WSGredientButton
                      onPress={async () => await Linking.openSettings()}
                      icon={{name: 'camera', size: 24, color: 'white'}}
                      size="lg"
                      title="Allow Camera Access"
                    />
                  </View>
                </View>
              )})
```

### Relevant log output

```shell
java.lang.NullPointerException: Attempt to invoke virtual method 'void android.view.View.dispatchCancelPendingInputEvents()' on a null object reference
	 at android.view.ViewGroup.dispatchCancelPendingInputEvents(ViewGroup.java:4948)
	 at android.view.ViewGroup.dispatchCancelPendingInputEvents(ViewGroup.java:4948)
	 at android.view.ViewGroup.dispatchCancelPendingInputEvents(ViewGroup.java:4948)
	 at android.view.ViewGroup.dispatchCancelPendingInputEvents(ViewGroup.java:4948)
	 at android.view.ViewGroup.dispatchCancelPendingInputEvents(ViewGroup.java:4948)
	 at android.view.ViewGroup.dispatchCancelPendingInputEvents(ViewGroup.java:4948)
	 at android.view.ViewGroup.dispatchCancelPendingInputEvents(ViewGroup.java:4948)
	 at android.view.ViewGroup.dispatchCancelPendingInputEvents(ViewGroup.java:4948)
	 at android.view.ViewGroup.dispatchCancelPendingInputEvents(ViewGroup.java:4948)
	 at android.view.ViewGroup.dispatchCancelPendingInputEvents(ViewGroup.java:4948)
	 at android.view.View.cancelPendingInputEvents(View.java:23371)
	 at android.app.Activity.cancelInputsAndStartExitTransition(Activity.java:5800)
	 at android.app.Activity.startActivityForResult(Activity.java:6600)
	 at android.app.Activity.requestPermissions(Activity.java:5665)
	 at com.facebook.react.ReactActivityDelegate.requestPermissions(ReactActivityDelegate.java:191)
	 at com.facebook.react.ReactActivity.requestPermissions(ReactActivity.java:110)
	 at com.mrousavy.camera.CameraViewModule.requestPermission(CameraViewModule.kt:226)
	 at com.mrousavy.camera.CameraViewModule.requestCameraPermission(CameraViewModule.kt:234)
	 at java.lang.reflect.Method.invoke(Native Method)
	 at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372)
	 at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:188)
	 at com.facebook.jni.NativeRunnable.run(Native Method)
	 at android.os.Handler.handleCallback(Handler.java:958)
	 at android.os.Handler.dispatchMessage(Handler.java:99)
	 at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:27)
	 at android.os.Looper.loopOnce(Looper.java:230)
	 at android.os.Looper.loop(Looper.java:319)
	 at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:228)
	 at java.lang.Thread.run(Thread.java:1012)
```

### Camera Device

```json
Not limited to a specific device
```

### Device

Samsung Galaxy S23 Ultra,,Google Pixel 7 , Samsung Galaxy S21 FE 5G , Samsung Galaxy S22,Samsung Galaxy S23 , Samsung Galaxy S24 , Samsung Galaxy Tab S6 Lite , Samsung Galaxy Tab S7+ 5G (SM-T978U)

### VisionCamera Version

4.0.0-beta.13

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rsainiWin, 2024-07-11:
It was the latest at the point of time
And current latest build requires us to update react native to 0.74
We've updated to the latest build but this is pipeline

mrousavy, 2024-07-11:
Not sure if I understood that, but can you please try to upgrade to VisionCamera 4.4.2 and try if this problem still exists?

rsainiWin, 2024-07-15:
I've tried with latest version as well
The crash happened randomly after giving the permission .
I wasn't able to replicate it next time

mrousavy, 2024-07-23:
To be honest this doesn't sound like a VisionCamera issue - maybe you are rolling a custom RN fork or an old RN version, but I have never seen this happen before.

closed 2024-07-23 by mrousavy

rsainiWin, 2024-07-25:
I'm using the latest RN and I'm not using a custom fork 
Let me know if i can help any further

srgoogle23, 2024-08-29:
@mrousavy and @rsainiWin finded a issue on rn repo, https://github.com/facebook/react-native/issues/41077
and it's about react-native-vision-camera and the same problem