margelo/react-native-vision-camera #3071: 🐛 Camera access prompt appears prematurely

jkaufman, 2024-07-11:
### What's happening?

RNVC is triggering a system camera access prompt on iOS before any explicit use of the camera. In fact, we can reproduce with a reference to `Camera` in a conditional branch that never executes.

[We've posted a **standalone** minimal reproducible example project, here](https://github.com/sitewireco/rnvc-ios-permission-bug)

### Reproduceable Code

```tsx
useEffect(() => {
    console.log('foobar')
    if (false) {
      c = Camera
    }
  }, [])
  ```
```

### Relevant log output

```shell
Too large to submit. Please see here for the device logs: https://github.com/sitewireco/rnvc-ios-permission-bug/blob/main/rnvc-repro-log
```

### Camera Device

```json
n/a
```

### Device

iPhone 12

### VisionCamera Version

4.4.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-07-11:
Hm, it could be this code right here: https://github.com/mrousavy/react-native-vision-camera/blob/b1c8999602d25a9d74b673a7d7b19f948b8df636/package/ios/Core/Extensions/AVCaptureDevice%2BsensorOrientation.swift#L18-L66

mrousavy, 2024-07-11:
Unfortunately Apple/iOS does not provide an API to get the native sensor orientation of the given Camera device, so I have to construct a temporary capture session and figure out what orientation it usually creates buffers in - I don't start this capture session, but maybe iOS still doesn't allow it even if it is just created?

I am not sure about this I thought starting the session would be the point where you'd need permission...

jkaufman, 2024-07-11:
I'm guessing it's instantiation of `AVCaptureDeviceInput`.

> Capturing media requires explicit permission from the user. An app’s default authorization status is [AVAuthorizationStatus.notDetermined](https://developer.apple.com/documentation/avfoundation/avauthorizationstatus/notdetermined?language=objc), which means the user hasn’t yet granted it permission to capture media. The first time you create an [AVCaptureDeviceInput](https://developer.apple.com/documentation/avfoundation/avcapturedeviceinput?language=objc) object for a media type that requires permission, the system automatically displays an alert to request recording permission. Alternatively, call this method to prompt the user at a time of your choosing. The system saves the user’s selection so that it doesn’t have to prompt the user again. A user can change their authorization status in the Settings app.

https://developer.apple.com/documentation/avfoundation/avcapturedevice/1624584-requestaccess?language=swift

mrousavy, 2024-07-12:
Ahh, thanks this makes sense.

I created #3072 as a workaround.

I really need a better way to get sensor orientation...

closed 2024-07-12 by mrousavy