margelo/react-native-vision-camera #3081: application crushes during resize attemp🐛

BarakBon, 2024-07-15:
### What's happening?

i want to use vision camera on my device for future uses with Tflite model.
I need to resize the image data for this model, and I tried to use the resize plugin.
without the execution of the resizing, the app runs just fine. but when I try resizing the app crushes after 1 second the resizing is attempting to execute.
I'm running the application on my device through Expo eas development server

### Reproduceable Code

```tsx
function MainMenu({ navigation }) {
	const [isStartDriving, setStartDriving] = useState(false);
	const [permission, requestPermission] = useCameraPermissions();
	const device = useCameraDevice("front");
	const format = useCameraFormat(device, [
		{ videoResolution: { width: 720, height: 720 } },
		{ fps: 2 },
	]);
	const { resize } = useResizePlugin();

	// Ensure frameProcessor is defined only when needed
	const frameProcessor = useFrameProcessor((frame) => {
		"worklet";
		// const buffer = frame.toArrayBuffer();
		// const data = new Uint8Array(buffer);
		// console.log(`Pixel at 0,0: RGB(${data[0]}, ${data[1]}, ${data[2]})`);
		// console.log(buf);
		runAtTargetFps(2, () => {
			"worklet";
			console.log(`Received a ${frame.width} x ${frame.height} Frame!`);
			const resized = resize(frame, {
				scale: {
					width: 192,
					height: 192,
				},
				pixelFormat: "rgb",
				dataType: "float32",
			});
			console.log(`Resized to ${resized.width} x ${resized.height} Frame!`);
		});
	}, []);
return (
		<View style={styles.mainContainer}>
<MainButton
					color={Colors.letsDriveColor}
					icon={<Fontisto name="angle-dobule-right" size={28} color="white" />}
					onPress={letsDriveHandler}>
					Lets Drive!
				</MainButton>
			<Camera
				device={device}
				frameProcessor={frameProcessor}
				isActive={isStartDriving}
				pixelFormat="rgb"
				format={format}
				resizeMode="contain"
				preview={false}
			/>
		</View>
	);
}
```

### Relevant log output

```shell
07-12 00:01:53.359  1012  1091 E AndroidRuntime: FATAL EXCEPTION: AsyncTask #1
07-12 00:01:53.359  1012  1091 E AndroidRuntime: Process: com.oplus.uxdesign, PID: 1012
07-12 00:01:53.359  1012  1091 E AndroidRuntime: java.lang.RuntimeException: An error occurred while executing doInBackground()
07-12 00:01:53.359  1012  1091 E AndroidRuntime: 	at android.os.AsyncTask$4.done(AsyncTask.java:415)
[... 309 lines cut ...]
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

one plus 7 pro

### VisionCamera Version

4.4.1

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Marius-Adam, 2024-07-15:
`FATAL EXCEPTION: AsyncTask #1`

Issue is probably related to the 

Have you tried wrapping your function in `runOnJs`?
https://docs.swmansion.com/react-native-reanimated/docs/2.x/api/miscellaneous/runOnJS/

BarakBon, 2024-07-15:
> `FATAL EXCEPTION: AsyncTask #1`
> 
> Have you tried wrapping your function in `runOnJs` inside the `useFrameProcessor` hook? https://docs.swmansion.com/react-native-reanimated/docs/2.x/api/miscellaneous/runOnJS/

thank you for your answer.
actually, I didn't.
tryed to understand the way you suggest me to use it and ended up with this:

`
	const handleResize = (frame) => {
		const resized = resize(frame, {
			scale: {
				width: 800,
				height: 800,
			},
			pixelFormat: "rgb",
			dataType: "float32",
		});
		console.log(`Resized to ${resized.width} x ${resized.height} Frame!`);
	};
	// Ensure frameProcessor is defined only when needed
	const frameProcessor = useFrameProcessor((frame) => {
		"worklet";
		runAtTargetFps(2, () => {
			console.log(`Received a ${frame.width} x ${frame.height} Frame!`);
			runOnJS(handleResize)(frame);
		});
	}, []);
	
`

now i get this error:
Frame Processor Error: Property '_WORKLET' doesn't exist, js engine: VisionCamera
probably using it wrong.

Marius-Adam, 2024-07-15:
Try this - https://github.com/mrousavy/react-native-vision-camera/issues/1613#issuecomment-2192605759

BarakBon, 2024-07-16:
> Try this (last comment) - [#1613 (comment)](https://github.com/mrousavy/react-native-vision-camera/issues/1613#issuecomment-2192605759)

thank you for trying to help, I appreciate it.

the `createRunInJsFn` is deprecated as I saw, so I used `createRunOnJS` instead like this:

```
const handleResize = (frame) => {
	console.log(`Received a ${frame.width} x ${frame.height} Frame!`);
	const resized = resize(frame, {
		scale: {
			width: 800,
			height: 800,
		},
		pixelFormat: "rgb",
		dataType: "float32",
	});
	console.log(`Resized to ${resized.width} x ${resized.height} Frame!`);
};
const resizer = Worklets.createRunOnJS(handleResize);
const frameProcessor = useFrameProcessor((frame) => {
	"worklet";
	runAtTargetFps(2, () => {
		"worklet";

		// console.log(`Received a ${frame.width} x ${frame.height} Frame!`);
		resizer(frame);
	});
}, []);
```

still the same as the edit in my previous comment. a large gap between "Received.." prints and not any "Resized.." ones

BarakBon, 2024-07-16:
> > ```
> > java.lang.String com.oplus.uxdesign.language.bean.FileHostBean.getManual()
> > ```
> 
> this doesn't really look related to VisionCamera. Does everything run fine if you just remove the `resize(..)` call?

thank you for your response.

its not crushing if I comment the resize call.

mrousavy, 2024-07-16:
Can you please create an issue in the resize plugin then? If the plugin crashes, this is not a VisionCamera issue but rather an issue with the plugin you are calling.

closed 2024-07-16 by mrousavy