margelo/react-native-vision-camera #3083: 💭 How to get a 9/16 aspect ratio for photos and videos?

VictorioMolina, 2024-07-16:
### Question

I want to record videos and take photos on Instagram Stories aspect ratio.

I have tried with the Templates, but as a result I get horizontal photos.

How could we achieve this? I couldn't find any example where height > width in the documentation for camera format.

### What I tried

```
const format = useCameraFormat(device, [
    { videoResolution: "max" },
    { videoAspectRatio: 9 / 16 },
    { photoResolution: "max" },
    { photoAspectRatio: 9 / 16 },
  ]);
  ```
  
  But the resulting aspect ratio, in the format is 4032/3024:
  
  ```
  {
  "minFps": 1,
  "videoWidth": 4032, <-----------------
  "minISO": 32,
  "photoHeight": 3024, <-----------------
  "maxFps": 30,
  "fieldOfView": 102.90327453613281,
  "videoHeight": 3024, <-----------------
  "autoFocusSystem": "phase-detection",
  "supportsVideoHdr": false,
  "supportsPhotoHdr": false,
  "photoWidth": 4032, <-----------------
  "supportsDepthCapture": false,
  "maxISO": 3072,
  "pixelFormats": [
    "yuv",
    "yuv",
    "rgb"
  ],
  "maxZoom": 189,
  "videoStabilizationModes": [
    "auto",
    "off"
  ]
}
  ```

### VisionCamera Version

^3.9.2

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

VictorioMolina, 2024-07-16:
Also tried with last version 4.5.0 and outputOrientation = “device” and “preview”.

The photo orientation is always landscape-right on my iPhone 11 Pro

VictorioMolina, 2024-07-16:
[image]
[image]

mrousavy, 2024-07-16:
Don't use V3, use V4. 

Also, formats are always in horizontal (or device native) orientation, this is mentioned in the Format docs.

Please read the Format documentation - there is also a format selector for aspect ratio

closed 2024-07-16 by mrousavy

VictorioMolina, 2024-07-16:
@mrousavy Hey, thank you for your answer, I really appreciate it and the work you are doing with this library.

I read documentation some months ago, while implementing the camera system. Have checked the formats docs again, and yes, it is working as expected.

The error I was having was related to the photo.width always being bigger than photo.height (after mapping my formats, I detected there were no vertical snaps).

_1. When rendering with react native image, with absolute fill, the image is correctly visualized.
2. When saving in the camera roll (I suppose because of Exif metadata), it is rendered correctly in the device´s native gallery._
**3. When rendering the image without absolute fill, it seems we need to switch the properties when orientation !== portrait/reverse portrait??**

For example, here, using Skia Images:

```js
    const renderCanvas = () => (
      <SkiaCanvas ref={ref} style={[globalStyles.flexCenteredContainer, style]}>
        <Group transform={transformation}>
          <Group matrix={matrix}>
            <Image
              image={image}
              width={width} <---------- 4224px
              height={height} <----------  2376px (the image orientation is not correct, is it safe to switch the properties?).
              fit="cover"
            >
              {filter && <ColorMatrix matrix={filter.data} />}
            </Image>
          </Group>
        </Group>
      </SkiaCanvas>
    );
```

When I say switching the properties, I mean something like:

```
image = {
  ...image,
  uri: image.path,
  width: image.height,
  height: image.width,
};
```

Aspect ratio is good. Problem is with orientation.

mrousavy, 2024-07-17:
> The error I was having was related to the photo.width always being bigger than photo.height (after mapping my formats, I detected there were no vertical ones).

Exactly - this is because Camera sensors are always in a fixed native orientation, which is landscape in most of the cases.

In VisionCamera, I use transform matrixes, rotations and EXIF tags to rotate the images to the desired output rotation (portrait in your case).

This is expected.

mrousavy, 2024-07-17:
Wait sorry I thought you were referring to `format.photoWidth`, but you actually said `photo.width`?

Let me try this to see if I can reproduce this.

mrousavy, 2024-07-17:
To be honest I would probably rely on `<Image>` or some other component that actually loads the image to give you `width`/`height` properties - those are already accounting for pixel ratio and orientation.
VisionCamera gives you the true native dimensions, while Image loaders give you the dimensions after loading the image.

VictorioMolina, 2024-07-17:
> To be honest I would probably rely on <Image> or some other component that actually loads the image to give you width/height properties - those are already accounting for pixel ratio and orientation.

My use case requires Skia image and canvas, with explicit width and height animated properties 😥. For now, as a workaround, I am saving to camera roll and swapping the photo's file dimensions with the ones included in the camera roll asset, ignoring Exif metadata to prevent incorrect behaviour because of that swap.

iliapnmrv, 2024-10-25:
@VictorioMolina Hey! I'm having the same issue, could you please share this part of your code if it's possible?

> for now, as a workaround, I am saving to camera roll and swapping the photo's file dimensions with the ones included in the camera roll asset, ignoring Exif metadata to prevent incorrect behaviour because of that swap.

iliapnmrv, 2025-03-20:
@Sof2222 Hi! I was unable to make it work and unfortunately I had to use screen's aspect ratio. Like this

```tsx
const ScreenAspectRatio =
  Dimensions.get('screen').height / Dimensions.get('screen').width;

 const format = useCameraFormat(device, [
    {
      photoAspectRatio: ScreenAspectRatio,
    },
    {
      videoAspectRatio: ScreenAspectRatio,
    },
  ]);
```