margelo/react-native-vision-camera #3084: 🔧  Android Build Error Imported target "react-native-worklets-core::rnworklets" includes non-existent path

bfar97, 2024-07-16:
### How were you trying to build the app?

I created an app from zero using react-native-community/reproducer-react-native [here](https://github.com/bfar97/vision-camera-worklets-core-build-fail) and only installed react-native-vision-camera@4.5.0 and react-native-worklets-core@1.3.3. When building for android I get some errors in CMakeLists.txt ```Imported target "react-native-worklets-core::rnworklets" includes
non-existent path```

### Full build logs

```tsx
> Task :prepareKotlinBuildScriptModel UP-TO-DATE
Download https://dl.google.com/dl/android/maven2/androidx/core/core-ktx/1.2.0/core-ktx-1.2.0.pom, took 357 ms
Download https://dl.google.com/dl/android/maven2/androidx/arch/core/core-runtime/2.0.0/core-runtime-2.0.0.aar, took 165 ms
Download https://dl.google.com/dl/android/maven2/androidx/core/core-ktx/1.2.0/core-ktx-1.2.0.aar, took 181 ms
[CXX5304] This version only understands SDK XML versions up to 3 but an SDK XML file of version 4 was encountered. This can happen if you use versions of Android Studio and the command-line tools that were released at different times.
C/C++: VisionCamera: Frame Processors: ON!
C/C++: VisionCamera: Linking react-native-worklets...
C/C++: CMake Error in CMakeLists.txt:
C/C++:   Imported target "react-native-worklets-core::rnworklets" includes
C/C++:   non-existent path
C/C++:     "/Users/bernardo/Desktop/vision-camera-worklets-core-build-fail/ReproducerApp/node_modules/react-native-worklets-core/android/build/headers/rnworklets"
C/C++:   in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
C/C++:   * The path was deleted, renamed, or moved to another location.
C/C++:   * An install or uninstall procedure did not complete successfully.
C/C++:   * The installation package was faulty and references files it does not
C/C++:   provide.
C/C++: CMake Generate step failed.  Build files cannot be regenerated correctly.
[CXX1429] error when building with cmake using /Users/bernardo/Desktop/vision-camera-worklets-core-build-fail/ReproducerApp/node_modules/react-native-vision-camera/android/CMakeLists.txt: -- The C compiler identification is Clang 17.0.2
-- The CXX compiler identification is Clang 17.0.2
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/bernardo/Library/Android/sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/bernardo/Library/Android/sdk/ndk/26.1.10909125/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done

C++ build system [configure] failed while executing:
    /Users/bernardo/Library/Android/sdk/cmake/3.22.1/bin/cmake \
      -H/Users/bernardo/Desktop/vision-camera-worklets-core-build-fail/ReproducerApp/node_modules/react-native-vision-camera/android \
      -DCMAKE_SYSTEM_NAME=Android \
      -DCMAKE_EXPORT_COMPILE_COMMANDS=ON \
      -DCMAKE_SYSTEM_VERSION=23 \
      -DANDROID_PLATFORM=android-23 \
      -DANDROID_ABI=x86 \
      -DCMAKE_ANDROID_ARCH_ABI=x86 \
      -DANDROID_NDK=/Users/bernardo/Library/Android/sdk/ndk/26.1.10909125 \
      -DCMAKE_ANDROID_NDK=/Users/bernardo/Library/Android/sdk/ndk/26.1.10909125 \
      -DCMAKE_TOOLCHAIN_FILE=/Users/bernardo/Library/Android/sdk/ndk/26.1.10909125/build/cmake/android.toolchain.cmake \
      -DCMAKE_MAKE_PROGRAM=/Users/bernardo/Library/Android/sdk/cmake/3.22.1/bin/ninja \
      "-DCMAKE_CXX_FLAGS=-O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all" \
      -DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/Users/bernardo/Desktop/vision-camera-worklets-core-build-fail/ReproducerApp/node_modules/react-native-vision-camera/android/build/intermediates/cxx/Debug/4b5h1x1z/obj/x86 \
      -DCMAKE_RUNTIME_OUTPUT_DIRECTORY=/Users/bernardo/Desktop/vision-camera-worklets-core-build-fail/ReproducerApp/node_modules/react-native-vision-camera/android/build/intermediates/cxx/Debug/4b5h1x1z/obj/x86 \
      -DCMAKE_BUILD_TYPE=Debug \
      -DCMAKE_FIND_ROOT_PATH=/Users/bernardo/Desktop/vision-camera-worklets-core-build-fail/ReproducerApp/node_modules/react-native-vision-camera/android/.cxx/Debug/4b5h1x1z/prefab/x86/prefab \
      -B/Users/bernardo/Desktop/vision-camera-worklets-core-build-fail/ReproducerApp/node_modules/react-native-vision-camera/android/.cxx/Debug/4b5h1x1z/x86 \
      -GNinja \
      -DANDROID_STL=c++_shared \
      -DNODE_MODULES_DIR=/Users/bernardo/Desktop/vision-camera-worklets-core-build-fail/ReproducerApp/node_modules \
      -DENABLE_FRAME_PROCESSORS=ON
  from /Users/bernardo/Desktop/vision-camera-worklets-core-build-fail/ReproducerApp/node_modules/react-native-vision-camera/android
VisionCamera: Frame Processors: ON!
VisionCamera: Linking react-native-worklets...
CMake Error in CMakeLists.txt:
  Imported target "react-native-worklets-core::rnworklets" includes
  non-existent path

    "/Users/bernardo/Desktop/vision-camera-worklets-core-build-fail/ReproducerApp/node_modules/react-native-worklets-core/android/build/headers/rnworklets"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
```

### Project dependencies

```json
"dependencies": {
    "react": "18.2.0",
    "react-native": "0.74.3",
    "react-native-vision-camera": "^4.5.0",
    "react-native-worklets-core": "^1.3.3"
  },
```

### VisionCamera Version

4.5.0

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-07-16:
Hey - sorry but this is a duplicate issue, a lot of people encounter this error. Long story short; it's a react-native-worklets-core error (not a react-native-vision-camera error), and it is because of Gradle/CMake cache. I didn't have the time to investigate this yet, but if you find a fix for this please let me know & submit a PR to fix this once and for all.

Probably caused by this: https://github.com/margelo/react-native-worklets-core/blob/6e5cc18b693af64516cb2655f28b05d8e072d02b/android/build.gradle#L173-L180

mrousavy, 2024-07-16:
The current workaround is to just nuke gradle/cmake cache and rebuild.

closed 2024-07-16 by mrousavy

bfar97, 2024-07-16:
I have found similar issues to this one, but not quite exactly the same. Also, nuking caches is not allowing me to rebuild successfully, that's why I fired the issue. Will try to look into the issue and see if I can come up with a fix! Thanks for the quick reply

maxpowa, 2024-12-14:
I've run into this while doing a "Gradle Sync" within Android Studio, but I was able to just run a build using `npx react-native run-android` or `npx expo run:android` and it will generate the files needed. I haven't dug deep enough into Gradle to see if there's a way you could hook into the dependency-resolution step instead of just the generic "preBuild" (which isn't run during a "Gradle Sync").