margelo/react-native-vision-camera #3085: 🐛 Android Crash on video recording

amosaxe, 2024-07-16:
### What's happening?

I wanted to take a video, but it crashes after video is recorded and previews start. It crashes anywhere during the preview screen or gets stuck. I have tried on two devices, oneplus 6T and motorolla g60. This only happens with back camera.

### Reproduceable Code

```tsx
import { useFocusEffect, useNavigation } from '@react-navigation/native';
import React, { useCallback, useEffect, useRef, useState } from 'react';
import {
    ActivityIndicator,
    Dimensions,
    Image,
    Platform,
    StyleSheet,
    TouchableOpacity,
    View,
} from 'react-native';
import Video from 'react-native-video';
import {
    Camera,
    useCameraDevice,
    useCameraFormat,
    useCameraPermission,
    useMicrophonePermission,
} from 'react-native-vision-camera';
import { FontAwesomeIcon } from '@fortawesome/react-native-fontawesome';
import {
    faCameraRotate,
    faMoon,
    faBolt,
    faChevronLeft,
} from '@fortawesome/free-solid-svg-icons';
import { PressableOpacity } from 'react-native-pressable-opacity';
import StaticSafeAreaInsets from 'react-native-static-safe-area-insets';
import { CaptureButton } from './CaptureButton';
import Reanimated, {
    Extrapolate,
    interpolate,
    useAnimatedGestureHandler,
    useAnimatedProps,
    useSharedValue,
} from 'react-native-reanimated';
import {
    PinchGestureHandler,
    TapGestureHandler,
} from 'react-native-gesture-handler';
import { dynamicSize } from '../../utils/responsive';
import {
    getMediaDataObjectFromCropper,
    getMediaDataObjectFromVisionCamera,
} from '../../utils/mediaFormatter';
import { CheckIcon } from 'react-native-heroicons/solid';
import { colors } from '../../styles/colors';
import ImagePicker from 'react-native-image-crop-picker';

const CONTROL_BUTTON_SIZE = 40;
const CONTENT_SPACING = 15;
const MAX_ZOOM_FACTOR = 10;
const SCALE_FULL_ZOOM = 3;
const CAPTURE_BUTTON_SIZE = 78;
const BORDER_WIDTH = CAPTURE_BUTTON_SIZE * 0.1;

export const SCREEN_WIDTH = Dimensions.get('window').width;
export const SCREEN_HEIGHT = Platform.select({
    android:
        Dimensions.get('screen').height -
        StaticSafeAreaInsets.safeAreaInsetsBottom,
    ios: Dimensions.get('window').height,
});
[... 923 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

motorolla g60

### VisionCamera Version

4.3.2

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

amosaxe, 2024-07-17:
I even tried the exact same code of example app by migrating(copy pasting) on my app and it still crashes, but if I run the example app it works fine.

I have only remove the frameprocessor while migrating the example app code.

I also looked into android studio logs(logcat). The above logs i had attached from flipper crash reporter

in the logcat i feel this is the main issue:

`Unable to update properties for view tag 65 com.facebook.react.uimanager.IllegalViewOperationException: ViewManager for tag 65 could not be found.`

this error I saw that is an open issue and was marked as close as well after v4. Any suggestions what to do?

amosaxe, 2024-07-17:
update: finally it works, to have a complete copy I even added react-native-worklets-core and a frame processor and now it works, not sure what was the reason, still eager to know so keeping this open

Dingenis, 2024-07-22:
You are running out of memory, that is what the log is trying to tell you. So it might be that you have a memory leak somewhere, you could try to profile it using the [Android Studio Memory Profiler](https://developer.android.com/studio/profile/memory-profiler). I don't think it's related to this library, otherwise the example app would also crash for that very reason. Maybe you could try to find the source of the memory usage?

amosaxe, 2024-07-23:
Yep, I honestly dont blame the library, and will put more details after profiling. Will test once if example also crash if i remove the worklets and update here.

mrousavy, 2024-07-23:
Hey - to be honest this does not sound like a VisionCamera issue, if it works fine in Example but crashes with your code (and the Camera props remain mostly the same) it is very likely something else.

Maybe some other library is allocating a ton of stuff, maybe you have an endless loop allocating JSI Values - I don't know but it doesn't sound VisionCamera related.

mrousavy, 2024-07-23:
Let me know what you find out after profiling - worst case we can just reopen this issue :)

closed 2024-07-23 by mrousavy