margelo/react-native-vision-camera #3086: 💭 Async Frame Processors with Event Emitters

Nithin-Kamineni, 2024-07-17:
### Question

You might also run some very complex AI algorithms which are not fast enough to smoothly run at 30 FPS (33ms). To not drop any frames you can create a custom "frame queue" which processes the copied frames and calls back into JS via a React event emitter. For this you'll have to create a Native Module that handles the asynchronous native -> JS communication, see ["Sending events to JavaScript" (Android)](https://reactnative.dev/docs/native-modules-android#sending-events-to-javascript) and ["Sending events to JavaScript" (iOS)](https://reactnative.dev/docs/native-modules-ios#sending-events-to-javascript).
```
function App() {
  const frameProcessor = useFrameProcessor((frame) => {
    'worklet'
    SomeAI.process(frame) // does not block frame processor, runs async
  }, [])

  useEffect(() => {
    SomeAI.addListener((results) => {
      // gets called asynchronously, goes through the React Event Emitter system
      console.log(`AI results: ${results}`)
    })
  }, [])

  return (
    <Camera frameProcessor={frameProcessor} {...cameraProps} />
  )
}
```
This is the documentation for using the Even Emitters to process the frames that take too long to process. And this is the exact solution I am looking for. But I couldn't understand the documentation of running this plugin so that it does not block the frame processor and runs async. Does the line `SomeAI.process(frame)` in the documentation refer to calling a Turbo module or is this a frameprocessor plugin? 
1. If this is a frame processor plugin then how can I use an event emitter in it? 
2. If this is a Turbo module then how can I input the frame object(Since turbo modules only take some data types)

An example code for using the above code would be great.

If this is the Kotlin file I want to send the event emitter back to js how can I do it?
```
package com.visiondropapp

import android.util.Log
import com.mrousavy.camera.frameprocessors.Frame
import com.mrousavy.camera.frameprocessors.FrameProcessorPlugin
import com.mrousavy.camera.frameprocessors.VisionCameraProxy

import android.graphics.Bitmap
import java.nio.ByteBuffer

import org.opencv.android.Utils
import org.opencv.core.*
import org.opencv.core.Mat
import org.opencv.core.Point
import org.opencv.core.CvType
import org.opencv.core.Size

import org.opencv.core.MatOfPoint
import org.opencv.core.MatOfPoint2f
import org.opencv.imgproc.Imgproc

import android.media.Image
import kotlin.math.pow
import kotlin.math.sqrt
import kotlin.math.abs

import com.facebook.react.bridge.ReadableArray
import com.facebook.react.bridge.ReadableMap
import com.facebook.react.bridge.Arguments
import com.facebook.react.bridge.WritableArray
import com.facebook.react.bridge.WritableMap

import android.util.Base64
import java.io.ByteArrayOutputStream

import com.facebook.react.modules.core.DeviceEventManagerModule
import com.facebook.react.bridge.ReactApplicationContext

class MicrochipAdjustmentProcessFrameProcessorPlugin(proxy: VisionCameraProxy, options: Map<String, Any>?): FrameProcessorPlugin() {
    
    private var mEmitter: DeviceEventManagerModule.RCTDeviceEventEmitter? = null

    init {
        Log.d("MicrochipAdjustmentProcessPlugin", "MicrochipAdjustmentProcessFrameProcessorPlugin initialized with options: " + options?.toString())
    }

    private fun sendEvent(eventName: String, params: WritableMap?) {
        if (mEmitter == null) {
            mEmitter = reactContext.getJSModule(DeviceEventManagerModule.RCTDeviceEventEmitter::class.java)
        }
        mEmitter?.emit(eventName, params)
    }

    private fun convertImageToMat(image: Image): Mat {
        val planes = image.planes
        val yBuffer: ByteBuffer = planes[0].buffer
        val uBuffer: ByteBuffer = planes[1].buffer
        val vBuffer: ByteBuffer = planes[2].buffer

        val ySize = yBuffer.remaining()
        val uSize = uBuffer.remaining()
        val vSize = vBuffer.remaining()

        val nv21 = ByteArray(ySize + uSize + vSize)

        yBuffer.get(nv21, 0, ySize)
        vBuffer.get(nv21, ySize, vSize)
        uBuffer.get(nv21, ySize + vSize, uSize)

        // val yuvImage = Mat(Size(image.width.toDouble(), image.height.toDouble()), CvType.CV_8UC1)
        // yuvImage.put(0, 0, nv21)

        // val rgbImage = Mat()
        // Imgproc.cvtColor(yuvImage, rgbImage, Imgproc.COLOR_YUV2RGB_NV21, 3)

        val yuvImage = Mat(image.height + image.height / 2, image.width, CvType.CV_8UC1)
        yuvImage.put(0, 0, nv21)

        val rgbImage = Mat()
        Imgproc.cvtColor(yuvImage, rgbImage, Imgproc.COLOR_YUV2RGB_NV21)

        yuvImage.release()

        return rgbImage
    }

    private fun matToBase64String(mat: Mat): String {
        val bitmap = Bitmap.createBitmap(mat.cols(), mat.rows(), Bitmap.Config.ARGB_8888)
        Utils.matToBitmap(mat, bitmap)
        val byteArrayOutputStream = ByteArrayOutputStream()
        bitmap.compress(Bitmap.CompressFormat.JPEG, 100, byteArrayOutputStream)
        val byteArray = byteArrayOutputStream.toByteArray()
        return Base64.encodeToString(byteArray, Base64.DEFAULT)
    }

    override fun callback(frame: Frame, params: Map<String, Any>?): Any? {
        if (params == null) {
            return null
        }

        val image = frame.image
        Log.d(
            "MicrochipAdjustmentPlugin",
            image.width.toString() + " x " + image.height + " Image with format #" + image.format + ". Logging " + params.size + " parameters:"
        )

        for (key in params.keys) {
            val value = params[key]
            Log.d("MicrochipAdjustmentPlugin", "  -> " + if (value == null) "(null)" else value.toString() + " (" + value.javaClass.name + ")")
        }

        val debug = params["debug"]
        // val debug = (params["debug"] as? String)?.toString() ?: "debug"

        // Convert the image to a Mat
        val originalMat = convertImageToMat(image)

        // Get the dimensions of the Mat
        val width = originalMat.width()
        val height = originalMat.height()

        Log.d("MicrochipAdjustmentPlugin", "Image dimensions: width = $width, height = $height")

        val params2 = Arguments.createMap().apply {
            putString("message", "params in adjustment process")
        }
        sendEvent("EventReminder1", params2)

        return hashMapOf<String, Any>(
            "width" to width,
            "height" to height,
            // "maxCircle" to (maxCircle ?: emptyMap<String, Any?>())
            // "squares" to mutableListOf<Map<String, Any?>>(squares)
        )
        
    }
}
```

An open-source library like this is invaluable for solo developers, enabling them to work on their projects. Thanks to everyone contributing to the development of the Vision camera!

### What I tried

I tried inputting the frame into a turbo module(since I can use even an emitter in that module). However, the turbo module is not accepting the frame data type. Also, I tried getting an emitter function to send events from the frame processor plugin to even the listener, but that didn't work.

### VisionCamera Version

4.4.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

Visiondrop, 2024-07-18:
Our company requires this feature of async frame processors with event emitters. @mrousavy Can you provide an example project to solve this issue?

mrousavy, 2024-07-19:
> Does the line SomeAI.process(frame) in the documentation refer to calling a Turbo module or is this a frameprocessor plugin?

This is a Frame Processor. Or a pure `jsi::HostFunction` if you are in C++/JSI code.

> If this is a frame processor plugin then how can I use an event emitter in it?

I haven't built something like this before, but I'm guessing you would need a separate TurboModule that has an event emitter, and then push events to that TurboModule on the native side somehow.

I am working on a new React Native Framework which will make this a lot easier - then even Frame Processor Plugins can directly push events easily.

As of today, yes you kinda have to go through a Native-, Turbo- or Expo-Module.

closed 2024-07-19 by mrousavy

Visiondrop, 2024-07-19:
Is there a way to input the frame as the parameter into the turbo modules? I can't seem to figure out a way. I do not prefer to convert the image into base64 to send it to the turbo module(it would be the last option). The frame processing is running fast so I don't want to change the frame to another data type is there a way to do it? If not then what do you recommend, what data type I can convert the image(from frame) so that I can send the image to the turbo module?

Visiondrop, 2024-07-19:
> This is a Frame Processor. Or a pure `jsi::HostFunction` if you are in C++/JSI code.
>
>
> I haven't built something like this before, but I'm guessing you would need a separate TurboModule that has an [...]

Is there a way to input the frame as the parameter into the turbo modules? I can't seem to figure out a way. I do not prefer to convert the image into base64 to send it to the turbo module(it would be the last option). The frame processing is running fast so I don't want to change the frame to another data type is there a way to do it? If not then what do you recommend, what data type I can convert the image(from frame) so that I can send the image to the turbo module?

mrousavy, 2024-07-21:
> Is there a way to input the frame as the parameter into the turbo modules?

No, because Java/ObjC TurboModules aren't flexible enough yet. The React Native Framework I am working on will make this possible.

But I am not saying you should pass the Frame to a TurboModule, I'm saying you should pass the Frame to a Frame Processor, and in the native code you access the TurboModule and send an event somehow. It worked on the old arch, I have not tried it on new arch yet.