margelo/react-native-vision-camera #3087: External Storage Permission denied due to latest policy.

sandeep36butte, 2024-07-17:
### Question

Due to the latest google play policy the use of External Storage permission is restricted for financial apps like loan etc..
Because of which our app update is getting rejected.
Is there any workaround where we can use the clicked photo from vision camera on RN view without storing the temp file in external storage and by pass the storage permission.

### VisionCamera Version

3.7.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-07-23:
I don't think VisionCamera forces external storage.

I store my files in a `File.createTempFile(..., context.cacheDir)`: https://github.com/mrousavy/react-native-vision-camera/blob/9a0620427b81e513665a2034db8742de64a115d1/package/android/src/main/java/com/mrousavy/camera/core/utils/FileUtils.kt#L12-L15

How is this not following the Google Play policy?

mrousavy, 2024-07-23:
According to [Privacy policy issue MANAGE_EXTERNAL_STORAGE](https://support.google.com/googleplay/android-developer/thread/243412077/privacy-policy-issue-manage-external-storage?hl=en#:~:text=Apps%20should%20only%20request%20access,critical%20user%2Dfacing%20app%20functionality.) it sounds fine as this is only related to this one specific permission - I couldn't find any info about the new policy you are mentioning - can you please link that here?

sandeep36butte, 2024-07-23:
Hello @mrousavy 
My bad our team is using the external storage permission to get the temp file created by vision camera which is not needed as you said earlier.

closed 2024-07-23 by sandeep36butte