margelo/react-native-vision-camera #3093: Is there a way to get the focal length of the camera?

seadeep42, 2024-07-18:
### Question

I'm using the front camera with a frame processor. Is there a way to get the focal length of the image either as a camera parameter or a property of the frame?

### VisionCamera Version

4.5.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-07-23:
No, there is not at the moment. 

You can create a feature request for this - or even better - a PR to implement this, if you want :)

mrousavy, 2024-07-23:
On iOS we have [`lensPosition`](https://developer.apple.com/documentation/avfoundation/avcapturedevice/1624643-lensposition) - not sure if this is actually what we want:

> A lens position value doesn’t correspond to an exact physical distance, **nor does it represent a consistent focus distance from device to device**.

Why do you need focal length anyways?

mrousavy, 2024-07-23:
I found this code somewhere in a gist:

```swift
func getFocalLengthInMillimeters(captureDevice: AVCaptureDevice) -> Float? {
    guard let format = captureDevice.activeFormat else {
        return nil
    }
    
    // Get the intrinsic matrix, which provides focal length in pixels
    let dimensions = CMVideoFormatDescriptionGetDimensions(format.formatDescription)
    let sensorWidth = format.sensorDimensions.width
    let sensorHeight = format.sensorDimensions.height
    let intrinsicMatrix = format.formatDescription.intrinsicMatrix
    
    // Calculate the focal length in millimeters
    let focalLengthInPixels = intrinsicMatrix[0][0] // This is the fx value
    let sensorPhysicalWidth = format.sensorPhysicalSize.width
    let sensorPhysicalHeight = format.sensorPhysicalSize.height
    let focalLengthInMillimeters = (focalLengthInPixels / Float(dimensions.width)) * Float(sensorPhysicalWidth)
    
    return focalLengthInMillimeters
}
```

But it looks very hacky and depends on activeFormat, I don't think this can work reliably in `getAvailableCameraDevices()`. 

This _might_ only work if we pass it alongside with `Frame`, but I kinda want to avoid that since the intrinsic matrix and focal length has nothing to do with a Frame, but more with a Camera device - it's contextually in the wrong place.

closed 2024-07-23 by mrousavy

mrousavy, 2024-07-23:
Also maybe related to https://github.com/mrousavy/react-native-vision-camera/issues/3032

waleedshkt, 2024-10-26:
I'm having a similar requirement to get camera focal length and sensory data

I believe in Android, there a native module `android.hardware.camera2.CameraCharacteristics` which offers this data. But I didn't find any such nativ modules for iOS

I think what we also want is the image rotation, yaw, pitch angles to successfully build 3D point cloud images

Not sure how to pull, rebuild it and do PR for this.

sayan-avantai, 2025-04-16:
Any update regarding the focal length?