margelo/react-native-vision-camera #3095: error: package com.mrousavy.camera.frameprocessor does not exist

DanielWoldehana, 2024-07-22:
### How were you trying to build the app?

I tried building my app with "vision-camera-code-scanner": "^0.2.0" While on ReactNative 0.74 and Android 14, sdk 34
 and I keep getting an error

### Full build logs

```tsx
/node_modules/vision-camera-code-scanner/android/src/main/java/com/visioncameracodescanner/VisionCameraCodeScannerPluginPackage.java:9: error: package com.mrousavy.camera.frameprocessor does not exist
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;
```

### Project dependencies

```json
"dependencies": {
   "react": "18.2.0",
   "react-hook-form": "^7.43.9",
   "react-native": "0.74.0",
   "react-native-vision-camera": "^4.3.2",
   "vision-camera-code-scanner": "^0.2.0"
}
```

### VisionCamera Version

^4.3.2

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

DanielWoldehana, 2024-07-23:
@mrousavy I did search those 2 issues in the screenshot above but there was one solution and that was to use java 11 from AdoptOpenJDK and delete zulu 11 but I am on react-native 0.73.1 because Android 14 requires sdk 34 and that is the earliest version of react-native that supports sdk 34 but unfortunately 0.73.1 requires java 17 and currently I am on zulu 17 and getting an error related to the vision-camera-code-scanner package not being able to find the Barcode class from com.google.mlkit.vision.barcode

IgorDePaula, 2024-12-01:
@mrousavy 2 years and not fix?

IgorDePaula, 2024-12-02:
He created this error on purpose to monetize his component. This component is no longer open source. Consider using another module.