margelo/react-native-vision-camera #3108: 💭 USB Camera Android

rorigger, 2024-07-31:
### Question

Hey Guys, I need a solution for using an external USB camera I have seen the issues regarding UVC cameras etc. I'm just wondering if anyone has ever successfully used an external camera with this library and if so what specific android/camera device was used? any help would be greatly appreciated!  

### What I tried

I have tried multiple UVC camera modules and a Microsoft Lifecam with Pixel7 and Galaxy Tab A7 Devices

### VisionCamera Version

4.5.0

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

ppanwin10, 2024-12-24:
Based on the library code, I noticed that the device list is directly polled from the camera manager. This means if the Android phone doesn't detect the external camera, the library won't pick it up. It seems that some devices support this, while others do not, so requiring alternative methods to capture external camera data.

https://stackoverflow.com/questions/57846505/accessing-a-usb-camera-using-android-camera2-api

mrousavy, 2025-02-17:
closing in favor of https://github.com/mrousavy/react-native-vision-camera/issues/3159

closed 2025-02-17 by mrousavy

AssenDimitrov, 2025-03-07:
Turns out most of the Android builds does not have [FEATURE_CAMERA_EXTERNAL](https://developer.android.com/reference/android/content/pm/PackageManager#FEATURE_CAMERA_EXTERNAL) kernel feature enabled. So we have two options - use custom Android ROM which is built with such support (ex. Lineage OS) or make your own UVC integration like [this example](https://voidcomputing.hu/blog/android-uvc/)