margelo/react-native-vision-camera #3111: ✨Support 16kb page size for Android 15+ devices

k-saparia, 2024-07-31:
### What feature or enhancement are you suggesting?

### Description
I am encountering issues when running my app on Android 15 devices with a 16 KB page size, particularly with some native components. Are there any plans to recompile the library to address these issues and ensure compatibility with phones that have a 16 KB page size?

As stated on Android [doc](https://developer.android.com/guide/practices/page-sizes#compile-16-kb-alignment),

> Beginning with Android 15, Android supports devices that are configured to use a page size of 16 KB (16 KB devices). If your app uses any [NDK](https://developer.android.com/ndk) libraries, either directly or indirectly through an SDK, then you will need to rebuild your app for it to work on these 16 KB devices.

An analysis of my app using the APK Analyzer [as described by Android](https://developer.android.com/guide/practices/page-sizes#16-kb-impact) shows that it does use libcapture as native code.
[image]
This is the native binary reported by the APK analysis:

libVisionCamera.so

### Describe the solution you'd like

* I would like for libcapture to support 16kb page size phones so that apps supporting sdk 35 do not crash

### Additional context

* Android doc: https://developer.android.com/guide/practices/page-sizes#compile-16-kb-alignment
* react-native thread: https://github.com/facebook/react-native/issues/45054

### What Platforms whould this feature/enhancement affect?

Android

### Alternatives/Workarounds

No workaround I have investigated so far. 

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2024-08-01:
should be just a simple flag to CMake, but I think this will break backwards compat

k-saparia, 2024-08-01:
If possible, It should be backward compatible with old and new architecture 🤔 Otherwise, worth to have a force update.

mrousavy, 2024-08-01:
I'm talking about older android versions

vishalyad16, 2024-08-04:
@k-saparia
How can I reduce the size of a video that is 151 MB for a 1-minute recording? The bitRate setting does not seem to be working.
https://github.com/mrousavy/react-native-vision-camera/issues/3113

mrousavy, 2025-02-17:
This already works in latest VisionCamera afaik

yousrasd, 2025-05-23:
Is there an update on this, Google is mandating apps to support 16kb page size by November, this is a blocker for us at the moment

bobzurad, 2025-05-29:
I don't understand why this issue was closed. When I add react-native-vision-camera to my React Native app that targets Android 15, I get the same results as @k-saparia, and I see a popup on my app that says: "This app isn't 16 KB compatible. ELF alignment check failed."

I was able to fix this issue by removing react-native-vision-camera and adding this library instead: https://github.com/teslamotors/react-native-camera-kit/ 

I also had to apply the patch in this PR: https://github.com/teslamotors/react-native-camera-kit/pull/699

k-saparia, 2025-05-29:
They have created fix for vision camera 16 KB support https://github.com/mrousavy/react-native-vision-camera/pull/3543

k-saparia, 2025-05-29:
However, I fixed by adding support in their cmake file. Here's my patch work!

```
diff --git a/node_modules/react-native-vision-camera/android/CMakeLists.txt b/node_modules/react-native-vision-camera/android/CMakeLists.txt
index 17097c1..4db203e 100644
--- a/node_modules/react-native-vision-camera/android/CMakeLists.txt
+++ b/node_modules/react-native-vision-camera/android/CMakeLists.txt
@@ -86,3 +86,5 @@ if (ENABLE_FRAME_PROCESSORS)
             react-native-worklets-core::rnworklets
     )
 endif()
+
+target_link_options(${CMAKE_PROJECT_NAME} PRIVATE "-Wl,-z,max-page-size=16384")
```

androaddict, 2025-06-24:
Even I upgraded to latest version of vision camera lib, while releasing the app in play store I face same kind of issue throw by google.
below is error I faced,
dlopen failed empty/missing string DT_STR TAB.
I targeted to 35.

how to fix this latest version won't fix this issues

sLurPPPeRsTaR, 2025-09-10:
> What? The PR you linked was merged on Jun 5, and 4.6.4 was released on Feb 20. It cannot be inside that release. It's inside 4.7.0.
> 
> <img alt="Image" width="753" height="266" src="https://private-user-images.githubusercontent.com/15199031/486732287-c1f4936a-a6ef-4d10-a665-13dddf62769d.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTc0NzQ5MTksIm5iZiI6MTc1NzQ3NDYxOSwicGF0aCI6Ii8xNTE5OTAzMS80ODY3MzIyODctYzFmNDkzNmEtYTZlZi00ZDEwLWE2NjUtMTNkZGRmNjI3NjlkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTEwVDAzMjMzOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAyYjlmMDliNTBlOWRmZTViM2FlZjUzMzAwYzQ5NDE4ZGI3NWRjN2JmYzk5MTU2YzM5M2E3NjYwZjY5NjNjNmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.TLHB4lZOQ5agMOXkgmlZgkag1yv__rZ9wqM-nS5tvCM">

will try, thanks btw