margelo/react-native-vision-camera #3113: 🐛How can I reduce the size of a video that is 151 MB for a 1-minute recording? The bitRate setting does not seem to be working.

vishalyad16, 2024-08-02:
### What's happening?

When recording a 1-minute video, the file size becomes as large as 151 MB. However, using compression packages often results in the video becoming blurry.

### Reproduceable Code

```tsx
import React, { useRef, useState, useEffect, useCallback } from 'react'
import {
    View,
    StyleSheet,
    TouchableOpacity,
    Image,
    AppState,
    BackHandler,
    Alert,
    ToastAndroid
} from 'react-native'
// components
import MyText from '../../components/atoms/MyText';
//packages
import { Camera, useCameraDevice, useCameraFormat, useCameraPermission, useMicrophonePermission } from 'react-native-vision-camera';
import FlipIcon from 'react-native-vector-icons/MaterialIcons'
import { useNavigation } from '@react-navigation/native';
import Icon from 'react-native-vector-icons/Ionicons';
// assets
import colors from '../../assets/colors/colors';
import RecordBtn from '../../assets/images/recordBtn.png';
import PauseBtn from '../../assets/images/PauseBtn.png';
import CustomButton from '../../components/atoms/Button';
import _debounce from 'lodash/debounce';
// style
import styles from './style'

const RecordVideo = ({ route }) => {
    //var
    const { setCustomUrl } = route.params;
    const backCamera = useCameraDevice('back');
    const frontCamera = useCameraDevice('front');

    const format = useCameraFormat(backCamera, [
        { videoResolution: { width: 1280, height: 720 } },
        { fps: 15 },{pixelFormat: 'native'}
        ])
        console.log("format",format);
    //hook:useRef
    const cameraRef = useRef(Camera);
    //hook:navigation
    const navigation = useNavigation()
    // Camera and microphone permissions
    const { hasPermission, requestPermission } = useCameraPermission()
    const { hasPermission: hasMicPermission, requestPermission: requestMicPermission } = useMicrophonePermission()
    useEffect(() => {
        // requestCameraPermission()
        if (Platform.OS === 'ios') {
        hasMicPermission || requestMicPermission()
        }
    }, [])
    //hook:useState
    const [isRecording, setIsRecording] = useState(false)
    const [recordDuration, setRecordDuration] = useState(0)
    const [timerId, setTimerId] = useState(null);
    const [isRecordingPaused, setIsRecordingPaused] = useState(false);
    const [isRecordingCompleted, setisRecordingCompleted] = useState(false);
    const [videoPath, setVideoPath] = useState("");
    const [cameraActive, setCameraActive] = useState(true);
    const [selectedDevice, setSelectedDevice] = useState(backCamera); // State to keep track of selected camera device
    const [isMaxDurationReached, setIsMaxDurationReached] = useState(false);
    //Function:To get camera permission
    // const requestCameraPermission = async () => {
[... 327 lines cut ...]
    "fieldOfView": 84.9812741473671,
    "maxFps": 30,
    "photoHeight": 1800
}
```

### Device

Realme ,OnePlus

### VisionCamera Version

"react-native-vision-camera": "^4.4.2",

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

alextusinean, 2024-08-04:
This is not an issue with the library. 151mb isn't that odd for a one minute video at 1280x720 15fps.

iamvishaldev, 2024-08-04:
> This is not an issue with the library. 151mb isn't that odd for a one minute video at 1280x720 15fps. If you'd like to reduce the size even more, start working on video compression algorithms.

Can you send me some examples for compression algorithms and I don't want to record at 1280×720 it can record 480 also

alextusinean, 2024-08-04:
> Can you send me some examples for compression algorithms and I don't want to record at 1280×720 it can record 480 also

I would like to but I don't know about any better than mp4. Try recording at 480 then and see if it improves.

zzz08900, 2024-08-25:
One thing with video compression algorithm is:
Fast encoding + high quality = large file size
Small file size + high quality = slow encoding
Fast encoding + small file size = lousy quality

take your pick then.

zzz08900, 2024-08-25:
If you think you can outsmart [VCEG](https://en.wikipedia.org/wiki/VCEG)) with a few days of work and came with a much better video compression algorithm than H264/265, well, good luck then.

dontuse, 2024-09-30:
[image]

In the previous version 3, my files were 34 megabytes, but in the new version, they have doubled to 64.

`  const format = useCameraFormat(device, [
    { videoAspectRatio: 16 / 9 },
    { videoResolution: { width: 1280, height: 720 } },
    { fps: 25 },
    { videoHdr: false },
  ]);`
  
  ` cameraRef.current.startRecording({
          flash: 'off',
          fileType: 'mp4',
          videoCodec: 'h264',
          videoBitRate: 'extra-low',`
          
 ` <Camera
            ref={cameraRef}
            style={StyleSheet.absoluteFill}
            device={device}
            format={format}
            isActive={true}
            video={true}
            audio={true}
            photo={false}
            fps={25}
            videoStabilizationMode="auto"
            resizeMode={'contain'}
          />`

closed 2024-10-30 by mrousavy

brianw-tech, 2025-04-15:
> [ { videoAspectRatio: 16 / 9 }, { videoResolution: { width: 1280, height: 720 } }, { fps: 25 }, { videoHdr: false }, ]

did you get a solution to this?