margelo/react-native-vision-camera #3115: 🐛 onRecordingFinished returns width: 0, height: 0 on every recording

ChristopherGabba, 2024-08-03:
### What's happening?

No matter which camera I use (front or back) on my iPhone 12 physical device, I'm getting the width and height of 0.

```typescript
const onRecordingFinished = useCallback(
    async (video: VideoFile) => {
      console.log("Finished recording with", JSON.stringify(video, null, 4))
      setVideoUrl(video.path)
    },
    [setVideoUrl],
  )
```

```logs
Finished recording with {
    "path": "file:///private/var/mobile/Containers/Data/Application/19DBB76D-4103-4B38-AC3A-C57B843D3189/tmp/14C46EA6-AD67-4E6B-88C2-C1BFCFE7C83D.mp4",
    "width": 0,
    "height": 0,
    "duration": 5.033449333
}
```

### Reproduceable Code

```tsx
import { observer } from "mobx-react-lite"
import React, { FC, useCallback, useEffect, useMemo, useRef, useState } from "react"
import { View, ViewStyle } from "react-native"
import { DraggableOverlay } from "./DraggableOverlay"
import { MediaPlayer } from "./MediaPlayer"
import { AppStackScreenProps } from "app/navigators"
import { aS, colors, layout } from "../../theme"
import {
  Camera,
  CameraCaptureError,
  CameraRuntimeError,
  VideoFile,
  useCameraDevice,
  useCameraFormat,
  useFrameProcessor,
} from "react-native-vision-camera"
import {
  Face,
  useFaceDetector,
  FaceDetectionOptions,
} from "react-native-vision-camera-face-detector"
import { Worklets } from "react-native-worklets-core"
import {
  useIsForeground,
  useTimeout,
  useVideoLoadingError,
  useRecordingVolumeManager,
} from "app/hooks"
import { SharedTransitionAndSwipeDownWrapper } from "app/screens/MediaScreens/SharedTransitionAndSwipeDownWrapper"
import { reportCrash } from "app/utils/crashReporting"
import { CountdownTimer } from "./CountdownTimer"
import { PlayerStates } from "./media.types"
import { TimerStatus } from "./timer.types"
import { VolumeManager } from "react-native-volume-manager"
import { useIsFocused } from "@react-navigation/native"
import { EndRecordingButton } from "./EndRecordingButton"
import { TimerBar } from "./TimerBar"
import { AVPlaybackStatus, Audio, Video } from "expo-av"
import { trackAnalytics } from "app/utils/analytics"
import * as Haptics from "expo-haptics"
import { SmileData, generateLaughData } from "app/utils/generateLaughData"
import { generateThumbnailFromVideo } from "app/utils/generateThumbnail"
import NetInfo from "@react-native-community/netinfo"
import { fixLocalPath } from "app/utils/commonFunctions"
[... 517 lines cut ...]
  "hasFlash": true,
  "minZoom": 1,
  "hasTorch": true
}
```

### Device

iPhone 12 Physical

### VisionCamera Version

4.5.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

vishalyad16, 2024-08-04:
@ChristopherGabba 
How can I reduce the size of a video that is 151 MB for a 1-minute recording? The bitRate setting does not seem to be working.

ChristopherGabba, 2024-08-04:
I have provided my entire setup in the details above, not sure what else I could provide. I'm also following all the steps in the docs unless I missed something that I'm not seeing.

stewartiee, 2024-08-06:
I'm seeing the same after trying to upgrade from 4.3.2 -> latest.

Seems to be a bug introduced in [4.4.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.4.0), since that's where it starts to break for me, returning {width: 0, height: 0}.

mrousavy, 2024-08-06:
sorry the bot probably shouldn't have closed this.

thanks for letting me know, I'll try to find some time to look into this. sponsoring me will prioritize that ;)

ChristopherGabba, 2024-08-07:
> thanks for letting me know, I'll try to find some time to look into this. sponsoring me will prioritize that ;)

Done -- Even though this issue is relatively low priority, doesn't affect me that much just wanted to show some appreciation!

mrousavy, 2024-08-07:
Thanks so much Christopher! :)

mrousavy, 2024-08-07:
Hm, I'm browsing the diff but I really can't see anything that might've broken the `naturalSize` getter, for me it's also always returning `0` now... Are you guys both sure that 4.3.2 worked fine?

mrousavy, 2024-08-07:
Could this be an Xcode bug? lol.. I don't see how `naturalSize` can suddenly be 0x0 https://github.com/mrousavy/react-native-vision-camera/blob/77e98178f84824a0b1c76785469413b64dc96046/package/ios/Core/Recording/Track.swift#L59-L61

stewartiee, 2024-08-07:
> Hm, I'm browsing the diff but I really can't see anything that might've broken the `naturalSize` getter, for me it's also always returning `0` now... Are you guys both sure that 4.3.2 worked fine?

Yeah, you're totally correct. I'm not sure what I was seeing yesterday, but is indeed broken on 4.3.2 too.

ChristopherGabba, 2024-08-07:
I unfortunately bumped mine from 4.0.4 all the way to 4.5.1 in one big jump. I truthfully cannot confirm if the issue was there in 4.0.4 but I never remember seeing it until this new version. So I can't be of much help in the immediate until I revert versions and bump up until I find it.

mrousavy, 2024-08-08:
Okay I'll take another look at the changelog, maybe something was missing in the video configuration options

mrousavy, 2024-08-20:
Hey @ChristopherGabba sorry this is going to take a bit longer than usually because I'm in the middle of finishing Nitro Modules, and then I'm going to leave for a 5 day vacation.
It's on my TODO list, promised!

ChristopherGabba, 2024-08-20:
@mrousavy nitro on brother, no worries! Enjoy your vacation

jcgoodru, 2024-09-11:
Can confirm this is happening on Android as well (Pixel 5 emulator)

bitsmanent, 2024-09-26:
On my Android emulator the issue only happens when changing *facing* while recording. Otherwise it works properly.

ChristopherGabba, 2024-09-27:
@mrousavy Along the same lines as this issue where it's returning the incorrect recorded size, on my physical Google Pixel 3 (I know its old), I predefine the camera format like so:
```typescript
  const format = useCameraFormat(device, [
    { videoStabilizationMode: "standard" },
    { fps: targetFps },
    { videoAspectRatio: 16 / 9 },
    { videoResolution: { width: 308, height: 548 } }, // Note pre-defined size here
    { photoAspectRatio: 16 / 9 },
    { photoResolution: { width: 308, height: 548 } },
  ])
```

And after I finish recording a video, I get:
```logs
Finished recording with {
    "height": 1280,
    "width": 720,
    "duration": 5.579,
    "path": "/data/user/0/com.pwwindustries.reelfeel/cache/mrousavy-6443249496622499435.mp4"
}
```

Maybe while you are diagnosing out the `0, 0` issue on iOS you can also figure out why Android is returning a larger size than I describe up front in the initial config?

mrousavy, 2024-09-30:
I think this is unrelated - Android might simply not have a format that small for video capture and chooses a different one. 

But I'll figure out the iOS size issue either way! Lots of other priorities with Nitro atm, but I'll have some time soon

phazei, 2024-11-21:
I'm also having this issue.  version 4.5.3

iPhone XR iOS 17.6

It works fine on Android

haquex19, 2025-02-22:
Is there an update on this?

It seems I am running into this issue when the camera is switched while recording. For example, if I start recording using the camera, and while recording I switch to the front camera, dimensions are now 0.