margelo/react-native-vision-camera #3121: 💭 Possible to zoom with useSkiaFrameProcessor?

alexstanbury, 2024-08-05:
### Question

In the example from this repo, I am using a skia frame processor to draw scanned barcodes. Unfortunately, I don't seem to be able to zoom in with it enabled. If I remove the frame processor zooming works as expected.

I can't seem to see any way of setting `pointerEvents:'none'` on the Skia canvas, even when looking at the library code.

Am I missing something obvious here? 🤔

code snippets:
```
 const codeCorners = useSharedValue<Point[][]>([])
  const codeScannerFrame = useSharedValue<CodeScannerFrame>({ width: 0, height: 0 })
```

```
     <Camera
          frameProcessor={frameProcessor}
          format={format}
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={isActive}
          codeScanner={codeScanner}
          enableZoomGesture={true}
          zoom={device.neutralZoom}
          torch={torchState}
        />
```
```
  const frameProcessor = useSkiaFrameProcessor(
    (frame) => {
      'worklet'
      frame.render()
     // stuff n things here
    },
    [codeScannerFrame, device, codeCorners],
  )
```
```
  const codeScanner = useCodeScanner({
    codeTypes: [
      'code-128',
      'code-39',
      'code-93',
      'codabar',
      'ean-13',
      'ean-8',
      'itf',
      'upc-e',
      'upc-a',
      'qr',
      'pdf-417',
      'aztec',
      'data-matrix',
    ],
    onCodeScanned: onCodeScanned,
  })
```
```
const onCodeScanned = useCallback(
    (codes: Code[], frame: CodeScannerFrame) => {
      codeScannerFrame.value = frame
      if (codes.length > 0) {
        const allCorners = codes.map((code) => code.corners ?? [])
        if (allCorners.length > 0) codeCorners.value = allCorners
      }
      if (clearTimer.current !== undefined) clearTimeout(clearTimer.current)

      clearTimer.current = setTimeout(() => {
        console.log('clearing codeCorners')
        codeCorners.value = []
      }, 250)
    },
    [codeScannerFrame, codeCorners],
  )
```

### What I tried

Looking at the internals, adding/removing the processor

### VisionCamera Version

4.5.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

alexstanbury, 2024-08-05:
With all due respect Hans, why am I allowed to ask a question in the first place if you're just gonna put it straight in the bin? 😂...Damn bots!

I would have asked on Discord but the app won't even open for me at the moment. 😭

mrousavy, 2024-08-05:
damn maybe i should configure him to allow questions

alexstanbury, 2024-08-05:
Should I change the question to a bug?

mrousavy, 2024-08-05:
Yea I think a bug is better. I think we'd need logs for that, and this might be a valid issue. As a workaround you can implement your own zoom handler

closed 2024-08-05 by mrousavy

alexstanbury, 2024-08-05:
From reading the docs, the following sounds like it may be why it's not working:

```Skia Frame Processors will disable the native Preview to use a custom Skia Canvas instead.```

Presumably the zooming gesture code is attached to the preview view, I'll dig into it to confirm tomorrow, and have a play around with building my own gesture handler. 

For the issue, what sort of logs would be helpful?