margelo/react-native-vision-camera #3122: 💭 Low video resolution.

sw-tt-hitendrachavda, 2024-08-06:
### Question

i want to reduce the size of video capture in android currently 0.10 seconds video it's giving me 20 mb size so how can i reduce the size.

### What I tried

const frontformate = useCameraFormat(devicesFront, [
    { fps: 30 },
    { videoResolution: { width: 1280, height: 720 } },
    ]);
   
  const backformate =  useCameraFormat(devicesBack, [
    { fps: 30 },
    { videoResolution: { width: 1280, height: 720 } },
    ]);

<Camera
            ref={cameraRef}
            style={[styles.camera]}
            audio={true}
            video={true}
            device={cameraType === 'front' ? devicesFront : devicesBack}
            isActive={true}
            format={cameraType === 'front' ? frontformate : backformate}
            outputOrientation={'device'}
          /> 

### VisionCamera Version

4.4.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-08-06:
Reduce the `videoResolution`.

closed 2024-08-06 by mrousavy

sw-tt-hitendrachavda, 2024-08-07:
Yes i try that 

but i want 
{ videoResolution: { width: 1280, height: 720 } } 
with low size video