margelo/react-native-vision-camera #3127: 🐛 The picture looks blurry ( out of focus) in Camera View but comes out focused after clicking on Ipad

rsainiWin, 2024-08-08:
### What's happening?

I'm encountering problems with camera focus before taking a picture. The camera attempts to focus but consistently fails.

I've implemented tap-to-focus functionality, but it's returning undefined unexpectedly.

Expected Behavior:
The Camera should focus successfully before taking a picture.
Tap-to-focus should be able to focus at point.

Actual Behavior:
The Camera fails to focus.
Tap-to-focus returns undefined.

This is an issue with most users using IPad, I'd this issue on 4.3.1 as well

I tried Installing shadowlens it's working fine over there

### Reproduceable Code

```tsx
const camera = useRef<Camera>(null);
const onFocusTap = useCallback(
    async ({nativeEvent: event}: GestureResponderEvent) => {
      if (!device?.supportsFocus) return;
      await camera.current?.focus({
        x: event.locationX,
        y: event.locationY,
      });
    },
    [device?.supportsFocus],
  );

<Reanimated.View onTouchEnd={onFocusTap} style={StyleSheet.absoluteFill}>
  <ReanimatedCamera
    style={StyleSheet.absoluteFill}
    device={device}
    isActive={isEnabled}
    ref={camera}
    enableHighQualityPhotos={false}
    format={format}
    photoQualityBalance={'balanced'}
    orientation="portrait"
    photo={true}
  />
</Reanimated.View>
```

### Relevant log output

```shell
default	20:55:49.657092+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke_2: <private> changed... newDownplayUserPreferredCameraOverrideHistory = F (based on <private>), new user preferred camera history <private>
default	20:55:49.657197+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke: dispatching async to preferred camera property refresh queue
default	20:55:49.657313+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _cameraHistoryDispatchQueue]_block_invoke: using main queue for refreshing camera properties
default	20:55:49.657428+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke: async dispatch to preferred camera property refresh queue returned
default	20:55:49.657511+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke: change handler for <private> returning
default	20:55:49.657579+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke: not refreshing camera properties because camera history is not set up yet.
default	20:55:49.657721+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke: <private> changed... newDownplayUserPreferredCameraOverrideHistory = F (based on <private>), new user preferred camera override history <private>
default	20:55:49.657773+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke: dispatching async to preferred camera property refresh queue
default	20:55:49.657818+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke: async dispatch to preferred camera property refresh queue returned
default	20:55:49.657864+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke: change handler for <private> returning
default	20:55:49.657910+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke: not refreshing camera properties because camera history is not set up yet.
default	20:55:49.658056+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke: <private> changed... new streaming camera history <private>
default	20:55:49.658102+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke: dispatching async to preferred camera property refresh queue
default	20:55:49.658147+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke: async dispatch to preferred camera property refresh queue returned
default	20:55:49.658220+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke: change handler for <private> returning
default	20:55:49.658272+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke: not refreshing camera properties because camera history is not set up yet.
default	20:55:49.658320+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke: doing deferred dispatch async to preferred camera property refresh queue
default	20:55:49.658456+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _setUpCameraHistoryOnce]_block_invoke_2: deferred async dispatch to preferred camera property refresh queue returned
default	20:55:49.658569+0530	WINspect-dev	<<<< AVCaptureFigVideoDevice >>>> +[AVCaptureFigVideoDevice _refreshPreferredCameraProperties:]: getting current value for userPreferredCamera (allowsSuspendedPreferredCameras = F)
[... 269 lines cut ...]
 "supportsLowLightBoost": false,
 "maxExposure": 8,
 "sensorOrientation": "portrait"
}
```

### Device

IPad pro 11 inch 2gen

### VisionCamera Version

4.5.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rsainiWin, 2024-08-08:
These are logs from xcode only

rsainiWin, 2024-08-08:
https://github.com/user-attachments/assets/f876e971-ab7c-41a1-97f8-4eca2aa9a174

mrousavy, 2024-08-08:
What `format` are you using

rsainiWin, 2024-08-09:
This is what we use in general
    {autoFocusSystem: 'phase-detection'},
    {photoAspectRatio: 1},
    {
      photoResolution:
        Platform.OS === 'ios' ? 'max' : {width: 1080, height: 1080},
    },
    
    I've tried different focus system and 
    removed other items as well

mrousavy, 2024-08-09:
No, I'm saying show me your format. Log it. Print it's content. Not the format _filter_.

rsainiWin, 2024-08-09:
{"autoFocusSystem": "contrast-detection", "fieldOfView": 104.81095123291016, "maxFps": 60, "maxISO": 1728, "minFps": 2, "minISO": 18, "photoHeight": 1512, "photoWidth": 2016, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 480, "videoStabilizationModes": ["auto", "off"], "videoWidth": 640}

rsainiWin, 2024-08-09:
Also when i set to phase-detection 
the photo capture fails with below error 
 ` Failed to take photo! [capture/unknown: Error Domain=AVFoundationErrorDomain Code=-11800 "The operation could not be completed" UserInfo={NSUnderlyingError=0x302c23e40 {Error Domain=NSOSStatusErrorDomain Code=-16802 "(null)"}, NSLocalizedFailureReason=An unknown error occurred (-16802), AVErrorRecordingFailureDomainKey=4, NSLocalizedDescription=The operation could not be completed}]`

rsainiWin, 2024-08-09:
Also the Blur view has increased further on the latest version of library vs 4.3.1

rsainiWin, 2024-08-11:
one of my Iphone 15 pro user also reported the same

rsainiWin, 2024-08-14:
@mrousavy 
Findings:
If i  pass empty array to  filter in format then Blurry issue is present
If i dont pass the format prop itself then there's no blur issue

lukebars, 2024-10-28:
@rsainiWin I'm getting reports from users using iPhone 15 pro, that focus isn't working at all. Have you experienced anything such as that?

rsainiWin, 2024-10-28:
@lukebars I switched to native camera implementation.
As camera was one of the very key feature for us.