margelo/react-native-vision-camera #3128: 🐛  ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera

betterphp, 2024-08-09:
### What's happening?

Trying to implement a frame processor that will use `frame.toArrayBuffer()` to get the colour at the centre of the frame. Any call to that method throw the error in the title.

Tracked it down to [here](https://github.com/mrousavy/react-native-vision-camera/blob/00caff0288e278e1076cb5b0708bb1ba7f436f07/package/android/src/main/cpp/frameprocessors/FrameHostObject.cpp#L172) but looking at the docs for `AHardwareBuffer_lock` I can't really see why it would fail.

### Reproduceable Code

```tsx
import { useState } from "react";
import { StyleSheet } from "react-native";
import {
    Camera,
    useCameraPermission,
    useFrameProcessor,
    useCameraDevice,
    useSharedValue,
    runAtTargetFps,
} from "react-native-vision-camera";
import { Worklets } from "react-native-worklets-core";
import { Text, Box, Button, ButtonText } from "@gluestack-ui/themed";

const styles = StyleSheet.create({
    container: {
        flex: 1,
        backgroundColor: "#f1f1f1",
        alignItems: "center",
        justifyContent: "center",
    },
    camera: {
        width: 400,
        height: 400,
    },
});

export default () => {
    const [ selectedColor, setSelectedColor ] = useState(null);
    const device = useCameraDevice("back");
    const { hasPermission, requestPermission } = useCameraPermission();

    if (device === null) {
        // No camera
        return <Box />;
    }

    if (!hasPermission) {
        // Camera permissions are not granted yet.
        return (
            <Box style={styles.container}>
                <Text>We need your permission to show the camera</Text>
                <Button onPress={requestPermission}>
                    <ButtonText>
                        Grant permission
                    </ButtonText>
                </Button>
            </Box>
        );
    }

    const onColorSampled = Worklets.createRunOnJS(
        (color) => setSelectedColor(color)
    );

    const frameProcessor = useFrameProcessor((frame) => {
        "worklet";

        runAtTargetFps(1, () => {
            "worklet";

            // The exception is thrown here
            const buffer = frame.toArrayBuffer();

            // This was just placeholder testing stuff
            onColorSampled(`Frame: ${frame.width}x${frame.height} (${frame.pixelFormat})`);
        });
    }, [onColorSampled])

    return (
        <Box style={styles.container}>
            <Camera
                frameProcessor={frameProcessor}
                style={styles.camera}
                device={device}
                isActive={true}
            />
            <Button>
                <ButtonText>
                    {`color or something ${selectedColor}`}
                </ButtonText>
            </Button>
        </Box>
    );
}
```

### Relevant log output

```shell
ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
 ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
 ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
 ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
 ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
 ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
 ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
 ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
 ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
 ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
 ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
 ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
 ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
 ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
 ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
 ERROR  Frame Processor Error: Failed to lock HardwareBuffer for reading!, js engine: VisionCamera
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 1,
  "minZoom": 1,
  "maxExposure": 6,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "ultra-wide-angle-camera"
  ],
  "supportsFocus": false,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 999.999985098839,
  "minExposure": -6,
  "name": "10 (BACK) androidx.camera.camera2",
  "hasFlash": false,
  "hasTorch": false,
  "position": "back",
  "id": "10"
}
```

### Device

Android Emulator, Pixel 8 Pro API 34

### VisionCamera Version

4.5.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

betterphp, 2024-08-11:
Yeah I've tested in on a real phone and it works fine there.

Would still be helpful if it worked running on an emulator though.

betterphp, 2024-08-11:
I had a second issue relating to this; the returned array was cropped and only returned about a fifth of the image - tested by storing the array as a json string and using a rubbish PHP script to render it back to an image.

In the end I solved it by using https://github.com/mrousavy/vision-camera-resize-plugin to extract the data - which actually turned out to be closer to what I needed in the first place anyway :)

Just leaving this update in case anyone runs in to similar problems.

mrousavy, 2024-08-20:
Hey - thanks for opening this issue. To be honest, I think this is something I have no control over. This is likely just the HAL on the Emulator being implemented poorly, there's no way to get the data as a buffer then.

mrousavy, 2024-08-20:
Maybe try a different emulator, the ones with the Google Play Services are usually better

closed 2024-08-20 by mrousavy