margelo/react-native-vision-camera #3129: 🐛 Unable to take photos or videos with custom path. Android with expo file system.

emilje, 2024-08-10:
### What's happening?

On iOS everything runs fine. 

When taking a photo:
```
[capture/invalid-path]: The given path (file:///data/user/0/com.emilje.ourea/cache/Camera) is invalid, or not writable!
```

When taking a video:
```
 {"cause": {"message": "The given path (file:///data/user/0/com.emilje.ourea/cache/Camera/) is invalid, or not writable!", "stacktrace": "com.mrousavy.camera.core.InvalidPathError: The given path (file:///data/user/0/com.emilje.ourea/cache/Camera/) is invalid, or not writable!
        at com.mrousavy.camera.core.utils.FileUtils$Companion.getDirectory(FileUtils.kt:18)
        at com.mrousavy.camera.core.types.RecordVideoOptions$Companion.fromJSValue(RecordVideoOptions.kt:17)
        at com.mrousavy.camera.react.CameraViewModule$startRecording$1.invokeSuspend(CameraViewModule.kt:123)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:104)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
        at java.lang.Thread.run(Thread.java:1012)
"}, "code": "capture/invalid-path", "message": "The given path (file:///data/user/0/com.emilje.ourea/cache/Camera/) is invalid, or not writable!", "userInfo": null}
```

### Reproduceable Code

```tsx
const BACK_CAMERA = useCameraDevice("back");
  const FRONT_CAMERA = useCameraDevice("front");
  const [device, setDevice] = useState(BACK_CAMERA);
  const format = useCameraFormat(device, [
    {
      fps: 60,
      videoResolution: { width: 1280, height: 720 },
    },
  ]);

<View style={[theme.flex, { backgroundColor: "black" }]}>
      {renderBackButton()}
      {renderSideBar()}
      {renderBottomBar()}
      <Camera
        ref={camera}
        device={device}
        format={format}
        isActive={true}
        torch={isTorch ? "on" : "off"}
        enableZoomGesture
        photo={true}
        video={true}
        audio={true}
        style={{ flex: 1, zIndex: -1 }}
      />
    </View>
```

### Relevant log output

```shell
Not using xcode or android studio.
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung Galaxy A52s 5G

### VisionCamera Version

4.5.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

emilje, 2024-08-11:
@mrousavy 
Hi, I am unable to run the example project, so I could not tick that box. Could you perhaps reopen this and take a look?
```
FAILURE: Build failed with an exception.

* Where:
Build file 'C:\Users\Emil\Desktop\react-native-vision-camera\package\android\build.gradle' line: 70

* What went wrong:
A problem occurred evaluating project ':react-native-vision-camera'.
> react-native-vision-camera: Failed to find node_modules/ path!
```

emilje, 2024-08-11:
Here is a bit of info if anyone comes with the same issue:

Expo file system paths, both start with "file://":
ios: ```file:///var/mobile/Containers/Data/Application/C5758717-0976-440B-934B-14F94736EE25/Library/Caches/Camera/```
android: ```file:///data/user/0/com.emilje.ourea/cache/Camera/```

Passing these paths to ```path``` prop on iOS works fine, but does not work on Android.
Instead, stripping the "file://" from the path beforehand works on both Android and iOS.
ios: ```/var/mobile/Containers/Data/Application/C5758717-0976-440B-934B-14F94736EE25/Library/Caches/Camera/```
android: ```/data/user/0/com.emilje.ourea/cache/Camera/```

I've made some helper functions like this so I can modify the filepaths as required throughout my app.
```
export const stripFileFromFilePath = (filePath: string) => {
  if (filePath.startsWith("file://")) {
    return filePath.replace("file://", "");
  }

  return filePath;
};

export const insertFileToFilePath = (filePath: string) => {
  if (!filePath.startsWith("file://")) {
    return "file://" + filePath;
  }

  return filePath;
};
```

Here is an example of using it with expo file system.

```
 let cameraFolder = await FileSystem.getInfoAsync(CAMERA_FOLDER_PATH);
    if (!cameraFolder.exists) {
      await FileSystem.makeDirectoryAsync(CAMERA_FOLDER_PATH);
      cameraFolder = await FileSystem.getInfoAsync(CAMERA_FOLDER_PATH);

      if(!cameraFolder.exists || !cameraFolder.isDirectory) {
        logError("Could not create camera folder in VisionCameraScreen.");
        return;
      }
    }

    if (!isVideo) {
      setIsWaitingTransition(true);
      const photo = await camera.current.takePhoto({
        flash: isPhotoFlash ? "on" : "off",
        path: stripFileFromFilePath(CAMERA_FOLDER_PATH),
      });
```

jonjamz, 2024-08-14:
This works for me as well. I needed to prepend `file://` to the path after capture otherwise I received errors from Expo FileSystem when working with the path. Glad this feature exists now!

mrousavy, 2024-08-15:
Thanks!

We can add this `file://` parsing to Android as well at some point, but for now I consider this low priority as there is a simple workaround.

closed 2024-08-15 by mrousavy

caseybrookes, 2024-09-21:
> We can add this file:// parsing to Android as well at some point

@mrousavy is there an issue we can follow for this? would love to know when we can remove this workaround

aaa3334, 2024-12-18:
Thanks Emilje! Was struggling trying to debug this for an hour or two just then. Your work around worked great!