margelo/react-native-vision-camera #3130: 💭How to wrap an overlay around the preview and render a square output

samducker, 2024-08-12:
### Question

Looking to try to do what instagram do here, providing the user with a clear view into what they are taking, whilst also maintaining quality when i use photoAspectRatio: 1 in format the preview goes terrible quality even with video resolution set to 'max'

Looking to achieve the effect shown in this photo 
[image]

### What I tried

using useCameraFormat 

### VisionCamera Version

4.5.1

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

samducker, 2024-08-12:
Hi @mrousavy I just used the provided question template, doesn't seem like an bug or issue.

mrousavy, 2024-08-27:
Hey - for stuff like this, please use the community discord. This doesn't seem like any question related to camera internals, it's more of a "how do I lay out the react view properly" thing.

closed 2024-08-27 by mrousavy