margelo/react-native-vision-camera #3131: 💭 Issue with FrameProcessorsUnavailableError

KarthiPN, 2024-08-12:
### Question

I have used useFrameprocessor like the code mentioned below but no matter what I do I'm getting " ERROR  TypeError: Cannot read property 'FrameProcessorsUnavailableError' of undefined, js engine: hermes"

### What I tried

import React, { useState, useEffect } from 'react';
import { View, Text, StyleSheet, Button } from 'react-native';
import { Camera, useCameraDevice,CameraPermissionStatus, useFrameProcessor } from 'react-native-vision-camera';

const CameraVision = ()=>{
    const [frameCount, setFrameCount] = useState<number>(0);
    const [cameraPermission, setCameraPermission] = useState<CameraPermissionStatus>('not-determined');
    const device = useCameraDevice('front');

    useEffect(() => {
        const getPermission = async () => {
          const status = await Camera.requestCameraPermission();
          setCameraPermission(status);
        };
        getPermission();
    }, []);

      const frameProcessor = useFrameProcessor((frame) => {
        if (!frame){
            console.log("Frame error")
        }
        'worklet'; // Required for worklet code

        // Access frame properties
        console.log('Frame Dimensions are:', frame.width, frame.height);
        console.log('Pixel format:', frame.pixelFormat);
        setFrameCount(prevCount => prevCount + 1);
      }, []);

      if (cameraPermission === 'not-determined') {
        return <Text>Requesting camera permission...</Text>;
      }
    
      if (cameraPermission === 'denied') {
        return <Text>Camera permission denied</Text>;
      }
    
      if (!device) {
        return <Text>Loading camera...</Text>;
      }
    

    return(
        <View>
            <Camera

            
            frameProcessor={frameProcessor}
            style={{'width':'100%',height:'80%'}}
            device = {device}
            isActive={true}
            // onFrame={frame => {
            //     // Custom frame processing logic
            //     console.log('Received frame', frame);
            //   }}
          />
          <Text>Frame Count: {frameCount}</Text>

        </View>
        
        
    )
}

export {CameraVision}

### VisionCamera Version

latest

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

lucksp, 2024-08-12:
Did you install `react-native-worklets-core`?

mrousavy, 2024-08-14:
This is a bug, not a question. You used the wrong issue template.

ripon008, 2024-10-15:
TypeError: Cannot read property 'FrameProcessorsUnavailableError' of undefined, js engine: hermes

same issue 
plase help

daisy-cao-csy, 2024-12-03:
ERROR  Error: Requiring unknown module "undefined". If you are sure the module exists, try restarting Metro. You may also want to run `yarn` or `npm install`., js engine: hermes
 ERROR  TypeError: Cannot read property 'FrameProcessorsUnavailableError' of undefined, js engine: hermes
 same issue

soparifly, 2025-04-21:
npm i react-native-worklets-core