margelo/react-native-vision-camera #3136: 💭 Lanscape mode issue.

sw-tt-hitendrachavda, 2024-08-12:
### Question

i Have camera screen and when i open the model of camera by default it is coming in 90 clock wise rotate.

### What I tried

 const frontformate = useCameraFormat(devicesFront, [
    { fps: 30 },
    { videoResolution: { width: 1280, height: 720 } },
    ]);
   
  const backformate =  useCameraFormat(devicesBack, [
    { fps: 30 },
    { videoResolution: { width: 1280, height: 720 } },
    ]);
 
 <Camera
            ref={cameraRef}
            style={[styles.camera]}
            audio={true}
            video={true}
            device={cameraType === 'front' ? devicesFront : devicesBack}
            isActive={true}
            format={cameraType === 'front' ? frontformate : backformate}
          /> 

### VisionCamera Version

4.5.1

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

chandanpradhanin, 2024-08-13:
I am encountering an issue with image orientation, where the captured images are displayed differently than expected. This problem appears to be related to orientation and did not occur in version 2.13.2. However, it is now present in version 4.4.2. Could you please provide a solution for this issue?
[image]

mrousavy, 2024-08-14:
This is a bug, not a question. You used the wrong issue template.

closed 2024-08-14 by mrousavy