margelo/react-native-vision-camera #3138: 🔧 Build fails after upgrading to targetSdkVeersion 34

ncsdrgsh, 2024-08-13:
### How were you trying to build the app?

1. npx @react-native-community/cli@latest init VisionCameraRN71 --version 0.71.8
2. Upgrade to targetSdkVersion: 34 https://gist.github.com/ncsdrgsh/a1c6af8fbe4525763316e14a1bafc8cc
3. npm install react-native-vision-camera@latest
5. Run project

### Full build logs

```tsx
> Task :react-native-vision-camera:buildCMakeDebug[arm64-v8a]

> Task :react-native-vision-camera:compileDebugKotlin FAILED
e: Incompatible classes were found in dependencies. Remove them from the classpath or use '-Xskip-metadata-version-check' to suppress errors
e: /Users/dragosh/.gradle/caches/transforms-3/32d978deed3eb0bf21efaed8712457d2/transformed/jetified-kotlinx-coroutines-android-1.8.1.jar!/META-INF/kotlinx-coroutines-android.kotlin_module: Module was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
e: /Users/dragosh/.gradle/caches/transforms-3/5d6ab5c18871220d9a0bf6ca6cbc7623/transformed/jetified-kotlin-stdlib-1.9.21.jar!/META-INF/kotlin-stdlib-jdk7.kotlin_module: Module was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
e: /Users/dragosh/.gradle/caches/transforms-3/5d6ab5c18871220d9a0bf6ca6cbc7623/transformed/jetified-kotlin-stdlib-1.9.21.jar!/META-INF/kotlin-stdlib-jdk8.kotlin_module: Module was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
e: /Users/dragosh/.gradle/caches/transforms-3/5d6ab5c18871220d9a0bf6ca6cbc7623/transformed/jetified-kotlin-stdlib-1.9.21.jar!/META-INF/kotlin-stdlib.kotlin_module: Module was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
e: /Users/dragosh/.gradle/caches/transforms-3/824edf9723e6db6eb5944f5f3fc51869/transformed/jetified-kotlinx-coroutines-core-jvm-1.8.1.jar!/META-INF/kotlinx-coroutines-core.kotlin_module: Module was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraConfiguration.kt: (53, 33): Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
The class is loaded from /Users/dragosh/.gradle/caches/transforms-3/5d6ab5c18871220d9a0bf6ca6cbc7623/transformed/jetified-kotlin-stdlib-1.9.21.jar!/kotlin/Unit.class
e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraConfiguration.kt: (79, 4): Class 'kotlin.Suppress' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
The class is loaded from /Users/dragosh/.gradle/caches/transforms-3/5d6ab5c18871220d9a0bf6ca6cbc7623/transformed/jetified-kotlin-stdlib-1.9.21.jar!/kotlin/Suppress.class
e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraDeviceDetails.kt: (32, 20): Unresolved reference: atan2
e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraDeviceDetails.kt: (33, 20): Unresolved reference: min
e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraDeviceDetails.kt: (34, 20): Unresolved reference: sqrt
e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraDeviceDetails.kt: (37, 2): Class 'kotlin.Suppress' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
The class is loaded from /Users/dragosh/.gradle/caches/transforms-3/5d6ab5c18871220d9a0bf6ca6cbc7623/transformed/jetified-kotlin-stdlib-1.9.21.jar!/kotlin/Suppress.class

[...]

e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/utils/CallbackPromise.kt: (10, 7): Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
The class is loaded from /Users/dragosh/.gradle/caches/transforms-3/5d6ab5c18871220d9a0bf6ca6cbc7623/transformed/jetified-kotlin-stdlib-1.9.21.jar!/kotlin/Unit.class
e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/utils/CallbackPromise.kt: (10, 41): Unresolved reference: stackTraceToString
e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/utils/CallbackPromise.kt: (12, 9): Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
The class is loaded from /Users/dragosh/.gradle/caches/transforms-3/5d6ab5c18871220d9a0bf6ca6cbc7623/transformed/jetified-kotlin-stdlib-1.9.21.jar!/kotlin/Unit.class
e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/utils/CallbackPromise.kt: (19, 7): Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
The class is loaded from /Users/dragosh/.gradle/caches/transforms-3/5d6ab5c18871220d9a0bf6ca6cbc7623/transformed/jetified-kotlin-stdlib-1.9.21.jar!/kotlin/Unit.class
e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/utils/CallbackPromise.kt: (20, 7): Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
The class is loaded from /Users/dragosh/.gradle/caches/transforms-3/5d6ab5c18871220d9a0bf6ca6cbc7623/transformed/jetified-kotlin-stdlib-1.9.21.jar!/kotlin/Unit.class
e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/utils/CallbackPromise.kt: (21, 7): Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
The class is loaded from /Users/dragosh/.gradle/caches/transforms-3/5d6ab5c18871220d9a0bf6ca6cbc7623/transformed/jetified-kotlin-stdlib-1.9.21.jar!/kotlin/Unit.class
e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/utils/CallbackPromise.kt: (22, 7): Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
The class is loaded from /Users/dragosh/.gradle/caches/transforms-3/5d6ab5c18871220d9a0bf6ca6cbc7623/transformed/jetified-kotlin-stdlib-1.9.21.jar!/kotlin/Unit.class
e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/utils/withPromise.kt: (8, 3): Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
The class is loaded from /Users/dragosh/.gradle/caches/transforms-3/5d6ab5c18871220d9a0bf6ca6cbc7623/transformed/jetified-kotlin-stdlib-1.9.21.jar!/kotlin/Unit.class
e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/utils/withPromise.kt: (10, 13): Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
The class is loaded from /Users/dragosh/.gradle/caches/transforms-3/5d6ab5c18871220d9a0bf6ca6cbc7623/transformed/jetified-kotlin-stdlib-1.9.21.jar!/kotlin/Unit.class
e: /Users/dragosh/Projects/PheonixRN71/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/react/utils/withPromise.kt: (12, 7): Class 'kotlin.Unit' was compiled with an incompatible version of Kotlin. The binary version of its metadata is 1.9.0, expected version is 1.7.1.
[... cut ...]
### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ncsdrgsh, 2024-08-13:
If I specify the kotlin version to be 1.9.0 in my gradle files I endup with another error:

```
Executing tasks: [:app:clean, :react-native-gradle-plugin:clean, :react-native-vision-camera:clean, :app:assembleDebugUnitTest, :app:assembleDebugAndroidTest, :app:assembleDebug, :react-native-gradle-plugin:assemble, :react-native-gradle-plugin:testClasses, :react-native-vision-camera:assembleDebug, :react-native-vision-camera:assembleDebugUnitTest, :react-native-vision-camera:assembleDebugAndroidTest] in project /Users/dragosh/Projects/PheonixRN71/android

Starting Gradle Daemon...
Gradle Daemon started in 559 ms
> Task :react-native-gradle-plugin:compileKotlin UP-TO-DATE
> Task :react-native-gradle-plugin:compileJava NO-SOURCE
> Task :react-native-gradle-plugin:pluginDescriptors UP-TO-DATE
> Task :react-native-gradle-plugin:processResources UP-TO-DATE
> Task :react-native-gradle-plugin:classes UP-TO-DATE
> Task :react-native-gradle-plugin:inspectClassesForKotlinIC UP-TO-DATE
> Task :react-native-gradle-plugin:jar UP-TO-DATE

> Configure project :react-native-vision-camera
[VisionCamera] Thank you for using VisionCamera ❤️
[VisionCamera] If you enjoy using VisionCamera, please consider sponsoring this project: https://github.com/sponsors/mrousavy
[VisionCamera] node_modules found at /Users/dragosh/Projects/PheonixRN71/node_modules
[VisionCamera] VisionCamera_enableFrameProcessors is set to true!
[VisionCamera] react-native-worklets-core not found, Frame Processors are disabled!
[VisionCamera] VisionCamera_enableCodeScanner is set to false!
Warning: This version only understands SDK XML versions up to 3 but an SDK XML file of version 4 was encountered. This can happen if you use versions of Android Studio and the command-line tools that were released at different times.
AGPBI: {"kind":"warning","text":"We recommend using a newer Android Gradle plugin to use compileSdk = 34\n\nThis Android Gradle plugin (7.3.1) was tested up to compileSdk = 33\n\nThis warning can be suppressed by adding\n    android.suppressUnsupportedCompileSdk=34\nto this project's gradle.properties\n\nThe build will continue, but you are strongly encouraged to update your project to\nuse a newer Android Gradle Plugin that has been tested with compileSdk = 34","sources":[{}]}

> Task :react-native-gradle-plugin:clean
> Task :react-native-gradle-plugin:assemble UP-TO-DATE
> Task :react-native-gradle-plugin:compileTestKotlin NO-SOURCE
> Task :react-native-gradle-plugin:compileTestJava NO-SOURCE
> Task :react-native-gradle-plugin:processTestResources NO-SOURCE
> Task :react-native-gradle-plugin:testClasses UP-TO-DATE
> Task :app:clean UP-TO-DATE
> Task :react-native-vision-camera:deleteCmakeCache
> Task :react-native-vision-camera:externalNativeBuildCleanDebug
> Task :react-native-vision-camera:externalNativeBuildCleanRelease
> Task :react-native-vision-camera:clean UP-TO-DATE
> Task :app:assembleDebugUnitTest UP-TO-DATE
> Task :app:buildCodegenCLI SKIPPED
> Task :app:generateCodegenSchemaFromJavaScript SKIPPED
> Task :app:generateCodegenArtifactsFromSchema SKIPPED
> Task :app:generatePackageList
> Task :app:preBuild
> Task :app:preDebugBuild
> Task :react-native-vision-camera:prepareHeaders
> Task :react-native-vision-camera:preBuild
> Task :react-native-vision-camera:preDebugBuild
> Task :react-native-vision-camera:compileDebugAidl NO-SOURCE
> Task :app:compileDebugAidl NO-SOURCE
> Task :react-native-vision-camera:packageDebugRenderscript NO-SOURCE
> Task :app:compileDebugRenderscript NO-SOURCE
> Task :app:dataBindingMergeDependencyArtifactsDebug
> Task :app:generateDebugResValues
> Task :app:generateDebugResources
> Task :react-native-vision-camera:compileDebugRenderscript NO-SOURCE
> Task :react-native-vision-camera:generateDebugResValues
> Task :react-native-vision-camera:generateDebugResources
> Task :react-native-vision-camera:packageDebugResources
> Task :app:generateDebugBuildConfig
> Task :app:mapDebugSourceSetPaths
> Task :app:createDebugCompatibleScreenManifests
> Task :app:extractDeepLinksDebug
> Task :react-native-vision-camera:extractDeepLinksDebug
> Task :react-native-vision-camera:generateDebugBuildConfig
> Task :app:preDebugAndroidTestBuild SKIPPED
> Task :app:compileDebugAndroidTestAidl NO-SOURCE
> Task :react-native-vision-camera:writeDebugAarMetadata
> Task :app:javaPreCompileDebug
> Task :react-native-vision-camera:javaPreCompileDebug
> Task :react-native-vision-camera:compileDebugLibraryResources
> Task :app:dataBindingMergeDependencyArtifactsDebugAndroidTest
[... 319 lines cut ...]
         > Error while dexing.
   > Failed to transform camera-view-1.4.0-beta02.aar (androidx.camera:camera-view:1.4.0-beta02) to match attributes {artifactType=android-dex, asm-transformed-variant=NONE, dexing-enable-desugaring=true, dexing-enable-jacoco-instrumentation=false, dexing-is-debuggable=true, dexing-min-sdk=21, org.gradle.category=library, org.gradle.dependency.bundling=external, org.gradle.libraryelements=aar, org.gradle.status=release, org.gradle.usage=java-runtime}.
      > Execution failed for DexingWithClasspathTransform: /Users/dragosh/.gradle/caches/transforms-3/6e487541fd4d802f265ca65bb263191d/transformed/jetified-camera-view-1.4.0-beta02-runtime.jar.
         > Error while dexing.
   > Failed to transform camera-video-1.4.0-beta02.aar (androidx.camera:camera-video:1.4.0-beta02) to match attributes {artifactType=android-dex, asm-transformed-variant=NONE, dexing-enable-desugaring=true, dexing-enable-jacoco-instrumenta [... cut ...]
      > Execution failed for DexingWithClasspathTransform: /Users/dragosh/.gradle/caches/transforms-3/3da703ec181ad35dd0573c196163b692/transformed/jetified-camera-core-1.4.0-beta02-runtime.jar.
         > Error while dexing.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

* Get more help at https://help.gradle.org

BUILD FAILED in 28s
110 actionable tasks: 100 executed, 10 up-to-date
```

ncsdrgsh, 2024-08-13:
"have you tried to turn it off and on"

I tried to clear invalidate android caches and repair IDE but with no success. I the end I uninstall android studio, restarted my laptop and reinstalled android studio and it "magically worked" got no clue why.

ThiagoInstivo, 2024-08-14:
I have de same problem, but I was see releases notes https://github.com/mrousavy/react-native-vision-camera/releases and in version  4.3.2 is require upgrade version on react native to 0.73 or above, maybe resolves this problem but I din`t try yet.

mrousavy, 2024-08-20:
Yea you need RN 0.73 or higher for the latest VisionCamera version. Also, make sure your Kotlin version matches the one I have in example app here.

closed 2024-08-20 by mrousavy

DeepakSharma-core, 2024-08-23:
I'm running 0.73.9 React Native version And Kotlin version is 1.9.0. Still I'm facing below error:

* What went wrong:
Execution failed for task ':app:mergeDebugGlobalSynthetics'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Failed to transform camera-extensions-1.4.0-rc01.aar (androidx.camera:camera-extensions:1.4.0-rc01) to match attributes {artifactType=android-global-synthetics, asm-transformed-variant=NONE, dexing-enable-desugaring=true, dexing-enable-jacoco-instrumentation=false, dexing-is-debuggable=true, dexing-min-sdk=28, org.gradle.category=library, org.gradle.dependency.bundling=external, org.gradle.libraryelements=aar, org.gradle.status=release, org.gradle.usage=java-runtime}.
      > Execution failed for DexingNoClasspathTransform: /Users/deepaksharma/.gradle/caches/transforms-3/965b48e571e90681b681e06c8d20a112/transformed/jetified-camera-extensions-1.4.0-rc01-runtime.jar.
         > Error while dexing.
   > Failed to transform camera-video-1.4.0-rc01.aar (androidx.camera:camera-video:1.4.0-rc01) to match attributes {artifactType=android-global-synthetics, asm-transformed-variant=NONE, dexing-enable-desugaring=true, dexing-enable-jacoco-instrumentation=false, dexing-is-debuggable=true, dexing-min-sdk=28, org.gradle.category=library, org.gradle.dependency.bundling=external, org.gradle.libraryelements=aar, org.gradle.status=release, org.gradle.usage=java-runtime}.
      > Execution failed for DexingNoClasspathTransform: /Users/deepaksharma/.gradle/caches/transforms-3/3b6e0915d86d56804a6648cb2e9682ba/transformed/jetified-camera-video-1.4.0-rc01-runtime.jar.

Please suggest me for making any changes.

jmazier-j2d, 2024-09-09:
> Yea you need RN 0.73 or higher for the latest VisionCamera version. Also, make sure your Kotlin version matches the one I have in example app here.

Could you please refer to this specific compatibility issue within the Readme or documentation ? (React native / react-native-vision-camera versions compatibility), that could be really useful ! I did switch to the 4.3.1 with my react native 0.72.17 and can confirm it does fix the issue (which make sense). Thanks for the great work by the way :)

danielchan303, 2025-01-02:
Please update the docs describing the RN version and library version compatibility, as well as need to specify the kotlin version in android/build.gradle

kotlinVersion = "1.9.22"

Spending hours until I find this, and another github issue

shengl711, 2025-03-08:
> Could you please refer to this specific compatibility issue within the Readme or documentation ? (React native / react-native-vision-camera versions compatibility), that could be really useful ! I [...]

@jmazier-j2d - Do you mind sharing after you install 4.3.1 version of react-native-vision-camera, did you add any other options other than below in gradle.properties for your app to work?  I am using RN 0.72.3 and Kotlin 1.8.20 but my app is either crashing or it failed when building react-native-vision-camera.  I find the documentation unclear as to any other setting or any other command to properly build this after installation. 

VisionCamera_enableCodeScanner=true
VisionCamera_enableFrameProcessors=false