margelo/react-native-vision-camera #3141: 💭 Camera format attribute supportsPhotoHdr is always 'false' on iPhone 12 (that does support HDR)

GreenCurveDev, 2024-08-17:
### Question

Hi,

Am writing an application (using vision-camera) which has a 'take photo' feature - I would like these images to be HDR if available but the camera format attributes (for all devices) is always 'false'.

I have followed the instructions for enabling HDR Video and Photo's in the VisionCamera documentation and for Video all is good. For photo's the images are always non-HDR even when manually setting the supportsPhotoHdr prop to true on the Camera.

Looking at the code for V4.5.1 (specifically **ios\Core\Extensions\AVCaptureDevice.Format+toDictionary.swift**) I see two functions that determine whether a device supports HDR Video and Photo. The function "supportsVideoHdr" is complete and works well but the equivalent function for photos, supportsPhotoHdr is annotated as 'TODO'.

var supportsPhotoHdr: Bool {    // TODO: Supports Photo HDR on iOS?    return false  }

Am posing this as a 'question' as I appreciate this may not be anywhere near the top of your priority list and would be happy to change and contribute. How can I modify the code to complete this function along the lines of...

var supportsPhotoHdr: Bool {
    let pixelFormat = CMFormatDescriptionGetMediaSubType(formatDescription)
    let hdrFormats = [
      // What formats should go here (Apple documentation not helpful in this respect)?
    ]
    return hdrFormats.contains(pixelFormat)
  }

And would this have the desired effect or am I missing other necessary changes?

Appreciate any advice you may have,

Thanks & regards,

Paul

### VisionCamera Version

V4.5.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

GreenCurveDev, 2024-08-17:
Hi Hans,

Appreciate your prompt response.

The XCode logs are not showing any errors to report/share.

The Vision Camera hook function call I used is const alldevices = useCameraDevices(); 

This call gives the results shown in the attached file where all 'supportsPhotoHdr' attributes are 'false'

[DeviceInformation.json](https://github.com/user-attachments/files/16644813/DeviceInformation.json)

At this point, the root cause would seem to be that the function used to determine whether there is Photo HDR support or not has yet to be implemented.

If this is not the case, and there are other code paths that determine whether photo HDR is supported, then confirmation of that would be helpful (though I cannot find those 'other' code paths).

Regards,

Paul

jonjamz, 2024-08-17:
I was actually just creating a similar bug report and saw yours, so I will report my experience here.

My iPhone XS supports HDR capture yet there are no formats on any of the devices that show support for HDR. I get the same empty result with my Pixel 6a. This goes for both photo and video HDR support.

```typescript
const allDevices = useCameraDevices();
useEffect(() => {
    const hdr = allDevices.map((device) => {
      return device?.formats.filter((fmt) => {
        return fmt.supportsPhotoHdr || fmt.supportsVideoHdr;
      });
    });
    console.log('DEVICES SUPPORTING HDR:', hdr);
}, [allDevices]);

// Logs: DEVICES SUPPORTING HDR: [[], [], [], [], []]
```

GreenCurveDev, 2024-08-17:
Hi, am glad it is not just me then.

@jonjamz - with regard to your specific iPhone issue - I don't think HDR video was supported until iPhone 12.
I am testing on an iPhone 12 and the HDR video seems to work fine (and is reported as 'supported' in the device format information. 

However, re: HDR photo's, there is hardly any information online regarding this feature (that I can find anyway) to the point it seems it may not be possible - ie it is a feature specific to the iOS native camera application and not available to 3rd party packages (someone please correct me if I am wrong about this). It may be the app developer needs to take 3 bracketed photo's and merge as described in the Vision documentation under section HDR.

The Apple documentation makes no mention of it in the avcapturephotosettings

https://developer.apple.com/documentation/avfoundation/avcapturephotosettings

And there are a couple of ancient posts on Reddit supporting this theory.

https://www.reddit.com/r/iOSProgramming/comments/778ypz/is_hdr_capture_a_thing_in_ios/

I am really hoping I am wrong about there being no Apple support for 3rd party HDR photo's - seems odd it is well supported for Video HDR.

In summary, I cannot see how to make this work from Apple docs (but I am not an experienced Swift developer) nor do I see its implementation in the Vision code - the CameraDeviceFormat.swift file shows at as being present in the serializable representation of the AVCaptureDevice.Format but even forcing this to 'true' (by rebuilding the package does not have any effect. If it did, then something 'ugly' could at least could be done via camera props.

In summary, at this point it would just be good to know (anyone) whether its possible or not irrespective of when/if it gets implemented.

Paul

GreenCurveDev, 2024-08-17:
Ok, after wasting a few hours on this I see the following post from last year 

https://github.com/mrousavy/react-native-vision-camera/issues/1473

So it seems definitely not supported and the "photoHdr" camera prop is just a placeholder?

jonjamz, 2024-08-17:
@GreenCurveDev thank you very much for the information. 

You are right, while the iPhone XS supports [HDR photos](https://support.apple.com/guide/iphone/adjust-hdr-camera-settings-iph2cafe2ebc/ios#:~:text=control%20HDR%20instead.-,On%20iPhone%C2%A0XS,-%2C%20iPhone%C2%A0XR), I do not see the HDR button when I switch to record a video.

It must be the same for Pixel 6a. The Ultra HDR setting mentions images but not videos.

jonjamz, 2024-08-17:
I checked out the issue you posted and followed the link to the AVFoundation docs about capturing a bracketed photo sequence. I copied the code from there into ChatGPT just to see what it would say about combining the sequence into an HDR image: https://chatgpt.com/share/9e558fc9-c892-4483-a56e-286ed78ff902

Apparently it is possible using CoreImage, but I have not verified the proposed code against the docs or tried it in a module yet.

GreenCurveDev, 2024-08-17:
@jonjamz Thanks - what a great idea and use of ChatGPT!

Will check it out.

mrousavy, 2024-08-27:
Hey - yea as mentioned in #1473, photo HDR is not yet supported in VisionCamera because it's far from easy to implement that, as you probably also noticed in the ChatGPT output. (I don't think that works as is, I think it needs a bit more code but I haven't tried).

There's also not a lot of resources in the internet, as not a lot of people have actually built custom photo HDR. I don't think any Camera library (not even for native iOS or native Android) does support photo HDR.

I'm closing this question in favor of keeping #1473 (a feature request) open, but I am not planning on implementing this in my own free time. Feel free to contact me for sponsored/consultancy work in which I can build this feature.

closed 2024-08-27 by mrousavy