margelo/react-native-vision-camera #3142: 💭 Is there a way to record a video together with an overlay?

VasilP202, 2024-08-19:
### Question

I'm using Skia Frame Processor to run pose detection on each frame and draw pose landmarks in real time. 
Is there a way to save the video along with the landmarks drawn over the video?

I'm using Camera's [startRecording(...)](https://react-native-vision-camera.com/docs/api/classes/Camera#startrecording) to record the video, but this only serves to save video frames, without any overlay.

### What I tried

I tried using **react-native-skia** / **react-native-video** to play video and draw landmarks separately (based on a timestamp of each pose). However, this causes significant desynchronization between the skeleton and the video. I still haven't found a way to access the video canvas at the exact timestamps. Therefore I want to record the video together with pose landmarks and play the video directly.

### VisionCamera Version

4.0.5

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

VasilP202, 2024-08-20:
How do I fill the issue template for queries?

mrousavy, 2024-08-27:
No, this is mentioned in the docs.

closed 2024-08-27 by mrousavy

mjhoracek, 2025-06-24:
Since it's almost a year later I think it might be fair to bump this again. Are there any examples for how this could be built? If not, what's a possible route to a solution? 

Thanks @mrousavy, your library is awesome.

mrousavy, 2025-06-25:
We build tailored solutions like this for our clients at [Margelo](https://margelo.com).

mjhoracek, 2025-06-25:
@VasilP202 did you ever figure out how to do this?