margelo/react-native-vision-camera #3143: 🐛 [Android] `takePhoto` is slow

jkaufman, 2024-08-20:
### What's happening?

`takePhoto` is slow. A minimal app executes `takePhoto` in ~1.40s on a Motorola Moto G Stylus 5G, 0.66s on Samsung Galaxy S21 5G, and 0.48s on a Google Pixel 3a. After performing extensive logging, we can see significant durations separating shutter button press, capture request creation, capture request fulfillment, and image save.

### Reproduceable Code

```tsx
A minimal reproduction case is available at [rnvc-playround/android-slow-capture-basic](https://github.com/sitewireco/rnvc-playground/tree/android-slow-capture-basic). Note the patch to RNVC that adds additional logging.
```

### Relevant log output

```shell
Logs and summary statistics drawn from `adb logcat`, `adb shell cmd media.camera watch`, and application console may be found in [/results](https://github.com/sitewireco/rnvc-playground/tree/android-slow-capture-basic/results). We reproduced the test on three different Android devices Google Pixel 3a, Motorola Moto G Stylus 5G, and Samsung Galaxy S21 5G.
```

### Camera Device

```json
The camera device is different across each of the Android handsets tested. The application logs at `results/app-output.txt` include the device name and selected format. Copied here:

Moto G

Device name: 0 (BACK) androidx.camera.camera2
      	Format: {
	  "videoStabilizationModes": [
	    "off",
	    "cinematic"
	  ],
	  "autoFocusSystem": "contrast-detection",
	  "photoWidth": 4080,
	  "supportsPhotoHdr": false,
	  "supportsDepthCapture": false,
	  "maxISO": 6400,
	  "minISO": 100,
	  "minFps": 10,
	  "videoWidth": 1920,
	  "supportsVideoHdr": false,
	  "videoHeight": 1080,
	  "fieldOfView": 75.04000877522147,
	  "maxFps": 30,
	  "photoHeight": 2296
	}

S21
      	Device name: 0 (BACK) androidx.camera.camera2
      	Format: {
	  "videoStabilizationModes": [
	    "off"
	  ],
	  "autoFocusSystem": "contrast-detection",
	  "photoWidth": 4032,
	  "supportsPhotoHdr": true,
	  "supportsDepthCapture": false,
	  "maxISO": 3200,
	  "minISO": 50,
	  "minFps": 7,
	  "videoWidth": 3840,
	  "supportsVideoHdr": false,
	  "videoHeight": 2160,
	  "fieldOfView": 80.0605180983776,
	  "maxFps": 30,
	  "photoHeight": 3024
	}

Pixel 3a
      	Device name: 0 (BACK) androidx.camera.camera2
      	Format: {
	  "videoStabilizationModes": [
	    "off",
	    "cinematic"
	  ],
	  "autoFocusSystem": "contrast-detection",
	  "photoWidth": 4032,
	  "supportsPhotoHdr": false,
	  "supportsDepthCapture": false,
	  "maxISO": 7111,
	  "minISO": 55,
	  "minFps": 7,
	  "videoWidth": 3840,
	  "supportsVideoHdr": false,
	  "videoHeight": 2160,
	  "fieldOfView": 76.94092946489485,
	  "maxFps": 30,
	  "photoHeight": 3024
	}
```

### Device

Google Pixel 3a, Motorola Moto G Stylus 5G, and Samsung Galaxy S21 5G

### VisionCamera Version

4.5.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-08-20:
Interesting. I guess we'd have to compare with a bare CameraX android app to see if VisionCamera does anything wrong?

mrousavy, 2024-08-20:
> 0.66s on a Samsung

I mean I wouldn't be surprised if this just takes really long on Android. Android is known for slower capture than iOS. 0.66s is pretty slow, so maybe this is under the highest quality and longest exposure options available, maybe some stabilization, quality prioritization or lower format resolution speeds this up.

AjaxVM, 2024-08-20:
We have tested this with a variety of configurations, and this is with flash off - if we lower quality to 720p, or set the balance to speed, it does speed up a bit, but is still very slow. Note that capturing logcat logs for the base camera is about 180ms on the slow end for full resolution capture, so this is much slower. It appears to be largely a delay between takePhoto starting and the callback executing for onCaptureStarted as a large part of the culprit, with the second delay (from onCaptureStarted to onImageSaved) being more related to resolution/settings.

jkaufman, 2024-08-20:
@mrousavy Definitely worth taking a peek at the linked minimal repro case, summary results (`brief.txt`) for each device, and raw logs. The delay from shutter press to capture start exceeds the entire capture duration in the native camera app on the S21. there may be an opportunity to improve performance with careful CameraX session management within RNVC.

mrousavy, 2024-08-20:
The native Samsung Camera app does not use CameraX.

If you remember, VisionCamera V3 didn't use CameraX either - it was Camera2 and it was much faster. Like, a lot faster.
But unfortunately it broke for like 20% of devices and I couldn't fix issues or quirks fast enough, so CameraX it is - but I still think CameraX is fast enough - maybe there's something wrong configured, or it's a new bug/regression in CameraX.

OnCaptureStarted is definitely the first event after precapture, and precapture does AF/AE/AWB triggers which are usually the parts that are really slow.
If AF/AE/AWB are already locked/focused/converged, it should be fast.

jkaufman, 2024-08-20:
Thanks for digging in. It's unsustainable to base a device-agnostic Android camera library on Camera2. No question that Samsung can achieve faster capture on a relatively small set of supported devices using Camera2, directly. I appreciate the move to CameraX and know it offers both greater reliability and more consistent quality.

The intended purpose of this ticket is to kickstart a conversation about what RNVC might do to improve on observed CameraX performance. Perhaps there's some sort of clever CameraX session management or AF/AE/AWB defaults that might help. Or maybe this is surfacing a bug in CameraX. In any case, we're finding that a 300ms delay between shutter tap and capture start exceeds what users expect, resulting in blurry images as they move on to framing their next photo.

jkaufman, 2024-08-21:
I ran a quick test using the Android project sample [CameraXBasic](https://github.com/android/camera-samples/tree/main/CameraXBasic). This is an outdated project running CameraX `1.1.0-beta01`, so take these results with a grain of salt. I captured and saved five consecutive photos with image analysis disabled in an average duration of 306.2ms.

jkaufman, 2024-08-21:
I repeated the test with [camerax-demo](https://github.com/robertlevonyan/camerax-demo), a modern sample project with more extensive use of CameraX APIs. The project was lightly modified it to the same CameraX version as the initial report, `1.4.0-beta02`, as well as log timings. I captured and saved five consecutive photos in an average duration of 294.6ms. Below is logcat output, including the timing of each `onCaptureStarted` and `onImageSaved` callback (tag `CAPTURE-TIMING`).

```
2024-08-20 20:37:55.197 24747-24747 ImageCapture            com.robertlevonyan.demo.camerax      D  takePictureInternal
2024-08-20 20:37:55.198 24747-24747 CameraOrientationUtil   com.robertlevonyan.demo.camerax      D  getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
2024-08-20 20:37:55.198 24747-24747 TakePictureManager      com.robertlevonyan.demo.camerax      D  Issue the next TakePictureRequest.
2024-08-20 20:37:55.241 24747-24784 ConvergenceUtils        com.robertlevonyan.demo.camerax      D  checkCaptureResult, AE=CONVERGED AF =PASSIVE_FOCUSED AWB=CONVERGED
2024-08-20 20:37:55.243 24747-24784 Camera2CameraImpl       com.robertlevonyan.demo.camerax      D  {Camera@cfc7967[id=0]} Issue capture request
2024-08-20 20:37:55.243 24747-24784 CaptureSession          com.robertlevonyan.demo.camerax      D  Issuing capture request.
2024-08-20 20:37:55.245 24747-24784 Camera2Cap...estBuilder com.robertlevonyan.demo.camerax      D  createCaptureRequest
2024-08-20 20:37:55.388 24747-24747 CAPTURE-TIMING          com.robertlevonyan.demo.camerax      E  onCaptureStarted: after 191ms
2024-08-20 20:37:55.414 24747-24747 TakePictureManager      com.robertlevonyan.demo.camerax      D  Issue the next TakePictureRequest.
2024-08-20 20:37:55.414 24747-24747 TakePictureManager      com.robertlevonyan.demo.camerax      D  No new request.
2024-08-20 20:37:55.550 24747-24747 CAPTURE-TIMING          com.robertlevonyan.demo.camerax      E  onImageSaved: after 353ms
2024-08-20 20:37:55.555 24747-24747 CameraXDemo             com.robertlevonyan.demo.camerax      D  Photo saved in content://media/external_primary/images/media/1000000114
2024-08-20 20:37:56.212 24747-24747 ImageCapture            com.robertlevonyan.demo.camerax      D  takePictureInternal
2024-08-20 20:37:56.214 24747-24747 CameraOrientationUtil   com.robertlevonyan.demo.camerax      D  getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
2024-08-20 20:37:56.214 24747-24747 TakePictureManager      com.robertlevonyan.demo.camerax      D  Issue the next TakePictureRequest.
2024-08-20 20:37:56.274 24747-24784 ConvergenceUtils        com.robertlevonyan.demo.camerax      D  checkCaptureResult, AE=CONVERGED AF =PASSIVE_FOCUSED AWB=CONVERGED
2024-08-20 20:37:56.278 24747-24784 Camera2CameraImpl       com.robertlevonyan.demo.camerax      D  {Camera@cfc7967[id=0]} Issue capture request
2024-08-20 20:37:56.278 24747-24784 CaptureSession          com.robertlevonyan.demo.camerax      D  Issuing capture request.
2024-08-20 20:37:56.281 24747-24784 Camera2Cap...estBuilder com.robertlevonyan.demo.camerax      D  createCaptureRequest
2024-08-20 20:37:56.420 24747-24747 CAPTURE-TIMING          com.robertlevonyan.demo.camerax      E  onCaptureStarted: after 208ms
2024-08-20 20:37:56.447 24747-24747 TakePictureManager      com.robertlevonyan.demo.camerax      D  Issue the next TakePictureRequest.
2024-08-20 20:37:56.447 24747-24747 TakePictureManager      com.robertlevonyan.demo.camerax      D  No new request.
2024-08-20 20:37:56.587 24747-24747 CAPTURE-TIMING          com.robertlevonyan.demo.camerax      E  onImageSaved: after 375ms
2024-08-20 20:37:56.591 24747-24747 CameraXDemo             com.robertlevonyan.demo.camerax      D  Photo saved in content://media/external_primary/images/media/1000000115
2024-08-20 20:37:57.254 24747-24747 ImageCapture            com.robertlevonyan.demo.camerax      D  takePictureInternal
2024-08-20 20:37:57.255 24747-24747 CameraOrientationUtil   com.robertlevonyan.demo.camerax      D  getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
2024-08-20 20:37:57.255 24747-24747 TakePictureManager      com.robertlevonyan.demo.camerax      D  Issue the next TakePictureRequest.
2024-08-20 20:37:57.308 24747-24779 ConvergenceUtils        com.robertlevonyan.demo.camerax      D  checkCaptureResult, AE=CONVERGED AF =PASSIVE_FOCUSED AWB=CONVERGED
2024-08-20 20:37:57.311 24747-24779 Camera2CameraImpl       com.robertlevonyan.demo.camerax      D  {Camera@cfc7967[id=0]} Issue capture request
2024-08-20 20:37:57.311 24747-24779 CaptureSession          com.robertlevonyan.demo.camerax      D  Issuing capture request.
2024-08-20 20:37:57.313 24747-24779 Camera2Cap...estBuilder com.robertlevonyan.demo.camerax      D  createCaptureRequest
2024-08-20 20:37:57.454 24747-24747 CAPTURE-TIMING          com.robertlevonyan.demo.camerax      E  onCaptureStarted: after 200ms
2024-08-20 20:37:57.480 24747-24747 TakePictureManager      com.robertlevonyan.demo.camerax      D  Issue the next TakePictureRequest.
2024-08-20 20:37:57.480 24747-24747 TakePictureManager      com.robertlevonyan.demo.camerax      D  No new request.
2024-08-20 20:37:57.609 24747-24747 CAPTURE-TIMING          com.robertlevonyan.demo.camerax      E  onImageSaved: after 355ms
[... cut ...]
2024-08-20 20:37:59.540 24747-24747 TakePictureManager      com.robertlevonyan.demo.camerax      D  Issue the next TakePictureRequest.
2024-08-20 20:37:59.541 24747-24747 TakePictureManager      com.robertlevonyan.demo.camerax      D  No new request.
2024-08-20 20:37:59.678 24747-24747 CAPTURE-TIMING          com.robertlevonyan.demo.camerax      E  onImageSaved: after 346ms
2024-08-20 20:37:59.683 24747-24747 CameraXDemo             com.robertlevonyan.demo.camerax      D  Photo saved in content://media/external_primary/images/media/1000000118
```

jkaufman, 2024-08-21:
I repeated the test with [camerax-demo](https://github.com/robertlevonyan/camerax-demo), a modern sample project with more extensive use of CameraX APIs. The project was [lightly modified](https://github.com/sitewireco/camerax-demo/tree/capture-timing) to use the same CameraX version as the initial report, `1.4.0-beta02`, as well as log timings. I captured and saved five consecutive photos in an average duration of **409.8ms**. Below is abbreviated logcat output, including the timing of each `onCaptureStarted` and `onImageSaved` callback (tag `CAPTURE-TIMING`).

In summary, the time to `onCaptureStarted` is 130ms longer, on average, under RNVC than in this sample project (using Pixel 3a). I'll need to dig in to the 3A settings to see if that ccounts for the difference. We'll also want to repeat these tests on the Samsung device since it exhibits the worst performance.

```
08-20 22:07:07.051 28990 28990 E CAPTURE-TIMING: onCaptureStarted: after 214ms
08-20 22:07:07.321 28990 28990 E CAPTURE-TIMING: onImageSaved: after 484ms
08-20 22:07:08.449 28990 28990 E CAPTURE-TIMING: onCaptureStarted: after 206ms
08-20 22:07:08.641 28990 28990 E CAPTURE-TIMING: onImageSaved: after 398ms
08-20 22:07:09.682 28990 28990 E CAPTURE-TIMING: onCaptureStarted: after 200ms
08-20 22:07:09.864 28990 28990 E CAPTURE-TIMING: onImageSaved: after 382ms
08-20 22:07:11.381 28990 28990 E CAPTURE-TIMING: onCaptureStarted: after 210ms
08-20 22:07:11.602 28990 28990 E CAPTURE-TIMING: onImageSaved: after 431ms
08-20 22:07:12.848 28990 28990 E CAPTURE-TIMING: onCaptureStarted: after 198ms
08-20 22:07:13.004 28990 28990 E CAPTURE-TIMING: onImageSaved: after 354ms
```

AjaxVM, 2024-08-21:
Using the sample app @jkaufman shared (capture-timing branch, with the change to use MINIMIZE_LATENCY), and tweaking settings a bit, I came up with these timings on the Samsung S21 device from before.
NOTE: the RNVC timings are the averages for that device. I am in the same location/quality of light, and the output images are the same dimensions.

Timings
  RNVC:
    onCaptureStarted: after 451ms
    onImageSaved: after 650ms (199ms)

  Preview on + min latency:
    onCaptureStarted: after 287ms
    onImageSaved: 777ms (490ms)

seba9999, 2024-10-04:
I'm also experiencing some delay taking picture with my Galaxy S20 ( around 0,8-1.2 sc ) 😞 

Does RNVC allow to change the jPeg quality ?

I'm not sur if I get it all... But it would be possible to add a parameter to choose the jpegQuality with [setJpegQuality](https://developer.android.com/reference/androidx/camera/core/ImageCapture.Builder#setJpegQuality(int))  right ? And set it even lower than 95 ( which is the value of CAPTURE_MODE_MINIMIZE_LATENCY )

Wouldn't it improve the delays ?

jkaufman, 2024-10-04:
That doesn’t represent much of the capture interval. Note the timings documented above.

thiagomachado1998, 2024-10-04:
I'm using the react native camera kit library and there I have similar difficulties, the photos are captured with a delay of 500-800ms, but on android I've made several modifications to the source code that have improved a little if one of these libraries solves the problem I will use it 

https://github.com/teslamotors/react-native-camera-kit/issues/674

seba9999, 2024-10-07:
Huum well, as far as I understood between a `onCaptureStarted` and a `onImageSaved` two things (at least) are happening. 

The picture is took
The picture is saved

For the second one, the image quality / size of the image would increase time to write it (save it) no ?
But I may be wrong and more than 90% of the time is from the first "event" : taking the shot ?

And what about the onCaptureStarted delay you show (~150ms) ? Is it linked to some AF/AE/AWB ? 

Sorry I dont get it all 😅

ismaelsousa, 2024-11-27:
I set up a native project with just the CameraX and these are the results

[image]

And I also tested it on the camera kit and it's the same time

That's the normal time using cameraX. Of course, this time, it is on my low-end device Moto G52 on a high-end it's faster

mrousavy, 2024-11-28:
@ismaelsousa how much is it in VisionCamera?

ismaelsousa, 2024-11-28:
pretty much the same time 1.5s on average @mrousavy

mrousavy, 2024-11-28:
cool, so it's not VisionCamera's fault. CameraX/Android is just slow.

sayan-avantai, 2025-03-21:
Any update regarding this? As I am taking multiple photos for bar code scan, it's very slow .