margelo/react-native-vision-camera #3144: [device/flash-not-available: The Camera Device does not have a flash unit! Make sure you select a device where `device.hasFlash`/`device.hasTorch` is true.]

chandanpradhanin, 2024-08-21:
### What's happening?

The back camera is working for me, but I am experiencing this issue when using the front camera to take selfies. I am using **getCameraDevice(devices, 'front')** to obtain the front camera.  I would appreciate your guidance in fixing this or letting me know if anything is incorrect. 

### Reproduceable Code

```tsx
const frontDevice = getCameraDevice(devices, 'front');
 <Camera
              ref={ref => {
                this.camera = ref;
              }}
              style={cameraStyle}
              device={frontDevice}
              isActive={isActive}
              photo={true}
              torch={"off"}
              enableZoomGesture={true}
            />
```

### Relevant log output

```shell
error [device/flash-not-available: The Camera Device does not have a flash unit! Make sure you select a device where `device.hasFlash`/`device.hasTorch` is true.]
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 4,
  "minZoom": 1,
  "maxExposure": 12,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 0,
  "minExposure": -24,
  "name": "1 (FRONT) androidx.camera.camera2",
  "hasFlash": false,
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Xiaomi Redmi Note 8 Pro - M1906G71

### VisionCamera Version

4.5.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

chandanpradhanin, 2024-08-21:
@mrousavy Thank you for your prompt response. As I do not intend to use a flash or torch while using the front camera, why is it still necessary that **device.hasFlash** and **device.hasTorch** be true? Also, I have used the **torch={"off"}** property, so why is this issue still occurring?

mrousavy, 2024-08-21:
If you don't have a device you shouldn't render the Camera. This is literally what the error message says. Use Typescript, you cannot pass null to device

chandanpradhanin, 2024-08-21:
@mrousavy i am not passing null to device prop. I am passing the which is return by getCameraDevice(devices, 'front') and camera view is opened in my app but takePhoto function is not working for me. So i have changed in the options flash: 'off' like below and now its working for me.

 const options = { base64: false, **flash: 'off'**, skipMetadata: true };

let response = await this.camera.takePhoto(options);

seanquinn, 2025-01-13:
Ran into this and burned some hours digging through to find this.

My issue was any references to `cameraRef` just silently failed on older android devices. Turns out, it was down to `flash` being set to `auto`. Worked fine in iOs and newer Android devices but not in some older devices.

The solution was to verify if the device supports flash before enabling flash: auto when capturing a photo. I was under the assumption `flash: auto` would have done this check.

whwz163, 2025-01-15:
请问解决了嘛 我也有这个问题

whwz163, 2025-01-15:
huawei 手机都不行

whwz163, 2025-01-15:
i test the huawei devices can't work

firefly-g, 2025-03-04:
我也是华为机不行 用react-native-torch灯光灯库 针对华为机做了单独处理， 其他机型下用vision-camera自带的 目前正常

wolever, 2025-04-26:
I am also experiencing this issue:

* With `react-native-vision-camera@4.6.3`
* On Android devices (including Pixel 6, Pixel 8 Pro, Pixel 9 Pro, and others)
* Using the "rear" camera
* Calling `takePhoto({ enableShutterSound: false, flash: 'off' })`

The `device/flash-not-available` error is raised.

LiamBateman, 2025-05-02:
Im encountering a similar issue, but it seems specifically with the back ultra wide camera, it works fine on the others. I've worked out to detect and pass false. 

I think the bahviour is correct, but in V2 you could pass true and it would ignore it if the device had no flash.