margelo/react-native-vision-camera #3146: 🐛 when frameprocessor is on app closes automatically

saransathish, 2024-08-22:
### What's happening?

`import React from 'react';
import { View } from 'react-native';
import { Camera, useCameraFormat, useFrameProcessor } from 'react-native-vision-camera';

interface FrameProcessingProps {
  device: any;
  model: any;
}

  const frameProcessor = useFrameProcessor((frame) => {
    'worklet'
    console.log(`Frame: ${frame.width}x${frame.height} (${frame.pixelFormat})`)
  }, [])

  return (
    <View style={{ flex: 1 }}>
      <Camera
        style={{ flex: 1 }}
        device={device}
        isActive={true}
        frameProcessor={frameProcessor}
      />
    </View>
  );
};

export default FrameProcessing;
`
In the above i uses the frameprocessor for some ml task but it wont work then i just used the simple console statement it also close the apps automatically pls any one help me to solve this issue.

### Reproduceable Code

```tsx
<Camera
        style={{ flex: 1 }}
        device={device}
        isActive={true}
        frameProcessor={frameProcessor}
      />
```

### Relevant log output

```shell
info Starting logkitty
[11:58:01] I | ReactNativeJS ▶︎ Running "newlearning" with {"rootTag":11}

[11:58:02] I | ReactNativeJS ▶︎ Loading react-native-worklets-core...

[11:58:02] I | ReactNativeJS ▶︎ Worklets loaded successfully

[11:58:02] I | ReactNativeJS ▶︎ Installing bindings...

[11:58:02] I | ReactNativeJS ▶︎ Successfully installed!

[11:58:02] I | ReactNativeJS ▶︎ Loading Tensorflow Lite Model 1

[11:58:02] I | ReactNativeJS ▶︎ Resolved Model path: http://localhost:8081/assets/src/assets/my-modal.tflite?platform=android&hash=a271e231db0296a6e9133ba046fd36a5

[11:58:02] I | ReactNativeJS ▶︎ Model loaded!
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

vivo t2 pro - android 14

### VisionCamera Version

4.5.2

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

parice02, 2024-08-22:
Same issue:

Error message:
```
Your app just crashed. See the error below.
com.facebook.jni.CppException: Regular javascript function 'rethrowErrorOnJS' cannot be shared. Try decorating the function with the 'worklet' keyword to allow the javascript function to be used as a worklet.
```

parice02, 2024-08-23:
Updating `react-native-worklets-core` to 1.3.3 fixes the problem.

mrousavy, 2024-08-27:
You probably don't have "processNestedWorklets" enabled in the reanimated babel plugin. You can also try to use it without reanimaed and only use worklets core.

closed 2024-08-27 by mrousavy

coofzilla, 2024-09-02:
> Updating `react-native-worklets-core` to 1.3.3 fixes the problem for me.

you saved me! had the exact error. thank you :)