margelo/react-native-vision-camera #3147: 🐛  Frame Drops in Recorded Video When Using useFrameProcessor,

rmethaniya, 2024-08-22:
### What's happening?

When the frame processor is disabled, the recorded video is smooth and free of frame drops. However, when the frame processor is enabled, the recorded video shows noticeable frame drops.

The issue occurs when using the `useFrameProcessor` to resize frames and play them every second with `runAtTargetFps`. Additionally, `runAsync` is used for frame processing, and frame detection is performed with `react-native-fast-tflite` using `runSync`.

### Reproduceable Code

```tsx
const format = getCameraFormat(device, [
    {fps: 120}, 
    {videoResolution: 'max'}
  ])

useFrameProcessor(
  (frame: Frame) => {
    'worklet'
    const resizes = resize(frame, {
      scale: {
        width: 224,
        height: 480
      },
      pixelFormat: 'rgb',
      dataType: 'uint8'
    })
    runAtTargetFps(1, () => {
      'worklet'
      runAsync(frame, () => {
        'worklet'
      
          const uint8 = Float32Array.from(resized, (value) => value / 255)
          const dict = model.runSync([uint8])[0]
         
      })
    })
  }
)
```

### Relevant log output

```shell
Installing bindings...
Successfully installed!
Looking up Frame Processor Plugin "resize"...
Frame Processor Plugin "resize" found! Initializing...
'--122------>', '{\n  "minFocusDistance": 15,\n  "formats": [],\n  "minZoom": 1,\n  "hasTorch": true,\n  "maxExposure": 8,\n  "supportsLowLightBoost": false,\n  "sensorOrientation": "landscape-right",\n  "isMultiCam": false,\n  "hardwareLevel": "full",\n  "position": "back",\n  "hasFlash": true,\n  "minExposure": -8,\n  "name": "Back Camera",\n  "maxZoom": 123.75,\n  "physicalDevices": [\n    "wide-angle-camera"\n  ],\n  "supportsRawCapture": false,\n  "neutralZoom": 1,\n  "supportsFocus": true,\n  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0"\n}'
Loading Tensorflow Lite Model 130
Resolved Model path: http://192.168.29.212:8081/assets/src/assets/detect_int8.tflite?platform=ios&hash=d9b3a7d2cdaab8be0c2b5d9ecd44c620
Fetching http://192.168.29.212:8081/assets/src/assets/detect_int8.tflite?platform=ios&hash=d9b3a7d2cdaab8be0c2b5d9ecd44c620...
19:02:18.580: [info] 📸 VisionCamera.didSetProps(_:): Updating 28 props: [onInitialized, cameraId, position, enableBufferCompression, preview, onCodeScanned, onStarted, top, borderWidth, right, isActive, pixelFormat, video, onViewReady, onError, onStopped, overflow, enableFrameProcessor, enableZoomGesture, format, left, audio, bottom, borderRadius, zoom, photo, borderColor, onShutter]
19:02:18.580: [info] 📸 VisionCamera.configure(_:): configure { ... }: Waiting for lock...
19:02:18.583: [info] 📸 VisionCamera.configure(_:): configure { ... }: Updating CameraSession Configuration... Difference(inputChanged: true, outputsChanged: true, videoStabilizationChanged: true, orientationChanged: true, formatChanged: true, sidePropsChanged: true, torchChanged: true, zoomChanged: true, exposureChanged: true, audioSessionChanged: true, locationChanged: true)
19:02:18.583: [info] 📸 VisionCamera.configureDevice(configuration:): Configuring Input Device...
19:02:18.583: [info] 📸 VisionCamera.configureDevice(configuration:): Configuring Camera com.apple.avfoundation.avcapturedevice.built-in_video:0...
19:02:18.587: [info] 📸 VisionCamera.configureDevice(configuration:): Successfully configured Input Device!
19:02:18.587: [info] 📸 VisionCamera.configureOutputs(configuration:): Configuring Outputs...
19:02:18.587: [info] 📸 VisionCamera.configureOutputs(configuration:): Adding Photo output...
19:02:18.588: [info] 📸 VisionCamera.configureOutputs(configuration:): Adding Video Data output...
19:02:18.589: [info] 📸 VisionCamera.configureOutputs(configuration:): Successfully configured all outputs!
19:02:18.589: [info] 📸 VisionCamera.configureFormat(configuration:device:): Configuring Format (1920x1080 | 1920x1080@120.0 (ISO: 34.0..1360.0))...
19:02:18.590: [info] 📸 VisionCamera.configureFormat(configuration:device:): Successfully configured Format!
19:02:18.592: [info] 📸 VisionCamera.getPixelFormat(for:): Available Pixel Formats: ["420v", "420f", "BGRA", "&8v0", "-8v0", "&8f0", "-8f0", "&BGA", "-BGA"], finding best match... (pixelFormat="yuv", enableHdr={false}, enableBufferCompression={false})
19:02:18.592: [info] 📸 VisionCamera.getPixelFormat(for:): Using PixelFormat: 420f...
19:02:18.842: [info] 📸 VisionCamera.onCameraStarted(): Camera started!
19:02:18.842: [info] 📸 VisionCamera.onSessionInitialized(): Camera initialized!
19:02:18.842: [info] 📸 VisionCamera.configure(_:): Beginning AudioSession configuration...
19:02:18.842: [info] 📸 VisionCamera.configureAudioSession(configuration:): Configuring Audio Session...
19:02:18.844: [info] 📸 VisionCamera.configure(_:): Committed AudioSession configuration!
19:02:18.846: [info] 📸 VisionCamera.configure(_:): Beginning Location Output configuration...
[... 471 lines cut ...]
  "hasFlash": true,
  "name": "Back Camera",
  "minExposure": -8
}
```

### Device

iPhone 14 (iOS 17.5.1)

### VisionCamera Version

4.0.5

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rmethaniya, 2024-08-28:
@mrousavy  I've attached the xCode logs. Please let me know if you need any more information.

mrousavy, 2024-08-28:
Well, resizing a `max` resolution frame might just take a while. I'm actually surprised that this even works, I think resizing should be handled inside `runAsync`.

closed 2024-08-28 by mrousavy

mrousavy, 2024-08-29:
Hm that's weird. I remember this working back when I tested it. As of right now, I don't have any free time to investigate this though.

justrustc, 2025-05-24:
@rmethaniya Have you solved this problem?