margelo/react-native-vision-camera #3148: getCameraCharacteristics:1041: Unable to retrieve camera characteristics for unknown device

stevenmathers, 2024-08-23:
### What's happening?

crash on specific android device S22+

I dont have this device, I am seeing this in our firebase crashlytics

[image]

### Reproduceable Code

```tsx
<Camera
                    ref={camera}
                    style={{flex:1}}
                    device={device}
                    isActive={isFocused && isForeground}
                    photo={true}
                    enableZoomGesture
                    orientation='portrait'
                    resizeMode={'contain'}
                />
```

### Relevant log output

```shell
see image above
```

### Camera Device

```json
unknown
```

### Device

Galaxy S22+

### VisionCamera Version

4.1.5

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-08-27:
Sorry, without logs I cannot help you. You need to reproduce this bug on an actual device and send me the adb logcat logs.

closed 2024-08-27 by mrousavy

stevenmathers, 2024-08-29:
Hi @mrousavy perhaps just keep this ticket open in case it is not just how Im using the library, but an issue for everybody who uses the library with this device.    Its been around for 2 years though.

mrousavy, 2024-08-29:
Yes but as I just said, there's nothing I can do without logs.

stevenmathers, 2024-08-29:
Ill try to track down an online real device of this type and see what
happens.