margelo/react-native-vision-camera #3151: 💭 Access photo library from within the camera

dbarros, 2024-08-24:
### Question

When showing the camera view to take a photo, is there a built-in prop that shows a button to be able to open the photo library from within the camera? In other words, sometimes you start off thinking you want to take a photo, or you do take one, but then want to go to the photo library and pick another photo.

### What I tried

I looked through the documentation but could not find anything applicable.

### VisionCamera Version

4.5.2

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-08-27:
No. This is a Camera library, not a photo gallery library.

closed 2024-08-27 by mrousavy

dbarros, 2024-08-28:
> No. This is a Camera library, not a photo gallery library.

I understand that. I wasn't looking for photo gallery functionality, instead simply asking if the camera library was able to show a button that would only take the user to the photo library. I'll do as the bot suggested and add the button manually. Thanks nonetheless.

mrousavy, 2024-08-28:
Yep, this is definitely something you'd need to build ontop of that. The Camera itself has no UI.