margelo/react-native-vision-camera #3152: ✨ Add possibility to cleanup resources when camera is being unmounted in frame processor plugin

AlexShukel, 2024-08-24:
### What feature or enhancement are you suggesting?

Currently, if frame processor plugin registers some listeners in it's constructor (for example, `OrientationEventListener` or any kind of sensors), there is no way to unregister those listeners when camera component unmounts.

The only way to unregister listeners is using `onHostPause` and `onHostDestroy` from `LifecycleEventListener`, but in that case listeners will drain battery even if user closed camera activity.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

It is possible to register native module that will be able to cleanup resources, and call the native method from JS in `useEffect` when camera is being unmounted.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2024-08-27:
Hm, it's hard to add such lifecycle listeners because the Camera has no connection or knowledge of any of it's plugins.

You can add that code in the `deinit`/`dealloc` methods on iOS, and on `finalize` on Android, but it will only be called once the JS object (instance of the plugin) will actually be destroyed by the garbage collector..

AlexShukel, 2024-08-27:
@mrousavy I think it's okey to manually cleanup resources from react's `useEffect` hook in Camera component, but it is also not fully functional workaround.

The problem occurs when users leaves application in background - then it would be nice to cleanup all listeners in order to prevent battery drain. So, if you clean listeners in `onHostPause` method of `com.facebook.react.bridge.LifecycleEventListener`, then you should register them in `onHostResume`. But, you shouldn't register listeners on `resume` if camera is not mounted. However, it is not possible to determine whether camera is mounted or not from plugin side.

Maybe it is possible to add some kind of lifecycle indicator in `VisionCameraProxy`?

mrousavy, 2024-08-27:
Good idea, yes we can add something like `addLifecycleListener` to VisionCameraProxy!

mrousavy, 2024-08-27:
Also, Nitro Modules will have proper memory management and can be eagerly released - but that is still not public