margelo/react-native-vision-camera #3156: 🐛 Incorrect Barcode Scanning Results for CODE-128

wellhenrique, 2024-08-26:
### What's happening?

When using react-native-vision-camera to scan a CODE-128 barcode on an Android device held in a portrait orientation, the scanned values are returned incorrectly. For example, a barcode with the value 11111111111111111111111.... is incorrectly recognized as 11134231111122313111111... 
This issue does not occur when the device is held landscape mode. This behavior has been consistently reproduced on multiple Android devices. Does anyone have any idea what might be causing this problem?

### Reproduceable Code

```tsx
My example project: https://github.com/wellhenrique/issue-react-native-vision-camera
Barcode used in testing: https://github.com/wellhenrique/issue-react-native-vision-camera/blob/main/barcode.jpg
Website where the barcode was generated: https://barcode.tec-it.com/en/?data=11111111111111111111111111111111111111111111
```

### Relevant log output

```shell
08-26 11:38:28.394  1069  8015 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
08-26 11:38:28.394  1069  8015 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
08-26 11:38:28.395  1069  8015 D vendor.qti.bluetooth@1.0-wake_lock: Acquire wakelock is acquired 
08-26 11:38:28.395  1069  8015 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
08-26 11:38:28.396  1078  1212 D SDM     : DisplayBase::BuildLayerStackStats: LayerStack layer_count: 5, app_layer_count: 4, gpu_target_index: 4, display type: 0
08-26 11:38:28.414  1070 19176 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-26 11:38:28.419  1239  1423 V Camera2ClientBase: notifyShutter: Shutter notification for request id 0 at time 4449356475439
08-26 11:38:28.429  1078  1212 D SDM     : DisplayBase::BuildLayerStackStats: LayerStack layer_count: 5, app_layer_count: 4, gpu_target_index: 4, display type: 0
08-26 11:38:28.448  1078  1212 D SDM     : DisplayBase::BuildLayerStackStats: LayerStack layer_count: 5, app_layer_count: 4, gpu_target_index: 4, display type: 0
08-26 11:38:28.448  1070 19176 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-26 11:38:28.451  1239  1883 V Camera2ClientBase: notifyShutter: Shutter notification for request id 0 at time 4449389876439
08-26 11:38:28.464  1078  1212 D SDM     : DisplayBase::BuildLayerStackStats: LayerStack layer_count: 5, app_layer_count: 4, gpu_target_index: 4, display type: 0
08-26 11:38:28.465  1069  8015 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
08-26 11:38:28.465  1069  8015 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
08-26 11:38:28.481  1070 19176 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-26 11:38:28.483  1239  1883 V Camera2ClientBase: notifyShutter: Shutter notification for request id 0 at time 4449423277439
08-26 11:38:28.497  1078  1212 D SDM     : DisplayBase::BuildLayerStackStats: LayerStack layer_count: 5, app_layer_count: 4, gpu_target_index: 4, display type: 0
08-26 11:38:28.516  1070 19176 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-26 11:38:28.518  1239  1883 V Camera2ClientBase: notifyShutter: Shutter notification for request id 0 at time 4449456677439
08-26 11:38:28.528  1069  8015 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
08-26 11:38:28.528  1069  8015 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
08-26 11:38:28.528  1069  8015 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
08-26 11:38:28.548  1078  1212 D SDM     : DisplayBase::BuildLayerStackStats: LayerStack layer_count: 5, app_layer_count: 4, gpu_target_index: 4, display type: 0
08-26 11:38:28.551  1070 19176 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-26 11:38:28.557  1239  1423 V Camera2ClientBase: notifyShutter: Shutter notification for request id 0 at time 4449490087439
08-26 11:38:28.564  1078  1212 D SDM     : DisplayBase::BuildLayerStackStats: LayerStack layer_count: 5, app_layer_count: 4, gpu_target_index: 4, display type: 0
08-26 11:38:28.569  1069  8015 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
08-26 11:38:28.569  1069  8015 D vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
08-26 11:38:28.583  1070 19176 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-26 11:38:28.586  1239  1423 V Camera2ClientBase: notifyShutter: Shutter notification for request id 0 at time 4449523484439
08-26 11:38:28.597  1078  1212 D SDM     : DisplayBase::BuildLayerStackStats: LayerStack layer_count: 5, app_layer_count: 4, gpu_target_index: 4, display type: 0
08-26 11:38:28.622  1070 19176 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-26 11:38:28.627  1239  1423 V Camera2ClientBase: notifyShutter: Shutter notification for request id 0 at time 4449556897439
08-26 11:38:28.648  1078  1212 D SDM     : DisplayBase::BuildLayerStackStats: LayerStack layer_count: 5, app_layer_count: 4, gpu_target_index: 4, display type: 0
08-26 11:38:28.655  1070 19176 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-26 11:38:28.658  1239  1423 V Camera2ClientBase: notifyShutter: Shutter notification for request id 0 at time 4449590295439
08-26 11:38:28.665  1078  1212 D SDM     : DisplayBase::BuildLayerStackStats: LayerStack layer_count: 5, app_layer_count: 4, gpu_target_index: 4, display type: 0
08-26 11:38:28.686  1070 19176 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-26 11:38:28.690  1239  1423 V Camera2ClientBase: notifyShutter: Shutter notification for request id 0 at time 4449623685439
08-26 11:38:28.697  1078  1212 D SDM     : DisplayBase::BuildLayerStackStats: LayerStack layer_count: 5, app_layer_count: 4, gpu_target_index: 4, display type: 0
08-26 11:38:28.717  1070 19176 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-26 11:38:28.719  1069  7942 D vendor.qti.bluetooth@1.0-wake_lock: Release wakelock is released 
08-26 11:38:28.722  3532  4235 I BistoHotwordHelper: (REDACTED) getHotwordActive::active query: %s, watch: %s, devices connected: %s
08-26 11:38:28.724  1239  1883 V Camera2ClientBase: notifyShutter: Shutter notification for request id 0 at time 4449657083439
08-26 11:38:28.731  1078  1212 D SDM     : DisplayBase::BuildLayerStackStats: LayerStack layer_count: 5, app_layer_count: 4, gpu_target_index: 4, display type: 0
08-26 11:38:28.752  1070 19176 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-26 11:38:28.757  1239  1883 V Camera2ClientBase: notifyShutter: Shutter notification for request id 0 at time 4449690504439
08-26 11:38:28.765  1078  1212 D SDM     : DisplayBase::BuildLayerStackStats: LayerStack layer_count: 5, app_layer_count: 4, gpu_target_index: 4, display type: 0
08-26 11:38:28.784  1070 19176 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-26 11:38:28.788  1239  1423 V Camera2ClientBase: notifyShutter: Shutter notification for request id 0 at time 4449723886439
08-26 11:38:28.797  1078  1212 D SDM     : DisplayBase::BuildLayerStackStats: LayerStack layer_count: 5, app_layer_count: 4, gpu_target_index: 4, display type: 0
08-26 11:38:28.820  1070 19176 E mm-camera: <MCT   ><ERROR> 1355: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr NULL
08-26 11:38:28.824  1239  1423 V Camera2ClientBase: notifyShutter: Shutter notification for request id 0 at time 4449757299439
08-26 11:38:28.832  1078  1212 D SDM     : DisplayBase::BuildLayerStackStats: LayerStack layer_count: 5, app_layer_count: 4, gpu_target_index: 4, display type: 0
[... 314 lines cut ...]
    "position": "front",
    "id": "1"
  }
]
```

### Device

moto X4

### VisionCamera Version

4.5.1

### Can you reproduce this issue in the VisionCamera Example app?

[... cut ...]
### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

zzz08900, 2024-08-29:
Try adjusting your camera settings. Your scan result is unstable which usually means you are feeding MLKit with subpar images.

I tried your barcode with our RNVC-based code scanner and I'm getting consistent all 1s.

xulihang, 2024-09-19:
You may have to check the results multiple times to have a steady result or use an advanced library like this one: https://github.com/tony-xlh/vision-camera-dynamsoft-barcode-reader/

EmanuilM, 2025-01-17:
@wellhenrique  Did you resolve the issue?