margelo/react-native-vision-camera #3164: 🐛 Cannot scan large Aztec code on Android

mszczotarz, 2024-08-30:
### What's happening?

I cannot scan large Aztec code (example provided bellow) on Android. iOS works fine and worked on android before upgrade react-native-vision-camera from `3.5.1` to `4.5.2`.

It work's for smaller Aztec without any problems, so i don't think this is some kind of configuration issue.

Working (smaller) Aztec code:
[image]

On Google ML Kit website https://developers.google.com/ml-kit/vision/barcode-scanning/android#performance-tips it's mentioned:

> If you use the [CameraX](https://developer.android.com/jetpack/androidx/releases/camera) API, be sure that backpressure strategy is set to its default value [ImageAnalysis.STRATEGY_KEEP_ONLY_LATEST](https://developer.android.com/reference/androidx/camera/core/ImageAnalysis#STRATEGY_KEEP_ONLY_LATEST). This guarantees only one image will be delivered for analysis at a time. If more images are produced when the analyzer is busy, they will be dropped automatically and not queued for delivery. Once the image being analyzed is closed by calling ImageProxy.close(), the next latest image will be delivered.

Maybe this is some hint, but if I understand correctly native code, strategy is not changed for scanner...

This Aztec code doesn't work (works on android with react-native-vision-camera 3.5.1:

[image]

P.S I checked on example app and the issue is the same (changed scanner config to `aztec`).

### Reproduceable Code

```tsx
const device = useCameraDevice('back', {
    physicalDevices: [
      'ultra-wide-angle-camera',
      'wide-angle-camera',
      'telephoto-camera',
    ],
  });

const onCodeScanned = useCallback(async (codes: Code[]) => {
    console.log(codes);
  }, []);

<Camera
            {...props
            device={device}
            codeScanner={codeScanner}
            isActive={aztecCode ? false : isActive}
          />
```

### Relevant log output

```shell
No logs produced
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung Galaxy A53 5G

### VisionCamera Version

4.5.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mszczotarz, 2024-08-30:
@mrousavy logs added

mrousavy, 2024-09-12:
If you have some custom use-cases, please create your own code scanner plugin and run it in a Frame Processor. This gives you more control, e.g. via tweaking resolution or accuracy. 

The integrated code scanner is a lightweight and simple codescanner, which uses device native features.

closed 2024-09-12 by mrousavy