margelo/react-native-vision-camera #3165: 💭

moloudayat, 2024-08-31:
### Question

my camera doesn't cover the entier screen.

### What I tried

I want to have a screen that can scan the barcode. Unfortunately, my camera only shows in a square box and the other part of the screen is black. this is my code:

```
const BarcodeScanner: React.FC<Navigation> = props => {

   const device = useCameraDevice('back')
   const codeScanner = useCodeScanner({
        codeTypes: [
            'code-128'
            , 'code-39'
            , 'code-93'
            , 'codabar'
            , 'ean-13'
            , 'ean-8'
            , 'itf'
            , 'upc-e'
            , 'upc-a'
            , 'qr'
            , 'pdf-417'
            , 'aztec'
            , 'data-matrix'

        ],
        onCodeScanned: (codes: Code[], frame: CodeScannerFrame) => {
            if (codes.length > 0) {
                onSelect(codes[0].value)
                navigation.goBack()
            }
        }
    });

    return device !== null ? (
        <StandalonePage
            title={t("common.barcodeScanner")}
            onBack={handleBackPress}
            navigation={props.navigation}>
            <Camera
                style={StyleSheet.absoluteFill}
                device={device}
                isActive={true}
                codeScanner={codeScanner}
                torch={isFlashOn ? "on" : "off"}/>
            <View style={styles.topRight} />
            <View style={styles.topLeft} />
            <View style={styles.bottomRight} />
            <View style={styles.bottomLeft} />
            <TouchableOpacity
                style={styles.imgContainer}
                onPress={toggleFlash}>
                <Image
                    style={styles.img}
                    source={isFlashOn ? FlashOff : FlashOn} />
            </TouchableOpacity>
        </StandalonePage>
    )
        : null
}
```

and this is what happened I used **resizeMode="cover"** but the result didn't change.

[image]

### VisionCamera Version

4.5.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

moloudayat, 2024-08-31:
How can I adjust the dimensions in my Camera component and check for aspect ratio?

mrousavy, 2024-09-02:
No title.

closed 2024-09-02 by mrousavy