margelo/react-native-vision-camera #3167: 💭 How to set a React state from a Frame processor?

FrancoisBck, 2024-09-02:
### Question

This was already asked here, but I didn't really find my answer: [#2731](https://github.com/mrousavy/react-native-vision-camera/issues/2731)

I use Frame Processor to run a `.tflite `model like this:

```.tsx
...
const [lastModelResult, setLastModelResult] = useState();

const objectDetection = useTensorflowModel(require("model_09_04.tflite")) // Load the model from the assets folder
const model = objectDetection.state === "loaded" ? objectDetection.model : undefined

const frameProcessor = useFrameProcessor((frame) => {
    'worklet'
    if (model == null) return
    const resized = resize(frame, {
        scale: {width: 320, height: 320},
        pixelFormat: 'rgb',
        dataType: 'uint8',
    })
    const result = model.runSync([resized])
    setLastModelResult(result) // => this cause error
},[model])
```

I need to retrieve the last result from the model to perform additional tasks in another function, such as appending the result to a list and analyzing the list.

### What I tried

```.tsx
import { useRunOnJS } from 'react-native-worklets-core';

const updateResults = useRunOnJS((results) => {
    setLastModelResult(results);
},[]);

useEffect(() => {
    if (lastModelResult)
        console.log(lastModelResult);
    }
}, [lastModelResult]);

const frameProcessor = useFrameProcessor((frame) => {
    'worklet'
    if (model == null) return
    const resized = resize(frame, {
        scale: {width: 320, height: 320},
        pixelFormat: 'rgb',
        dataType: 'uint8',
    })
    const result = model.runSync([resized])
    updateResults(result)
},[model])
 
```

Is there a better solution ?

### VisionCamera Version

4.5.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

stevengoldberg, 2024-09-05:
> I need to retrieve the last result from the model to perform additional tasks in another function

You don't want to call setState on every frame, but you can update a ref, which doesn't cause a re-render. This is what I do in my app:

```
const updateLabelResults = useRunOnJS((results) => {
        labelerResultsRef.current = results
    }, [])
    const frameProcessor = useFrameProcessor(
        (frame) => {
            'worklet'
            runAtTargetFps(3, () => {
                const data = scanImage(frame)
                if (previewImageUri === null) {
                    updateLabelResults(data)
                }
            })
        },
        [previewImageUri]
    )
```

When some other state change is triggered, I stop the camera and read from labelerResultsRef.current.

mrousavy, 2024-09-08:
> Is there a better solution ?

Why? The code above works perfectly and is as efficient as a JS thread hop can be. This is the solution, you can also use useRef, useSharedValue or other things instead of state.

closed 2024-09-08 by mrousavy