margelo/react-native-vision-camera #3168: 🐛 frameProcessor and codeScanner are not working together on android

raslankiwan, 2024-09-03:
### What's happening?

I'm using codeScanner to detect QR codes, and skiaFrameProcessor to draw green box over the detected QR codes. This is working perfectly on iOS, but when adding both codeScanner and frameProcessor properties on android the camera output isn't shown (black screen). When disable any of them the feed is back again.

### Reproduceable Code

```tsx
const codeScanner = useCodeScanner({
      codeTypes: barCodeTypes,
      onCodeScanned: (codes, frame) => {
        
        setState({
          activeBarcodes: codes,
          frameWidth: frame.width,
          frameHeight: frame.height
        });
          onComplete(codes);

      }
    });

  const frameProcessor = useSkiaFrameProcessor(
      async (frame) => {
        'worklet';
        frame.render();
        for (const barcode of activeBarcodes) {
          const { corners } = barcode;

          const paint = Skia.Paint();
          const points = corners.map((corner) => Skia.Point(corner.x, corner.y));
          const poly = Skia.Path.Make();
          poly.addPoly(points, true);
          const color = barcode.status.success ? 'green' : Colors.danger;
          paint.setColor(Skia.Color(color));
          paint.setStyle(PaintStyle.Fill);
          paint.setAlphaf(0.7);
          frame.drawPath(poly, paint);
        }
      },
      [activeBarcodes]
    );

 <Camera
              style={StyleSheet.absoluteFill}
              device={device}
              codeScanner={codeScanner}
              frameProcessor={ frameProcessor}
            />
```

### Relevant log output

```shell
Camera mount error {"name":"unknown/unknown","_code":"unknown/unknown","_message":"androidx.camera.core.internal.CameraUseCaseAdapter$CameraException: java.lang.IllegalArgumentException: No supported surface combination is found for camera device - Id : 2.  May be attempting to bind too many use cases. Existing surfaces: [] New configs: [androidx.camera.video.impl.VideoCaptureConfig@96f06ef, androidx.camera.core.impl.ImageAnalysisConfig@43cf5fc, androidx.camera.core.impl.ImageAnalysisConfig@dcfb485]","_cause":{"cause":{"cause":{"stacktrace":"java.lang.IllegalArgumentException: No supported surface combination is found for camera device - Id : 2.  May be attempting to bind too many use cases. Existing surfaces: [] New configs: [androidx.camera.video.impl.VideoCaptureConfig@96f06ef, androidx.camera.core.impl.ImageAnalysisConfig@43cf5fc, androidx.camera.core.impl.ImageAnalysisConfig@dcfb485]\n\tat androidx.camera.camera2.internal.SupportedSurfaceCombination.getSuggestedStreamSpecifications(SupportedSurfaceCombination.java:598)\n\tat androidx.camera.camera2.internal.Camera2DeviceSurfaceManager.getSuggestedStreamSpecs(Camera2DeviceSurfaceManager.java:185)\n\tat androidx.camera.core.internal.CameraUseCaseAdapter.calculateSuggestedStreamSpecs(CameraUseCaseAdapter.java:750)\n\tat androidx.camera.core.internal.CameraUseCaseAdapter.updateUseCases(CameraUseCaseAdapter.java:348)\n\tat androidx.camera.core.internal.CameraUseCaseAdapter.updateUseCases(CameraUseCaseAdapter.java:365)\n\tat androidx.camera.core.internal.CameraUseCaseAdapter.updateUseCases(CameraUseCaseAdapter.java:299)\n\tat androidx.camera.core.internal.CameraUseCaseAdapter.addUseCases(CameraUseCaseAdapter.java:277)\n\tat androidx.camera.lifecycle.LifecycleCamera.bind(LifecycleCamera.java:201)\n\tat androidx.camera.lifecycle.LifecycleCameraRepository.bindToLifecycleCamera(LifecycleCameraRepository.java:302)\n\tat androidx.camera.lifecycle.ProcessCameraProvider.bindToLifecycle$camera_lifecycle_release(ProcessCameraProvider.kt:526)\n\tat androidx.camera.lifecycle.ProcessCameraProvider.bindToLifecycle(ProcessCameraProvider.kt:206)\n\tat com.mrousavy.camera.core.CameraSession_ConfigurationKt.configureCamera(CameraSession+Configuration.kt:265)\n\tat com.mrousavy.camera.core.CameraSession.configure(CameraSession.kt:141)\n\tat com.mrousavy.camera.react.CameraView$update$1.invokeSuspend(CameraView.kt:155)\n\tat kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)\n\tat kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:104)\n\tat android.os.Handler.handleCallback(Handler.java:958)\n\tat android.os.Handler.dispatchMessage(Handler.java:99)\n\tat android.os.Looper.loopOnce(Looper.java:230)\n\tat android.os.Looper.loop(Looper.java:319)\n\tat android.app.ActivityThread.main(ActivityThread.java:8919)\n\tat java.lang.reflect.Method.invoke(Native Method)\n\tat com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:578)\n\tat com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1103)\n","message":"No supported surface combination is found for camera device - Id : 2.  May be attempting to bind too many use cases. Existing surfaces: [] New configs: [androidx.camera.video.impl.VideoCaptureConfig@96f06ef, androidx.camera.core.impl.ImageAnalysisConfig@43cf5fc, androidx.camera.core.impl.ImageAnalysisConfig@dcfb485]"},"stacktrace":"androidx.camera.core.internal.CameraUseCaseAdapter$CameraException: java.lang.IllegalArgumentException: No supported surface combination is found for camera device - Id : 2.  May be attempting to bind too many use cases. Existing surfaces: [] New configs: [androidx.camera.video.impl.VideoCaptureConfig@96f06ef, androidx.camera.core.impl.ImageAnalysisConfig@43cf5fc, androidx.camera.core.impl.ImageAnalysisConfig@dcfb485]\n\tat androidx.camera.core.internal.CameraUseCaseAdapter.addUseCases(CameraUseCaseAdapter.java:279)\n\tat androidx.camera.lifecycle.LifecycleCamera.bind(LifecycleCamera.java:201)\n\tat androidx.camera.lifecycle.LifecycleCameraRepository.bindToLifecycleCamera(LifecycleCameraRepository.java:302)\n\tat androidx.camera.lifecycle.ProcessCameraProvider.bindToLifecycle$camera_lifecycle_release(ProcessCameraProvider.kt:526)\n\tat androidx.camera.lifecycle.ProcessCameraProvider.bindToLifecycle(ProcessCameraProvider.kt:206)\n\tat com.mrousavy.camera.core.CameraSession_ConfigurationKt.configureCamera(CameraSession+Configuration.kt:265)\n\tat com.mrousavy.camera.core.CameraSession.configure(CameraSession.kt:141)\n\tat com.mrousavy.camera.react.CameraView$update$1.invokeSuspend(CameraView.kt:155)\n\tat kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)\n\tat kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:104)\n\tat android.os.Handler.handleCallback(Handler.java:958)\n\tat android.os.Handler.dispatchMessage(Handler.java:99)\n\tat android.os.Looper.loopOnce(Looper.java:230)\n\tat android.os.Looper.loop(Looper.java:319)\n\tat android.app.ActivityThread.main(ActivityThread.java:8919)\n\tat java.lang.reflect.Method.invoke(Native Method)\n\tat com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:578)\n\tat com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1103)\nCaused by: java.lang.IllegalArgumentException: No supported surface combination is found for camera device - Id : 2.  May be attempting to bind too many use cases. Existing surfaces: [] New configs: [androidx.camera.video.impl.VideoCaptureConfig@96f06ef, androidx.camera.core.impl.ImageAnalysisConfig@43cf5fc, androidx.camera.core.impl.ImageAnalysisConfig@dcfb485]\n\tat androidx.camera.camera2.internal.SupportedSurfaceCombination.getSuggestedStreamSpecifications(SupportedSurfaceCombination.java:598)\n\tat androidx.camera.camera2.internal.Camera2DeviceSurfaceManager.getSuggestedStreamSpecs(Camera2DeviceSurfaceManager.java:185)\n\tat androidx.camera.core.internal.CameraUseCaseAdapter.calculateSuggestedStreamSpecs(CameraUseCaseAdapter.java:750)\n\tat androidx.camera.core.internal.CameraUseCaseAdapter.updateUseCases(CameraUseCaseAdapter.java:348)\n\tat androidx.camera.core.internal.CameraUseCaseAdapter.updateUseCases(CameraUseCaseAdapter.java:365)\n\tat androidx.camera.core.internal.CameraUseCaseAdapter.updateUseCases(CameraUseCaseAdapter.java:299)\n\tat androidx.camera.core.internal.CameraUseCaseAdapter.addUseCases(CameraUseCaseAdapter.java:277)\n\t... 17 more\n","message":"java.lang.IllegalArgumentException: No supported surface combination is found for camera device - Id : 2.  May be attempting to bind too many use cases. Existing surfaces: [] New configs: [androidx.camera.video.impl.VideoCaptureConfig@96f06ef, androidx.camera.core.impl.ImageAnalysisConfig@43cf5fc, androidx.camera.core.impl.ImageAnalysisConfig@dcfb485]"},"stacktrace":"java.lang.IllegalArgumentException: androidx.camera.core.internal.CameraUseCaseAdapter$CameraException: java.lang. [... cut ...]
### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

raslankiwan, 2024-09-08:
@mrousavy

mrousavy, 2024-09-08:
> @mrousavy 
what?

mrousavy, 2024-09-12:
While this technically is a bug, I think there is nothing I can do about this as it is either your phone's Camera, or CameraX not properly figuring out suitable stream/resolution combinations.

You can just create your own Code Scanner plugin and run it in a Frame Processor, not really tricky to do tbh.

closed 2024-09-12 by mrousavy

levymetal, 2024-10-16:
For anyone stumbling across this issue, you can use `scanCodes` from [@mgcrea/vision-camera-barcode-scanner](https://github.com/mgcrea/vision-camera-barcode-scanner) within your own frame processor. I've posted an example implementation on a seperate issue here: https://github.com/mrousavy/react-native-vision-camera/issues/2684#issuecomment-2416489112