margelo/react-native-vision-camera #3169: 🐛 Camera preview blank screen when using Skia Frame Processors

oppfill, 2024-09-03:
### What's happening?

only useFrameProcessor works and useSkiaFrameProcessor just shows blank screen
i try out all the ways but it does not fixed.

### Reproduceable Code

```tsx
const device: any = useCameraDevice('back');

  const frameProcessor = useSkiaFrameProcessor(frame => {
    'worklet';
    frame.render();
  }, []);

      <Camera
        photo={true}
        device={device}
        frameProcessor={frameProcessor}
        style={{flex: 1}}
        isActive={true}
      />
```

### Relevant log output

```shell
at NativeBaseProvider (http://localhost:8081/index.bundle//&platform=android&dev=true&lazy=true&minify=false&app=com.pixipax.www&modulesOnly=false&runModule=true:241841:33)
    at RNCSafeAreaProvider
    at SafeAreaProvider (http://localhost:8081/index.bundle//&platform=android&dev=true&lazy=true&minify=false&app=com.pixipax.www&modulesOnly=false&runModule=true:200400:24)
    at App (http://localhost:8081/index.bundle//&platform=android&dev=true&lazy=true&minify=false&app=com.pixipax.www&modulesOnly=false&runModule=true:152831:26)
    at RCTView
    at View (http://localhost:8081/index.bundle//&platform=android&dev=true&lazy=true&minify=false&app=com.pixipax.www&modulesOnly=false&runModule=true:61807:43)
    at RCTView
    at View (http://localhost:8081/index.bundle//&platform=android&dev=true&lazy=true&minify=false&app=com.pixipax.www&modulesOnly=false&runModule=true:61807:43)
    at AppContainer (http://localhost:8081/index.bundle//&platform=android&dev=true&lazy=true&minify=false&app=com.pixipax.www&modulesOnly=false&runModule=true:61649:36)
    at OppFill(RootComponent) (http://localhost:8081/index.bundle//&platform=android&dev=true&lazy=true&minify=false&app=com.pixipax.www&modulesOnly=false&runModule=true:112678:28)
 LOG  Loading react-native-worklets-core...
 LOG  Worklets loaded successfully
```

### Camera Device

```json
LOG  {
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 24,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -24,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

xiaomi redmi note 9

### VisionCamera Version

4.5.2

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

oppfill, 2024-09-03:
RGB metadata [1.380859 1.000000 2.082031 ].
2024-09-03 13:02:41.741  1179-25744 pd_buf_mgr...ilmmipiraw camerahalserver                      D  total 0 pixels, sz 1099136, Stride 2480
2024-09-03 13:02:41.741  1179-25744 pd_buf_mgr...ilmmipiraw camerahalserver                      D  total[992 * 1108] = 1099136 pixels, Stride 2480
2024-09-03 13:02:41.743  1179-24849 isp_tuning_custom       camerahalserver                      D  [ratio mapping] sensorid(64210), sensormdoe(2)
2024-09-03 13:02:41.757  1179-24849 isp_tuning_custom       camerahalserver                      D  map_AppName_to_MappingInfo
2024-09-03 13:02:41.757  1179-24849 isp_tuning_custom       camerahalserver                      D  Appname(com.pixipax.www)
2024-09-03 13:02:41.757  1179-24849 isp_tuning_custom       camerahalserver                      D  [map_Target_Size_to_MappingInfo] width: 1920, height: 1080
2024-09-03 13:02:41.757  1179-24849 isp_tuning_custom       camerahalserver                      D  map_AppName_to_MappingInfo
2024-09-03 13:02:41.757  1179-24849 isp_tuning_custom       camerahalserver                      D  Appname(com.pixipax.www)
2024-09-03 13:02:41.757  1179-24849 isp_tuning_custom       camerahalserver                      D  [map_Target_Size_to_MappingInfo] width: 1920, height: 1080
2024-09-03 13:02:41.758  1179-25685 ifunc_cam_twin          camerahalserver                      W  [update_afterRunTwin_Dma] WARNING: slave_num 1 but channel.size() 1
2024-09-03 13:02:41.759  1179-25702 isp_tuning_custom       camerahalserver                      D  [map_Target_Size_to_MappingInfo] width: 1920, height: 1080
2024-09-03 13:02:41.760  1179-25422 mtkcam-Pip...delSession camerahalserver                      I  [submitRequest] submitRequest cannot get unique about CUSTOMER_DEMO_TEST
2024-09-03 13:02:41.760  1179-25422 mtkcam-Pip...delSession camerahalserver                      I  [submitRequest] submitRequest processUniqueKey = 0
2024-09-03 13:02:41.760  1179-24849 CAM_CUS_MSDK            camerahalserver                      D  CAM_CUS_MSDK [compareSensorIdAndModuleId] SID FAD2 is found
2024-09-03 13:02:41.760  1179-24849 CAM_CUS_MSDK            camerahalserver                      D  CAM_CUS_MSDK [compareSensorIdAndModuleId] target SID fad2 and MID 0 are found
2024-09-03 13:02:41.760  1179-24849 CAM_CUS_MSDK            camerahalserver                      D  CAM_CUS_MSDK Use MID data
2024-09-03 13:02:41.763  1179-24849 isp_tuning_custom       camerahalserver                      D  [ratio mapping] sensorid(64210), sensormdoe(2)

oppfill, 2024-09-03:
in my logcat everything is ok too, i have no any error

oppfill, 2024-09-03:
@mrousavy dude

mrousavy, 2024-09-03:
> i try out all the ways but it does not fixed.

What all ways did you try?

> @mrousavy dude

what dude?

> in my logcat everything is ok too, i have no any error

Without sharing full logs I cannot help you. Please post the full logs when you start the example app and use Skia Frame Processor. It works in there.

closed 2024-09-03 by mrousavy