margelo/react-native-vision-camera #3170: 🐛 Device torch flashing when takePhoto function called

FoundersApproach, 2024-09-03:
### What's happening?

Camera's torch is "on", When taking photo by takePhoto function, A already on torch is flashing.

### Reproduceable Code

```tsx
Takephoto function: (by calling takePhoto, torch flashes)
const photo = await cameraRef.takePhoto({
                photoQualityBalance: "quality",
                flash: "off",
                enableAutoStabilization: true,
                enableShutterSound: false
              });

Camera device:
<Camera
                onError={(error)=>props.onError(error)}
                orientation={getCurrentOrientation()}
                torch={"on"}
                onInitialized={() => {
                  setIsInitialized(true);
                  setCameraRef(cameraRef)
                }}
                format={Platform.OS == "android" ? format : null}
                codeScanner={codeScanner} 
                style={
                  isInitialized
                    ? {
                        position: "absolute",
                        top: 0,
                        left: 0,
                        right: 0,
                        bottom: 0,
                      }
                    : {
                        width: 0,
                        height: 0,
                      }
                }
                device={device}
                isActive={true}
                photo={true}
                ref={(ref) => cameraRef = ref}
              />
```

### Relevant log output

```shell
There is no log
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 24,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -24,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

All android phone Android

### VisionCamera Version

3.9.2

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

FoundersApproach, 2024-09-03:
@mrousavy can you please take a look ?

mrousavy, 2024-09-03:
The bot isn't meant to tell you to just ping you whenever you want.

Also you're using V3, which is no longer supported.

mrousavy, 2024-09-03:
> Device
>
> All android phone Android

This is also not helpful at all. I'm assuming you didn't actually test this on **all** 3 billion Android phones.

closed 2024-09-03 by mrousavy

wangchenglong1234, 2024-09-11:
I also encountered this issue in version v3