margelo/react-native-vision-camera #3171: 💭 Image Compression and Frame Processing Question

rmroot, 2024-09-04:
### Question

I have two questions. One relating to image capture and one related to frame processing.

*What we are trying to do:*
We are trying to apply a technique called "Background-Oriented Schlieren" to identify airflow using images. To do this you can subtract the pixels of 2 stationary images from each other to see if there has been a pixel shift that indicates airflow (dependent on air temperature). 

*What we have implemented:*
So far I have been able to use this library to implement the image capture portion. We then send the images to a server for processing to identify the pixels shifting and return manipulated images to the phone for display. It's a pretty straightforward approach but not the most user friendly. Ideally we would like to do this processing on the device but it's fairly computationally intensive to iterate through all the pixels and a cloud base solution has been the most effective so far.

*Question 1: How does this library handle the compression of image files?*
When I call `takePhoto()` or `takeSnapshot()` it creates a temporary image file that I am able to grab from the device and send off for processing. Would you be able to tell me what sort of compression is being done to those .png files that are temporarily stored within the application data? Is this something that the library is doing or is it device/hardware specific?

*Question 2: Is it possible to use 2 frames within a frame processor?*
My first attempt at an implementation was using a frame processor but I pivoted from their because of inexperience and trouble using 2 frames within the frame processor. Is it possible to use 2 frames within a frame processor? All of the examples I have seen are looking at single frames and I would appreciate knowing if I'm on a fools errand trying to compare two frames within the frame processor architecture. 

Thank you for any info that can be provided!

### VisionCamera Version

4.5.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-09-12:
Hi!

> 1.: How does this library handle the compression of image files?
Compression is handled using normal device JPEG encoders. That's it. For very obvious reasons, I don't choose any opinionated compression algorithms or settings. _"to those .png files"_ What png files? VisionCamera does not use png.

> 2. Is it possible to use 2 frames within a frame processor?

A Frame Processor is a stream of Frames. You can queue up frames if you want. So yes if that answers your question.

closed 2024-09-12 by mrousavy

rmroot, 2024-09-12:
> Hi!
>
>
> Compression is handled using normal device JPEG encoders. That's it. For very obvious reasons, I don't choose any opinionated compression algorithms or settings. _"to those .png files"_ What [...]

My bad, definitely meant "jpg" not "png". Thanks for the feedback. Good to know I can use multiple frames and I am not trying to do something impossible. Appreciate your work!