margelo/react-native-vision-camera #3172: 💭 Fatal Error on App Initialization: IllegalArgumentException - width must be positive in CameraDeviceDetails.getFormats

danielbcarvalho, 2024-09-05:
### Question

I’m experiencing a fatal error on app initialization after updating to react-native-vision-camera version 4.2.0. The error occurs before the camera is opened, and I’m seeing a high number of these crashes reported in Firebase Crashlytics, especially on Xiaomi devices (74%) running Android 10 (81%).

After updating to react-native-vision-camera version 4.2.0, I’ve observed a significant number of crashes related to this error in Firebase Crashlytics, the error occurs before the camera is opened and is predominantly on Xiaomi devices. 

I’m reaching out to see if anyone has experienced this issue before and can offer guidance on how to resolve it. Any tips or solutions would be greatly appreciated since the issue isn’t reproducible on all devices I have access to.

### What I tried

	1.	The app crashes on startup before the camera screen is displayed.
	2.	I have not been able to reproduce the issue on my local device as I don’t have access to the specific models where this error is prevalent.

Below is a simplified example of how the <Camera> component is being used in the app:

```
import React, { useState, useEffect } from 'react';
import { Camera, useCameraDevice, useCodeScanner } from 'react-native-vision-camera';
import { View } from 'react-native';

const QrCodePaymentScreen = () => {
  const [activeCamera, setCameraActive] = useState(true);
  const device = useCameraDevice('back');

  const codeScanner = useCodeScanner({
    codeTypes: ['qr'],
    onCodeScanned: (codes) => {
      // Handle scanned QR codes
    },
  });

  useEffect(() => {
    // Example logic that might trigger the camera to start
    setCameraActive(true);
  }, []);

  if (!device) {
    return <ActivityIndicator size="large" />;
  }

  return (
    <View style={{ flex: 1 }}>
      <Camera
        audio={false}
        device={device}
        style={{ flex: 1 }}
        isActive={activeCamera}
        codeScanner={codeScanner}
      />
    </View>
  );
};

```
export default QrCodePaymentScreen;

### VisionCamera Version

4.2.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

danielbcarvalho, 2024-09-05:
Logs from Crashlytics:

`CameraDeviceDetails.getFormats
java.lang.IllegalArgumentException - width must be positive

Fatal Exception: java.lang.IllegalArgumentException
width must be positive
com.android.internal.util.Preconditions.checkArgumentPositive (Preconditions.java:275)
android.hardware.camera2.params.StreamConfigurationDuration.<init> (StreamConfigurationDuration.java:58)
android.hardware.camera2.marshal.impl.MarshalQueryableStreamConfigurationDuration$MarshalerStreamConfigurationDuration.unmarshal (MarshalQueryableStreamConfigurationDuration.java:68)
android.hardware.camera2.marshal.impl.MarshalQueryableStreamConfigurationDuration$MarshalerStreamConfigurationDuration.unmarshal (MarshalQueryableStreamConfigurationDuration.java:45)
android.hardware.camera2.marshal.impl.MarshalQueryableArray$MarshalerArray.unmarshal (MarshalQueryableArray.java:93)
android.hardware.camera2.impl.CameraMetadataNative.getBase (CameraMetadataNative.java:567)
android.hardware.camera2.impl.CameraMetadataNative.getBase (CameraMetadataNative.java:537)
android.hardware.camera2.impl.CameraMetadataNative.getStreamConfigurationMap (CameraMetadataNative.java:1258)
android.hardware.camera2.impl.CameraMetadataNative.access$400 (CameraMetadataNative.java:89)
android.hardware.camera2.impl.CameraMetadataNative$5.getValue (CameraMetadataNative.java:605)
android.hardware.camera2.impl.CameraMetadataNative.get (CameraMetadataNative.java:437)
android.hardware.camera2.impl.CameraMetadataNative.get (CameraMetadataNative.java:400)
android.hardware.camera2.CameraCharacteristics.get (CameraCharacteristics.java:235)
androidx.camera.camera2.internal.compat.CameraCharacteristicsBaseImpl.get (CameraCharacteristicsBaseImpl.java:40)
androidx.camera.camera2.internal.compat.CameraCharacteristicsCompat.get (CameraCharacteristicsCompat.java:108)
androidx.camera.camera2.internal.compat.CameraCharacteristicsCompat.getStreamConfigurationMapCompat (CameraCharacteristicsCompat.java:134)
androidx.camera.camera2.internal.Camera2CameraInfoImpl.getSupportedResolutions (Camera2CameraInfoImpl.java:434)
com.mrousavy.camera.core.CameraDeviceDetails.getFormats (CameraDeviceDetails.kt:118)
com.mrousavy.camera.core.CameraDeviceDetails.toMap (CameraDeviceDetails.kt:79)
com.mrousavy.camera.react.CameraDevicesManager.getDevicesJson (CameraDevicesManager.kt:87)
com.mrousavy.camera.react.CameraDevicesManager.sendAvailableDevicesChangedEvent (CameraDevicesManager.kt:94)
com.mrousavy.camera.react.CameraDevicesManager$initialize$1.invokeSuspend (CameraDevicesManager.kt:71)
kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith (ContinuationImpl.kt:33)
kotlinx.coroutines.DispatchedTask.run (DispatchedTask.kt:108)
java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
java.lang.Thread.run (Thread.java:919)
`

ncoen97, 2024-09-08:
Same error here!
I see it mostly on Redmi Note 8 (android 9 10 and 11)

mrousavy, 2024-09-08:
You created a question. Not a bug report.
Please fill out the bug report template, **that's what it's for!**

closed 2024-09-08 by mrousavy

danielbcarvalho, 2024-09-09:
> You created a question. Not a bug report. Please fill out the bug report template, **that's what it's for!**

Done!

https://github.com/mrousavy/react-native-vision-camera/issues/3180

Thank you very much.