margelo/react-native-vision-camera #3174: 💭 V3 is not reading Data-Matrix codes  with black background on Android.

KarimovAbdullo, 2024-09-06:
### Question

[image]
I have a project related to scanning codes. The issue is that when I scan Data-Matrix codes on a black background, the scanner cannot detect them, but everything works fine on iOS.

not detecting these codes
[image]

detecting these codes
[image]

### What I tried

react-native : 0.71.1
re

const ScannerCamera = forwardRef<ScannerAnimationRefI, ScannerCameraProps>(
  (props, animationRef) => {
    const device = useCameraDevice('back', {
      physicalDevices: ['wide-angle-camera'],
    });

    const {hasPermission, requestPermission} = useCameraPermission();

    useEffect(() => {
      if (!hasPermission) {
        requestPermission();
      }
    }, [hasPermission, requestPermission]);

    const [torch, setTorch] = useState<'on' | 'off'>('off');

    return (
      <View style={styles.scannerContainer}>
        <View style={styles.scanner}>
          {device && hasPermission ? (
            <Camera
              {...props}
              torch={torch}
              device={device}
              style={StyleSheet.absoluteFillObject}
            />
          ) : null}
          <ScanerAnimation ref={animationRef} />
        </View>
        {hasPermission ? null : <CText RED>Нет разрешения на камеру</CText>}
        <CButton
          onPress={() => setTorch(prev => (prev === 'on' ? 'off' : 'on'))}
          outline>
          {torch === 'on' ? 'Выключить фонарик' : 'Включить фонарик'}
        </CButton>

      
      </View>
    );
  },
);
export default ScannerCamera;

/////////////
  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13', 'ean-8', 'data-matrix'],
    onCodeScanned,
  });

  <ScannerCamera
              ref={animationRef}
              codeScanner={codeScanner}
              isActive={true}
            />

### VisionCamera Version

react-native-vision-camera:3.4.6

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

KarimovAbdullo, 2024-09-11:
@mrousavy

mrousavy, 2024-09-11:
Listen - the purpose of the bot is not to just ping me when you need quick help.

It told you to update logs, add more info to this, and technically you even created the wrong template - this is a bug not a question.

closed 2024-09-16 by mrousavy

xulihang, 2024-09-19:
You can use Dynamsoft Barcode Reader to read the inverted DPM code.

Online demo: https://demo.dynamsoft.com/barcode-reader/

Vision Camera Plugin: https://github.com/tony-xlh/vision-camera-dynamsoft-barcode-reader/