margelo/react-native-vision-camera #3177: 🐛 [4.5.3] Google Api Error: Invalid request - Your app declares support for 'riscv64' architecture, but the following libraries are missing

efstathiosntonas, 2024-09-09:
### What's happening?

After trying to upload a production release on Google Play Store it throwed this error. Downgrading to 4.5.2 the build was accepted. It seems that CameraX@15 has some issues with rn?

```log
[!] Google Api Error: Invalid request - Your app declares support for 'riscv64' architecture, but the following libraries are missing:
 [libVisionCamera.so, libandroidlame.so, libanimation-decoder-gif.so, libavcodec.so, libavdevice.so, libavfilter.so, libavformat.so, libavif_android.so, libavutil.so, libbarhopper_v3.so, libbugsnag-ndk.so, libbugsnag-plugin-android-anr.so, libbugsnag-root-detection.so, libc++_shared.so, libcrsqlite.so, libcrypto.so, libexpo-av.so, libexpo-modules-core.so, libfabricjni.so, libfbjni.so, libffmpegkit.so, libffmpegkit_abidetect.so, libfolly_runtime.so, libgifimage.so, libglog.so, libhermes.so, libhermes_executor.so, libhermesinstancejni.so, libimagepipeline.so, libjscinstance.so, libjsi.so, libjsijniprofiler.so, libjsinspector.so, libmapbox-common.so, libmapbox-maps.so, libmapbufferjni.so, libnative-filters.so, libnative-imagetranscoder.so, libop-sqlite.so, libreact-native-keys.so, libreact_codegen_rncore.so, libreact_debug.so, libreact_devsupportjni.so, libreact_featureflags.so, libreact_featureflagsjni.so, libreact_nativemodule_core.so, libreact_nativemodule_defaults.so, libreact_nativemodule_dom.so, libreact_nativemodule_featureflags.so, libreact_nativemodule_microtasks.so, libreact_newarchdefaults.so, libreact_performance_timeline.so, libreact_render_componentregistry.so, libreact_render_consistency.so, libreact_render_core.so, libreact_render_debug.so, libreact_render_graphics.so, libreact_render_imagemanager.so, libreact_render_mapbuffer.so, libreact_render_observers_events.so, libreact_render_uimanager_consistency.so, libreact_utils.so, libreactnativeblob.so, libreactnativejni.so, libreanimated.so, librninstance.so, librnscreens.so, librrc_image.so, librrc_legacyviewmanagerinterop.so, librrc_textinput.so, librrc_view.so, libruntimeexecutor.so, libsqlite_vec.so, libstatic-webp.so, libswresample.so, libswscale.so, libturbomodulejsijni.so, libucrop.so, libuimanagerjni.so, libunistyles.so, libworklets.so, libyoga.so, libavcodec_neon.so, libavdevice_neon.so, libavfilter_neon.so, libavformat_neon.so, libavutil_neon.so, libffmpegkit_armv7a_neon.so, libswresample_neon.so, libswscale_neon.so].

 Update your app to rely on libraries that support 'riscv64' architecture.
```

Probably related: 
* https://github.com/expo/expo/issues/31370

related: 
* https://github.com/facebook/react-native/issues/46397

### Reproduceable Code

```tsx
no repro, just try to upload a release build on Play Store.
```

### Relevant log output

```shell
no logs
```

### Camera Device

```json
not needed
```

### Device

Android

### VisionCamera Version

4.5.3

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Jonnboy91, 2024-09-10:
I'm getting the same issue as well with 4.5.3, now testing to downgrade to 4.5.2 to see if that works.

SKHRAPP, 2024-09-10:
@Jonnboy91 is that working with version 4.5.2

Jonnboy91, 2024-09-10:
Yeah 4.5.2 went through. 👌

SKHRAPP, 2024-09-11:
@Jonnboy91 did you solve this issue

Jonnboy91, 2024-09-11:
As mentioned on the earlier comment, the version 4.5.2 works 👌

SKHRAPP, 2024-09-11:
@Jonnboy91 how did you solve this issue 

```
* What went wrong:
Execution failed for task ':app:packageStagingReleaseBundle'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.PackageBundleTask$BundleToolWorkAction
   > Unrecognized native architecture for directory 'lib/riscv64'.
```

Jonnboy91, 2024-09-11:
That was the issue that came with 4.5.3 version 😅 So I solved it by downgrading to 4.5.2 as mentioned earlier.

SKHRAPP, 2024-09-11:
@Jonnboy91 me also downgraded from 4.5.3 to 4.5.2 still getting error for creating production Build

efstathiosntonas, 2024-09-11:
do you gradle clean before production builds when downgrading from 4.5.3 to 4.5.2?

SKHRAPP, 2024-09-11:
@efstathiosntonas yes

LeXuanKhanh, 2024-09-11:
TDLR:
just happened to me today, the workaround is to go to that cache folder and manually delete that risv64 folder
another way is to use a filter the API architecture when building with Android but I haven't tried it yet
Source and explanation:
https://stackoverflow.com/questions/76753627/how-to-fix-android-project-build-failing-with-so-is-not-an-abi 
https://stackoverflow.com/questions/78609800/agp-8-0-so-is-not-an-abi-for-3rd-party-library
https://issuetracker.google.com/issues/279516850?pli=1

amir-ibrahim, 2024-09-12:
so we solved the issue by using version 4.5.2 and removing the ^ from the version
@Amurmurmur 

[image]

MahmoudHemaid, 2024-09-12:
> so we solved the issue by using version 4.5.2 and removing the ^ from the version @Amurmurmur
> 
> ![Screenshot 2024-09-12 at 9 53 26 AM](https://private-user-images.githubusercontent.com/41516271/366760882-97080522-78c5-4b37-804a-c7a313582d3e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYxNDA3MzYsIm5iZiI6MTcyNjE0MDQzNiwicGF0aCI6Ii80MTUxNjI3MS8zNjY3NjA4ODItOTcwODA1MjItNzhjNS00YjM3LTgwNGEtYzdhMzEzNTgyZDNlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTEyVDExMjcxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMwYTllZmE2YWIwYmM4ZmQxMWY4MTJmMjNhYzE2OWExNzNkODk5ZTQ4ODdkOTAxYjYzOWE5NmZkZDJmNmFiNzAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.tT__9FXDsF6SmtQORc9bPW67OTNjx5AgeuLHJrVMmDw)

Worked with me

mrousavy, 2024-09-12:
I am pretty sure this is not a VisionCamera issue, but rather an issue of adopting support for a new architecture. CameraX is just already ready for RISCv64, while some other native libs (like RN) are not.

Add this to your app's build.gradle to temporarily remove riscv64:

```groovy
android {
  // ..
    defaultConfig {
      // ..
      ndk {
        abiFilters "armeabi-v7a", "arm64-v8a", "x86", "x86_64"
      }
```

mrousavy, 2024-09-12:
Yea, stability improvements.

I just removed riscv64 from my app and everything works fine. I don't think I should throttle down development speed just because the ecosystem isn't fully ready yet

Jonnboy91, 2024-09-12:
Alright fair enough! Just was interested in hearing the reasons 💪

mrousavy, 2024-09-12:
No worries - yea I was also surprised to find out that this simply breaks the build. I guess we have to be extra careful about such things.

lucksp, 2024-09-23:
Following...i also have this issue after upgrading to:

```
"expo": "^51.0.0",
"react-native": "0.74.5",
"react-native-vision-camera": "^4.5.3",
```

> Add this to your app's build.gradle to temporarily remove riscv64:
> 
> ```groovy
> android {
>   // ..
>     defaultConfig {
>       // ..
>       ndk {
>         abiFilters "armeabi-v7a", "arm64-v8a", "x86", "x86_64"
>       }
> ```

Will `riscv64` be removed from RNVC in a future release?  I use expo & EAS so I don't plan to edit the build.gradle.

I downgraded back to RNVC 4.5.1 and my build also worked fine now.

JoaoPauloCMarra, 2024-09-23:
> Following...i also have this issue after upgrading to:
>
> ```
> "expo": "^51.0.0",
> "react-native": "0.74.5",
> "react-native-vision-camera": "^4.5.3",
> ```
>
>
> Will `riscv64` be removed from RNVC in a future [...]

4.5.2 should work

lucksp, 2024-09-23:
Yes, 4.5.2 should work but for now, however that might be the limit to
upgrade version if riscv64 remains enabled in 4.5.3 and above

rieg-ec, 2024-09-24:
in case anyone using expo wants to keep with latest version, this plugin adds the abiFilters that @mrousavy mentioned

```
const { withAppBuildGradle, withPlugins } = require('@expo/config-plugins');

const withCustomBuildGradle = (config) => withAppBuildGradle(config, (config) => {
  let buildGradle = config.modResults.contents;

  const ndkBlockInDefaultConfigRegex = /defaultConfig\s*{[^}]*ndk\s*{[^}]*abiFilters[^}]*}[^}]*}/s;
  if (!ndkBlockInDefaultConfigRegex.test(buildGradle)) {
    const defaultConfigStartRegex = /(defaultConfig\s*{)/;
    buildGradle = buildGradle.replace(
      defaultConfigStartRegex,
      `$1
        ndk {
            abiFilters "armeabi-v7a", "arm64-v8a", "x86", "x86_64"
        }`,
    );
  }

  config.modResults.contents = buildGradle;
  return config;
});

module.exports = function withAndroidNdkAbiFilters(config) {
  return withPlugins(config, [withCustomBuildGradle]);
};
```

it now builds with `eas build --profile production`  using `"react-native-vision-camera": "^4.5.3"`

patlux, 2024-10-01:
For those who are using `split`. You need to exclude `riscv` as following:

```gradle
android {
    // ...
    packagingOptions {
        exclude "lib/riscv64/**"  // Exclude riscv64 libraries
    }
}

```