margelo/react-native-vision-camera #3178: 💭 Can't find VisionCamera.xcodeproj as Readme

LeXuanKhanh, 2024-09-09:
### Question

Hi, I try to poke around the iOS native code of the library to make some contributions. However, the [iOS-platform-specific code guide ](https://github.com/mrousavy/react-native-vision-camera/tree/main/package/ios) said that there should be a VisionCamera.xcodeproj to open the folder in Xcode but as far as I see the file doesn't exist. Do you know if it is missing from the project or how can I generate it ? tks

### VisionCamera Version

main branch

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

LeXuanKhanh, 2024-09-09:
@mrousavy can you take a look ?

mrousavy, 2024-09-10:
It's in `example/`, that doc is outdated

closed 2024-09-10 by mrousavy