margelo/react-native-vision-camera #3179: 🐛 InitializationException - Device reporting less cameras than anticipated

ncoen97, 2024-09-09:
### What's happening?

I am encountering a crash reported in Crashlytics, related to `androidx.camera.core.InitializationException`. The issue seems to occur when the app attempts to initialize the camera, and it reports fewer cameras than expected, resulting in a `CameraUnavailableException`. This crash happens on real devices, and it mentions retrying initialization. However, I cannot reproduce the issue locally on my test devices, so I’m unable to investigate further in a controlled environment.

### Reproduceable Code

```tsx
const QRCodeScanner: React.FC<QRCodeScannerProps> = ({
  onCodeScanned,
  loading,
  handlePermissionBlocked,
}) => {
  const device = useCameraDevice('back');
  const camera = useRef<Camera>(null);
  const isFocused = useIsFocused();
  const appState = useAppState();
  const [cameraReady, setCameraReady] = useState(false);
  const {onLayout, layout} = useLayout();
  const hasPermission = useCameraPermissions(handlePermissionBlocked);
  const codeScanner = useCodeScanner({
    codeTypes: ['qr'],
    onCodeScanned,
  });

  const isActive = isFocused && appState === 'active' && !!layout && !loading;
  const isScanning = isActive && cameraReady;

  const onError = (err: CameraRuntimeError | string) =>
    Analytics.logError(err, 'Camera error');

  return device && hasPermission ? (
    <View style={styles.container} onLayout={onLayout}>
      {isScanning ? (
        <QRMask
          containerHeight={layout.height}
          containerWidth={layout.width}
        />
      ) : null}
      {!isScanning ? (
        <View style={styles.loaderContainer}>
          <DefaultLoader />
        </View>
      ) : null}
      <Camera
        ref={camera}
        onError={onError}
        photo={false}
        device={device}
        codeScanner={codeScanner}
        isActive={isActive}
        style={StyleSheet.absoluteFill}
        onPreviewStarted={() => setCameraReady(true)}
        onStopped={() => setCameraReady(false)}
      />
    </View>
  ) : null;
};
```

### Relevant log output

```shell
Crashlytics report:

Fatal Exception: androidx.camera.core.InitializationException
androidx.camera.core.CameraUnavailableException: Device reporting less cameras than anticipated. On real devices: Retrying initialization might resolve temporary camera errors. On emulators: Ensure virtual camera configuration matches supported camera features as reported by PackageManager#hasSystemFeature. Available cameras: 0

androidx.camera.core.CameraX.lambda$initAndRetryRecursively$2$androidx-camera-core-CameraX (CameraX.java:433)
```
```

### Camera Device

```json
I cannot provide the JSON for the camera device since the issue occurs in production and I cannot reproduce it locally.
```

### Device

The crash affects multiple devices, with the following breakdown:  Motorola (69%) Samsung (17%) Xiaomi (14%). Operating system versions:  Android 11 (62%) Android 13 (21%) Android 12 (17%).

### VisionCamera Version

4.5.2

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

LukasMod, 2024-09-11:
Hey, I have same problem on production, multiple crashes in crashlytics (its small % of users, but enough to be one of main reason for our crashes currently). Maybe my information will help you somehow to debug this.
    
```
"react-native": "0.73.8",
"react-native-vision-camera": "4.5.0"
```
I add stacktarce from one of crashes. 
[stacktrace.txt](https://github.com/user-attachments/files/16963668/stacktrace.txt)

Name of error is `androidx.camera.core.CameraX.lambda$initAndRetryRecursively$2$androidx-camera-core-CameraX (CameraX.java:369)`. Mostly (46%) its Xiaomi devices, different androids.

mrousavy, 2024-09-12:
Well, Camera APIs on Android suck.

efstathiosntonas, 2024-09-16:
Started seing this out of the sudden on Android 9 on 4.5.2, rn 0.75.3. 

Can reproduce it locally on Pixel emulator running Android 9. @mrousavy do you need any logs or screw camera apis 😂?

efstathiosntonas, 2024-09-17:
Some extra information around the bug: 

1. Open device camera app
2. open your app
3. try to pick a photo, throws: ` [session/recoverable-error: An unknown error occurred while creating the Camera Session, but the Camera can recover from it.]
 ERROR  [device/camera-already-in-use: The given Camera Device is already in use!]`
4. Close your app
5. try to open device camera app again: Android throws: `Camera error Can't connect to camera` so it seems it brakes Android system too.

stack:

```logcat
FATAL EXCEPTION: pool-26-thread-2 (Ask Gemini)
Process: com.xxxxx, PID: 10365
androidx.camera.core.InitializationException: androidx.camera.core.CameraUnavailableException: Device reporting less cameras than anticipated. On real devices: Retrying initialization might resolve temporary camera errors. On emulators: Ensure virtual camera configuration matches supported camera features as reported by PackageManager#hasSystemFeature. Available cameras: 0
 at androidx.camera.core.CameraX.lambda$initAndRetryRecursively$2$androidx-camera-core-CameraX(CameraX.java:433)
 at androidx.camera.core.CameraX$$ExternalSyntheticLambda2.run(D8$$SyntheticClass:0)
 at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
 at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
 at java.lang.Thread.run(Thread.java:764)
 Suppressed: kotlinx.coroutines.internal.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@81d901b, java.util.concurrent.ThreadPoolExecutor@660c5b8[Running, pool size = 2, active threads = 1, queued tasks = 0, completed tasks = 2]]
Caused by: androidx.camera.core.CameraUnavailableException: Device reporting less cameras than anticipated. On real devices: Retrying initialization might resolve temporary camera errors. On emulators: Ensure virtual camera configuration matches supported camera features as reported by PackageManager#hasSystemFeature. Available cameras: 0
```

LuongTruong, 2024-09-19:
Hello guys, I just found a way to reproduce the issue.

1. From Edit Android Emulator -> Show Advanced Settings -> Camera: Front -> None and Back -> None
2. Run your app on that emulator and the issue will appear

Thanks guys for pointing nearly all the versions that have the issue. As a result I use `react-native-vision-camera: 4.2.1`.

Below is my final version in `package.json`

```
"dependencies": {
  "react": "18.2.0",
  "react-native": "0.73.9",
  "react-native-reanimated": "3.15.2",
  "react-native-vision-camera": "4.2.1",
},
```

Hope it can help.

Rag0n, 2024-09-21:
Same issue with react-native-vision-camera 4.5.2

Nader-CS, 2024-09-23:
same issue

ShchipskiiYura, 2024-10-08:
same problem

imamrobani, 2024-10-10:
same issue 

"react-native": "0.73.9",
"react-native-vision-camera": "^4.5.2"

Nader-CS, 2024-10-16:
any help ?

39otrebla, 2024-10-22:
Same issue here, but telemetry (Sentry) suggests these may not be real users.

DarshanSavaliyaGitHub, 2024-10-29:
Same issue here...

marcos-cunha-gringo, 2024-10-31:
We are having the same issue here:
 - react-native-vision-camera: 4.5.0;
 - react-native: 0.74.5;

mannoeu, 2024-12-03:
> any updates ?

Unfortunately, no one can reproduce the error concisely. :(

UKMIITB, 2024-12-10:
Has the camerax version has been updated to 1.4.0 ?

I am also facing this issue post version update & have raised it here.
Please upvote if the issue is same

Have raised it here
https://issuetracker.google.com/issues/383220395

mrousavy, 2024-12-10:
> Has the camerax version has been updated to 1.4.0 ?

https://github.com/mrousavy/react-native-vision-camera/blob/54df86ace1716039f690e865c97f29f87dfe43b7/package/android/build.gradle#L185-L192

hongying-yang, 2024-12-11:
@mrousavy We are using version 4.6.1, and our build.gradle file is the same as the one you provided, 1.5.0-alpha03. However, some phones are still reporting this error.

Nader-CS, 2024-12-11:
still same

hozdemir, 2024-12-24:
still same, we get this error on some devices. do you have any suggestions for a solution?

kadiraydinli, 2024-12-27:
Still same. Is there a solution for this?

Nader-CS, 2024-12-27:
any updates ? thie issue happen today for 15 user.

JuhilGajeraGitHub, 2025-01-01:
I am also getting this crash in firebase crashlytics

levepic, 2025-01-06:
So do i

litinskii, 2025-01-12:
+1 Same issue

rb-adnan, 2025-01-12:
i am also getting the same error.
-- i am not able to reproduce exactly but getting this log from some devices listed below with OS version
- here's some stacktrace from dynatrace:

details:
"react-native": "0.75.4",
"react-native-vision-camera": "^4.5.3",

**Devices**
P30 Pro (46.6 %)
Galaxy A51 (26.2 %)
HUAWEI nova 5T (3.4 %)
Infinix HOT 40i (2.43 %)
Others (21.4 %)

**Operating system**
Android 10 (50 %)
Android 13 (31.1 %)
Android 11 (6.31 %)
Android 12 (4.37 %)
Others (8.25 %)

`androidx.camera.core.InitializationException: androidx.camera.core.CameraUnavailableException: Device reporting less cameras than anticipated. On real devices: Retrying initialization might resolve temporary camera errors. On emulators: Ensure virtual camera configuration matches supported camera features as reported by PackageManager#hasSystemFeature. Available cameras: 0

at

androidx.camera.core.CameraX.lambda$initAndRetryRecursively$2$androidx-camera-core-CameraX(CameraX.java:433)

at

androidx.camera.core.CameraX$$ExternalSyntheticLambda2.run(D8$$SyntheticClass:0)

at

java.util.concurrent.ThreadPoolExecutor.processTask(ThreadPoolExecutor.java:1187)

at

java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1152)

at

java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)

at

java.lang.Thread.run(Thread.java:784)`

mannoeu, 2025-01-17:
> Hi everyone, As mentioned in the above issue tracker that this is an expected behaviour It was happening previously as well (with older versions), just that it was failing silently. And now the exception is thrown.
> 
> Handling for this is to wrap the `cameraProviderFuture.get()` call with `try catch` block Sample is shared here https://issuetracker.google.com/issues/383220395#comment3

The app will crash? Can i ignore this error?

mrousavy, 2025-01-20:
created a PR to potentially fix this: https://github.com/mrousavy/react-native-vision-camera/pull/3373

Nader-CS, 2025-01-24:
anyone can confirm if this pr works ?

rb-adnan, 2025-01-26:
@Nader-CS you need fix the variable declaration for `provider`, then should work fine.

sorenfrederiksen, 2025-01-27:
> created a PR to potentially fix this: [#3373](https://github.com/mrousavy/react-native-vision-camera/pull/3373)

Awesome! Thanks for looking into this, @mrousavy - really appreciate your hard work 💪 

How many people in this long thread are considering supporting Marc's work? We really should be.

PR #3385 (2025-01-30): fix: camera initialization error (#3179)

## What

This PR prevents crashes caused by Android camera initialization exceptions, on devices without cameras.

## Changes

This PR ensures calls to `ProcessCameraProvider.getInstance(reactContext).await(executor)` are safely wrapped in a try-catch statement and logs resulting errors.

Please feel free to make changes to my edits and do with this what you will - really appreciate your work 🙏

## Tested on

### Android simulated device setup to have no cameras

#### Before 🐛

Catastrophic failure, resulting in a crash of the application.

https://git [... cut]
closed 2025-01-30 by mrousavy