margelo/react-native-vision-camera #3180: 🐛 Fatal Error on App Initialization: IllegalArgumentException - width must be positive in CameraDeviceDetails.getFormats

danielbcarvalho, 2024-09-09:
### What's happening?

I’m experiencing a fatal error on app initialization after updating to react-native-vision-camera version 4.2.0. The error occurs before the camera is opened, and I’m seeing a high number of these crashes reported in Firebase Crashlytics, especially on Xiaomi devices (74%) running Android 10 (81%). I see it mostly on Redmi Note 8 .

After updating to react-native-vision-camera version 4.2.0, I’ve observed a significant number of crashes related to this error in Firebase Crashlytics, the error occurs before the camera is opened and is predominantly on Xiaomi devices.

I’m reaching out to see if anyone has experienced this issue before and can offer guidance on how to resolve it. Any tips or solutions would be greatly appreciated since the issue isn’t reproducible on all devices I have access to.

1.	The app crashes on startup before the camera screen is displayed.
2.	I have not been able to reproduce the issue on my local device as I don’t have access to the specific models where this error is prevalent.

### Reproduceable Code

```tsx
import React, { useState, useEffect } from 'react';
import { Camera, useCameraDevice, useCodeScanner } from 'react-native-vision-camera';
import { View } from 'react-native';

const QrCodePaymentScreen = () => {
  const [activeCamera, setCameraActive] = useState(true);
  const device = useCameraDevice('back');

  const codeScanner = useCodeScanner({
    codeTypes: ['qr'],
    onCodeScanned: (codes) => {
      // Handle scanned QR codes
    },
  });

  useEffect(() => {
    // Example logic that might trigger the camera to start
    setCameraActive(true);
  }, []);

  if (!device) {
    return <ActivityIndicator size="large" />;
  }

  return (
    <View style={{ flex: 1 }}>
      <Camera
        audio={false}
        device={device}
        style={{ flex: 1 }}
        isActive={activeCamera}
        codeScanner={codeScanner}
      />
    </View>
  );
};
```

### Relevant log output

```shell
`CameraDeviceDetails.getFormats
java.lang.IllegalArgumentException - width must be positive

Fatal Exception: java.lang.IllegalArgumentException
width must be positive
com.android.internal.util.Preconditions.checkArgumentPositive (Preconditions.java:275)
android.hardware.camera2.params.StreamConfigurationDuration. (StreamConfigurationDuration.java:58)
android.hardware.camera2.marshal.impl.MarshalQueryableStreamConfigurationDuration$MarshalerStreamConfigurationDuration.unmarshal (MarshalQueryableStreamConfigurationDuration.java:68)
android.hardware.camera2.marshal.impl.MarshalQueryableStreamConfigurationDuration$MarshalerStreamConfigurationDuration.unmarshal (MarshalQueryableStreamConfigurationDuration.java:45)
android.hardware.camera2.marshal.impl.MarshalQueryableArray$MarshalerArray.unmarshal (MarshalQueryableArray.java:93)
android.hardware.camera2.impl.CameraMetadataNative.getBase (CameraMetadataNative.java:567)
android.hardware.camera2.impl.CameraMetadataNative.getBase (CameraMetadataNative.java:537)
android.hardware.camera2.impl.CameraMetadataNative.getStreamConfigurationMap (CameraMetadataNative.java:1258)
android.hardware.camera2.impl.CameraMetadataNative.access$400 (CameraMetadataNative.java:89)
android.hardware.camera2.impl.CameraMetadataNative$5.getValue (CameraMetadataNative.java:605)
android.hardware.camera2.impl.CameraMetadataNative.get (CameraMetadataNative.java:437)
android.hardware.camera2.impl.CameraMetadataNative.get (CameraMetadataNative.java:400)
android.hardware.camera2.CameraCharacteristics.get (CameraCharacteristics.java:235)
androidx.camera.camera2.internal.compat.CameraCharacteristicsBaseImpl.get (CameraCharacteristicsBaseImpl.java:40)
androidx.camera.camera2.internal.compat.CameraCharacteristicsCompat.get (CameraCharacteristicsCompat.java:108)
androidx.camera.camera2.internal.compat.CameraCharacteristicsCompat.getStreamConfigurationMapCompat (CameraCharacteristicsCompat.java:134)
androidx.camera.camera2.internal.Camera2CameraInfoImpl.getSupportedResolutions (Camera2CameraInfoImpl.java:434)
com.mrousavy.camera.core.CameraDeviceDetails.getFormats (CameraDeviceDetails.kt:118)
com.mrousavy.camera.core.CameraDeviceDetails.toMap (CameraDeviceDetails.kt:79)
com.mrousavy.camera.react.CameraDevicesManager.getDevicesJson (CameraDevicesManager.kt:87)
com.mrousavy.camera.react.CameraDevicesManager.sendAvailableDevicesChangedEvent (CameraDevicesManager.kt:94)
com.mrousavy.camera.react.CameraDevicesManager$initialize$1.invokeSuspend (CameraDevicesManager.kt:71)
kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith (ContinuationImpl.kt:33)
kotlinx.coroutines.DispatchedTask.run (DispatchedTask.kt:108)
java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
java.lang.Thread.run (Thread.java:919)
`
```

### Camera Device

```json
I have not been able to reproduce the issue on my local device as I don’t have access to the specific models where this error is prevalent.
```

### Device

Redmi Note 8 

### VisionCamera Version

4.2.0

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-10-04:
bruh why does this throw an error

GonzalezVictorJavier, 2024-10-04:
Hi, how are you?. We have this same issue (Same exception and Xiaomi participation percentage at devices with errors), but at our proprietary CameraX app implementation. Seems to be some kind of Xiaomi - CameraX error, right?

eTryp, 2024-10-09:
Hi, thanks for this great library.
Just to highlight that I'm having the same issue on a Sunmi D3 device (https://www.sunmi.com). Even without any call to the camera, just by including it in the application.

mrousavy, 2024-10-10:
Hey all! Fixed this in https://github.com/mrousavy/react-native-vision-camera/pull/3236.

closed 2024-10-10 by mrousavy

danielbcarvalho, 2025-04-01:
Thank you very much.