margelo/react-native-vision-camera #3181: 🔧 v4.5.3 fails to build on Android - jetified-camera-core-1.5.0-alpha01/jni/riscv64/libimage_processing_util_jni.so is not an ABI

interstates21, 2024-09-10:
### How were you trying to build the app?

I updated react-native-vision-camera to v4.5.3
the app crashes at a build time.

Downgrading to v4.5.1 solves the issue

### Full build logs

```tsx
Execution failed for task ':app:mergeDebugNativeLibs'. > jni extracted from path /Users/oleksiikupin/.gradle/caches/transforms-3//transformed/jetified-camera-core-1.5.0-alpha01/jni/riscv64/libimage_processing_util_jni.so is not an ABI
```

### Project dependencies

```json
"react-native-vision-camera": "4.5.1",
```

### VisionCamera Version

4.5.1

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

DragonDare, 2024-09-11:
Even I am facing this issue with 4.5.3. Switching to 4.5.1 and modifying `camerax_version` from `1.5.0-alpha01` to `1.4.0-rc01` helped me build my app.

SKHRAPP, 2024-09-11:
@DragonDare please help to modifying camerax_version where this page located and please share screen 

getting error when make production build 

```
* What went wrong:
Execution failed for task ':app:packageStagingReleaseBundle'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.PackageBundleTask$BundleToolWorkAction
   > Unrecognized native architecture for directory 'lib/riscv64'.
```
i downgraded library from 4.5.3 to 4.5.2

amir-ibrahim, 2024-09-11:
i face this error when trying to upload the app to Play Store v4.5.3 the error is :
'riscv64' architecture is only supported for application targeting WearOS.

DragonDare, 2024-09-11:
> @DragonDare please help to modifying camerax_version where this page located and please share screen
>
> getting error when make production build
>
> ```
> * What went wrong:
> Execution failed for task [...]

Steps:
1. `"react-native-vision-camera": "4.5.1"`. Change version to 4.5.1 in package.json.
2. yarn/npm install.
3. Go to `node_modules\react-native-vision-camera\android\build.gradle` and change line 185 to `def camerax_version = "1.4.0-beta02"`.

Gradle clean and try to build again. should work

LeXuanKhanh, 2024-09-11:
go to your cache folder `/Users/oleksiikupin/.gradle/caches/transforms-3//transformed/jetified-camera-core-1.5.0-alpha01/` and delete the `riscv64` folder in `jni` file
another is using target ABI when build but I haven't tried it yet
source and explanation: 
https://stackoverflow.com/questions/78609800/agp-8-0-so-is-not-an-abi-for-3rd-party-library
https://stackoverflow.com/questions/76753627/how-to-fix-android-project-build-failing-with-so-is-not-an-abi
https://issuetracker.google.com/issues/279516850

mrousavy, 2024-09-12:
I am pretty sure this is not a VisionCamera issue. CameraX is just already ready for RISCv64, while some other native libs (like RN) are not.

Add this to your app's build.gradle:

```groovy
android {
  // ..
    defaultConfig {
      // ..
      ndk {
        abiFilters "armeabi-v7a", "arm64-v8a", "x86", "x86_64"
      }
```

closed 2024-09-12 by mrousavy

muhammadhuzaifakhalilhnhtechsolutions, 2024-09-12:
> Steps:
>
> 1. `"react-native-vision-camera": "4.5.1"`. Change version to 4.5.1 in package.json.
> 2. yarn/npm install.
> 3. Go to `node_modules\react-native-vision-camera\android\build.gradle` and change [...]

Thanks it's work for me ❤️

DragonDare, 2024-09-13:
> I am pretty sure this is not a VisionCamera issue. CameraX is just already ready for RISCv64, while some other native libs (like RN) are not.
>
> Add this to your app's build.gradle:
>
> ```groovy
> android [...]

This didn't work. Still getting the `
> jni extracted from path E:\.gradle\caches\transforms-3\59932ef85550e6bd54309370421ab708\transformed\jetified-camera-core-1.5.0-alpha01\jni\riscv64\libimage_processing_util_jni.so is not an ABI` error

DragonDare, 2024-09-13:
> go to your cache folder `/Users/oleksiikupin/.gradle/caches/transforms-3//transformed/jetified-camera-core-1.5.0-alpha01/` and delete the `riscv64` folder in `jni` file another is using target ABI [...]

This, however worked! Thank You! Now I can use the latest version of vision camera (4.5.3) as well.

rinusah, 2024-09-26:
> Steps:
>
> 1. `"react-native-vision-camera": "4.5.1"`. Change version to 4.5.1 in package.json.
> 2. yarn/npm install.
> 3. Go to `node_modules\react-native-vision-camera\android\build.gradle` and change [...]

Indeed! This solution works. It is extremely effective.

chiragramimi, 2024-10-03:
i have the same problem

bhawyaicodelabs, 2024-10-15:
> Indeed! This solution works. It is extremely effective.

Oh!!, Thankss It Workedd for me.

mddmey, 2024-10-16:
> go to your cache folder `/Users/oleksiikupin/.gradle/caches/transforms-3//transformed/jetified-camera-core-1.5.0-alpha01/` and delete the `riscv64` folder in `jni` folder another way is using target ABI when build but I haven't tried it yet source and explanation: https://stackoverflow.com/questions/78609800/agp-8-0-so-is-not-an-abi-for-3rd-party-library https://stackoverflow.com/questions/76753627/how-to-fix-android-project-build-failing-with-so-is-not-an-abi https://issuetracker.google.com/issues/279516850

Thankss.

leonardosilva97, 2024-10-23:
I'm facing an issue when trying to build my Expo project using EAS. The build works fine locally after applying a solution suggested by @DragonDare , but when I try to submit the project to EAS, I encounter the following error:

`FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:mergeReleaseNativeLibs'.
> jni extracted from path /home/expo/.gradle/caches/transforms-3/707afcf529a3c4be15bcde4b3c1b766c/transformed/jetified-camera-core-1.5.0-alpha01/jni/riscv64/libimage_processing_util_jni.so is not an ABI

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 9m 26s

Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.
You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.
For more on this, please refer to https://docs.gradle.org/8.3/userguide/command_line_interface.html#sec:command_line_warnings in the Gradle documentation.

1109 actionable tasks: 1109 executed
Error: Gradle build failed with unknown error. See logs for the "Run gradlew" phase for more information.
`

devofthings, 2024-10-24:
If someone also don't want to manually change the build.gradle. version 4.3.2 works for me without any flaws and it seems to be the last version where `node_modules\react-native-vision-camera\android\build.gradle` has `def camerax_version = "1.4.0-beta02"`. Thanks for finding the fix @DragonDare ✌🏼

illia-port, 2024-10-25:
Thanks, guys, downgrading from `"react-native-vision-camera": "^4.5.3"` to `4.5.1` worked without any additional changes.